esp8266/Arduino #92: Question: Boot Mode GPIO2 to GND what is for ?

luc-github, 2015-04-21:
Hi sorry in advance if it is a stupid question, but I have checked https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes and cannot find what it is done when starting ESP8266 (01) with GPIO2 connected to GND.
Seems not loading the FW but sending many things to serial, I cannot read on Serial monitor, so I suspect it use specific baud rate that is not in Arduino Monitor.

I originally wanted to use this configuration to create a kind of recovery mode, but as it is used already - I would like to know for what to learn something.

Anyone could tell me what it is ? or where I can find the information ?

Thanks

Links2004, 2015-04-21:
its the Bootloader witch is used to flash the application (Sketch).
the baud rate is 76800
more info:
http://www.esp8266.com/wiki/doku.php?id=getting-started-with-the-esp8266#got_an_esp-nn_board_in_the_mail_now_what

luc-github, 2015-04-22:
Thanks a lot, but this is talking about GPIO ___0___ used to upload sketch, I do not see anything about GPIO ___2___

you mean GPIO ___2___ set to GND is used to change the bootloader ? what tool is used for this ?

Thanks,

Links2004, 2015-04-22:
if you are in Bootloader mode (GPIO0 + GPIO15 LOW) then GPIO2 ist used as alternative TX to the normal TX. You can attache the RX of the USB / Serial adapter if you like.

luc-github, 2015-04-22:
OK this is good to know - thanks 

but in the configuration I mention :
GPIO0 - OPEN = HIGH
GPIO2  to GND = LOW
CH_PD to VCC
VCC to VCC
GND to GND
RX to RX
TX to TX
RST - OPEN

there is no GPIO 15 on 01 as far I can see or it is named differently

and I do not put GPIO0 to LOW

[image]

Links2004, 2015-04-22:
yes the ESP-1 don´t have this the ESP-7 or ESP-12 have it, may some other modules too (i only one ESP-7 and ESP-12).
GPIO2 to GND = LOW is a bad idea i thing then you may damage the second Uart on the device. 
GPIO2 is TX --> so its Output. If you use an pull-down there is no problem.

Links2004, 2015-04-22:
if you want to boatload something you need connect like 

this:

[image]
 
GPIO2 is left open

luc-github, 2015-04-22:
Thanks a lot,
so you mean I cannot use GPIO2 as input pull up   ? it will damage the host Uart ?
```pinMode(2, INPUT_PULLUP);```

Links2004, 2015-04-22:
no this is not a problem. 
you only get problems if you connect GPIO2 direct (or over a switch) to GND or VCC on the hardware side .
if you do it in software there is no problem. 
Your software dont run on the same time then the bootloader ;)

luc-github, 2015-04-22:
Ok thanks for your explanation, 
I already get it working but wanted to understand,  currently I use the ```pinMode(2, INPUT_PULLUP);``` and read if GPIO2 is connected to GND 10 seconds after module started, to launch a function, I understand jumper must not be connected before module start/boot is finished to handle properly the PULLUP, and anyway it was not working  

Thanks a lot for your time really appreciate,

closed 2015-04-22 by luc-github