esp8266/Arduino #94: SSID name and password for AP aren't working

lemio, 2015-04-21:
When uploading this code, a nework called ESP_A0EFD1 with no password appears instead of the network with 'test' as SSID and 'test' as password...
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
 
const char* ssid = "test";
const char* password = "test";

ESP8266WebServer server(80);
 
const int led = 13;
 
void handle_root() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  delay(100);
  digitalWrite(led, 0);
}
 
void setup(void)
{
  WiFi.mode(WIFI_AP);
  Serial.begin(115200);
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);

  // Connect to WiFi network
  WiFi.softAP(ssid, password);
  Serial.println("");

  // Wait for connection
  
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
   
  server.on("/", handle_root);
  
  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });
  
  server.begin();
  Serial.println("HTTP server started");
}
 
void loop(void)
{
  server.handleClient();
} 
```
[image]

igrr, 2015-04-23:
"test" doesn't look like a valid WPA2 password — it has to be at least 8 characters.
Perhaps you could try again with "testtest" as a password?

lemio, 2015-04-29:
That did the job, thanks! Can we prevent users from choosing to small passwords

igrr, 2015-05-12:
This is now documented in Readme.

closed 2015-05-12 by igrr