esp8266/Arduino #95: Has anyone else experienced "flakey" uploads?

tim-in-oakton, 2015-04-22:
My config is as follows:
* ESP8266 -03 modules
* Regulated 3.3v supply - arduino style battery input/USB regulator thing
* 3.3v FTDI USB:serial converter

On my first module I successfully compiled and uploaded maybe 500 times - occasionally I would need to upload twice if the first upload failed to execute correctly.  This got worse over time, it seemed like restarting the Arduino IDE and rebooting my PC was some help (hard to prove, it was intemittent).  I did try to reflash with a minimal sketch as well, that didn't really seem to help.
The program is a datalogging app for a Beer Keg - maybe 1000 lines of code or so.  Not very big.

Recently it got harder & harder to get a "good" upload (I used the serial monitor & wireless behavior to confirm if it was "working").  This morning I was completely unable to get an upload to work - the upload using the Arduino IDE would complete with no errors, but the uploaded executable image would not execute.

Using the Python ESPTool uploader, I was able to "erase" the flash, but not upload the 512K "blank" file - it would consistenly fail at 32% (hex800000).
I replaced the ESP module with another, the FIRST upload worked fine and executed perfectly, when I made a minor change to the code  however it would upload with no errors but not execute.  Frustrating.

The IDE reports:
Sketch uses 205,912 bytes (39%) of program storage space. Maximum is 524,288 bytes.
warning: unsupported baud rate: 115200, using 115200
Uploading 38112 bytes from C:\Users\timm\AppData\Local\Temp\build7944810091012342573.tmp/Kegerator-v4.3.cpp_00000.bin to flash at 0x00000000
......................................
Uploading 167848 bytes from C:\Users\timm\AppData\Local\Temp\build7944810091012342573.tmp/Kegerator-v4.3.cpp_40000.bin to flash at 0x00040000
....................................................................................................................................................................

I will try to upgrade the power supply to something I trust more than an off-brand Arduino-style regulator, but is this something that others have seen?  Are there any tricks to reliable upload&execute?

I am using the 1.6.3 IDE with a GIT pull from maybe 2 weeks ago - have there been any changes to the upload functionality or to the linker directives?

Any thoughts?  Not sure if I have a bug or a PEBCAK issue- are there any known "gotchas" to be aware of that complicate uploading?
thanks!
tim

tim-in-oakton, 2015-04-22:
I forgot to specify - Windows build, will try a fresh clone tonight-

duncan-a, 2015-04-22:
Hi

I'm now having issues like this too. Everything was fine but now it appears
to upload successfully, then nothing except 'gobbledygook' on the serial
monitor.

From what little I've been able to understand, the 'gobbledygook' is a
bootloader message at 76800 baud - but I don't have that as a baud rate
option for the serial monitor.

It's not the PSU - have tried 3 and they're all 3.3 +/- 0.03V.
It's not the FTDI Friend that I'm using as USB to Serial - that works fine
with an Arduino Pro Mini 3.3V.
The wiring hasn't changed - Vcc, CH_PD and RST (via 1K) to 3V3, GND to 0V,
ESP Rx to FTDI Tx, ESP Tx to FTDI Rx).
The method remains the same - ground GPIO0, ground and release RST, release
GPIO0, upload, ground and release RST.
It's the same problem with all modules - ESP-01 and Olimex.

I'm running Windows XP with the 1.6.1-p1 release from GitHub (I can't
fathom out all the 'pull', 'fork' and everything else to do it any other
way). I've tried closing down the IDE and restarting and I've tried hard
rebooting the laptop.

There's an occasional "4j - Error when starting" type error when I restart
the IDE but that doesn't appear to be consistent (and I have no idea what
it is anyway).

It might (though not guaranteed) be helpful to read the bootloader message
after a reset, so discovering how to get 76800 baud rate in the IDE Serial
Monitor could be a good start...

From what I can understand, it appears that either the 1.6.1-p1 IDE is
failing somehow or seven ESP modules have failed at the same time - which
seems improbable as four of them were previously OK and the other three
were brand new...

All help and advice gratefully received.......

tim-in-oakton, 2015-04-22:
Your description matches mine exactly, thanks for the 76800bps tip, I may look into that (did not know the magic number was 76800!).

I tried a third module late last night - same behavior...  first upload works, second LOOKS fine uploading, but doesn't work.  Resetting after upload shows "gobbletygook" in the serial monitor....

AFAIK I changed NOTHING in my upload environment, and I have 3 PCs used for development - all are behaving in this manner (or, changing to a different PC doesn't seem to change the upload-not-working behavior).  All PC's are WIndows7, using the earlier 1.6.1 Zip download with patched libs for the UDP port issue (thanks IGRR).

Once in this state, it doesn't appear that I can upload a functioning "minimal sketch" to a module either - although this problem seemed to "grow over time" - from an occasional failed upload to "can't upload at all anymore".  Very strange, the common factors would seem to be the IDE/uploader code, the host OS (windows7 with auto-updates)  & the power supply in my case...  I'll have a better bench supply this evening UMT-5 to test with.

Anyone else seeing flakiness?  If so, can you share your system/project specs?

duncan-a, 2015-04-22:
I don't get the "first upload works, second LOOKS fine uploading, but
doesn't work" - it's now just "LOOKS fine uploading, but doesn't work",
even with a fresh module.

I'm limited to one (rather ancient) Win XP laptop using "1.6.1 Zip
download" but without "patched libs for the UDP port issue" (I don't
understand how to do things like that!) - but I can use that IDE OK with a
Pro Micro 3.3V (Board: Pro Micro, Processor: ATMega 328 8MHz, Programmer:
AVR ISP) so that's proving that the FTDI Friend is OK and that the 'genuine
Arduino' part of the IDE is OK.

If you discover a way of adding 76800 into the options for the serial
monitor, please share it (in simple terms if possible - anything too techie
is way over my head).

abl, 2015-04-22:
Heh, my debug output works at 77400. :)

My devices are all VERY voltage-sensitive; they pull too much current for the 3.3V lines on most of my serial adapters.

My advice would be to grab PuTTY, monitor the serial port, and try to restart the module in to flash mode.

If you see a boot mode of anything but (1,7) in the debug output, something's wrong. First number is the boot type - 1 means boot from UART (aka flash over serial), 3 means boot from SPI flash (aka normal startup), 5 means boot from SDIO (aka...nobody's doing this yet.) The second number appears to be related to whether or not the chip started cleanly - 7 is good, anything less and you might not be able to flash.

duncan-a, 2015-04-22:
Thanks for joining in Alexander

Hmmm...

In my case, I'm running a 5V 1.2A wall wart (stable +/- 0.1V according to a
multimeter) and following it with an AMS117 type 3.3V regulator which,
according to the specs, should be good for 1A - again, on the multimeter,
the 3.3V rail is rock steady.

I'll try and get PuTTY installed and hope that provides an answer to the
what (the heck's happening?) - that may subsequently lead to a why (this is
happening) and a how (to fix it).

One thing I'm wondering relates to tim-in-oakton and I both using the
1.61-p1 from the .zip file on the 'releases' page - most other people are
being much more techie and doing clever things (that I can't follow) from
the regular builds on the main GitHub page. Could it be that this problem
has appeared before, has been fixed by some devious and unfathomable means
on the main page but that it hasn't made it to the .zip file?

Just a thought...

abl, 2015-04-22:
Yeah, I don't think it's your power supply - which is significantly more stable than mine - I just got VERY similar problems when I had a flaky power supply.

PuTTY should be the easiest way to get a terminal open - no install necessary, you can get it as a standalone executable.

The underlying flash tools probably haven't changed, and that appears to be where the issue lies if it isn't the chip; I'm not familiar with this particular project but I'm fairly familiar at this point (in that when one falls out of a tree, one becomes familiar with many branches) with the ESP8266 and the Arduino platform in general.

I'll dig deeper once I get home.

duncan-a, 2015-04-22:
Yup, the belief that USB can power anything and you can always get 3.3V
from an Arduino seems to be a widespread erroneous belief within some
sections of the maker movement. I may be useless with software, out of my
depth with a lot of the modern stuff and totally confused by the
proliferation of unfathomable terms, but fifty years of engineering from
the days of all valves (tubes) has taught me the value of adequate power
supplies.

I could try a chunky electrolytic close to the ESP but I really doubt it
will make any difference - the multimeter may be slow to react but I think
there should be more than enough in hand, even if the 117 is a 'Chinese
copy' of dubious pedigree...

I can't rule out ESP chip failure, but at the attrition rate I've
experienced with 'fresh from the packet' ones not working at the first try
since this problem materialised, I doubt that's at the root of the problem
- and living on a pension doesn't allow me to keep throwing them at the
problem!!!

Ho hum...............

abl, 2015-04-23:
...but it says 3.3 right on the board!

I've done some moderately heinous things to my ESPs - the flashing process is flaky, and without the debug info at that magical baud rate it's really hard to tell what's failing. I suspect your chips are fine. Looking forward to seeing what you get out of PuTTY :)

codewise-nicolas, 2015-04-23:
Thank goodness Im not the only with this problem - I thought I was going crazy last night.
Im using the lates github files with the arduino from the readme.

I started to notice problems when I was experimenting with deepsleep.
I noticed the problem when deepsleep would cause a fatal exception instead of the usual "deep sleep 15s".

I would change nothing, simply click verify then upload, and randomly it would work.

I still have not found a solution, but I think its the problem you guys are having as i believe it is programmer related.

My setup:
ESP-01 with rst <> xpdc wire
LD1117 3.3v regulator feeding the ESP
Logic level converter (https://www.sparkfun.com/products/12009)
USB-UART is a CP2102 based.
Ground is common across esp and usb-uart.

duncan-a, 2015-04-23:
Alexander

Yep, it does say 3.3V and usually it is - but it depends on what you're
trying to do with it, what else is going on with the Arduino and how the
Aeduino is being powered ...

   - Desktop/Laptop/Tablet USB ports are very 'variable' in the current
   they can supply - I've seen things crap out at 120mA before now from a
   laptop, others have been happy at 500mA.
   - The 5V and 3.3V regulators used in Arduino vary from type to type -
   and the 'clones' complicate things further.
   - The 3.3V reg follows the 5V reg so is dependent upon what's being
   drawn off 5V not just the current drawn on the 3.3V line.

Running an Arduino off USB is often a primary source of 'unexplained
problems' - particularly with things taking high peak currents - WiFi,
Bluetooth Classic, RF and Ethernet modules for instance...!!!

Back to the problem in hand...

I can't get PuTTY to work - due, I'm sure, to total incompetence and lack
of experience with terminal programs.

The USB adaptor is on Com8 so, in the Connection/Serial part of the PuTTY
config window, I'm trying to set:

Serial Line: COM8
Speed: 76800
Data Bits: 8
Stop Bits: 1
Parity: None
Flow Control: tried all options...!!!

Hitting "Open" gives me a 'Windows Bong' (feel like I could use a bong at
the moment - flashback many years!)  - no more, no less...

What am I doing wrong???

duncan-a, 2015-04-23:
Although not directly related to "the problem", I tripped over this on the
ESP8266 Community Forum:

*"Did you a add a capacitor across Vcc and GND near the ESP?I also
experienced many unwanted resets like you you have. The majority of these
resets disappeared when I added a pulup resistor from RST to Vcc. Before I
did that I could cause a reset by nearing the ESP with a hand."*

I also read something else about pickup on connecting leads causing all
sorts of problems, but I can't remember the context.

Where I'm living in Spain, there is no mains earth anywhere in the house
wiring - that may (or may not?) be significant.

Several schematics I've seen include a resistor and capacitor (to add a
delay to one of the pulses) when using an 'automatic upload' option - I'll
search for one and add details when I find it - any port in a storm!!!

One thing I have concluded in my trawling today is that there do appear to
be some issues with ESPTool - though most of the tecnicalities were over my
head. There seem to be quite a few posts relating to problems in various
configurations (using ESP8266/Arduino, direct to the ESP, etc) so we're not
the only ones with problems, though we do seem to be leading in the"it did
work, now it won't"contest.

For those who are very techie: Is there anything in the 1.6.1-p1 zip
version that might have 'filled up', 'overflowed' or anything like that?
I'm a bit loath to try a reinstall, but has anybody tried it?

Just thinking out loud...!

codewise-nicolas, 2015-04-23:
I did experience flakyness the very first few times (long ago) with the esp8266 and so the cap+resistors fixed it. Its now standard in all my ESP work.

I think this is somehow related to the compiler or the programmer (esptool).

Im trying to make it happen again so I can grab the crash report and post it for more knowleadgeable people to figure out. But to no avail.

Does this method of programming (arduino ide) also make use of the bank 0/1 concept of the ESP; that is, does it program the same file to both banks ?
Does the ESPtool verify everything it writes ?

tim-in-oakton, 2015-04-23:
ABL - thanks for the Putty recommendation, I didn't even thin to use it for Serial!!
Following are results, I uploaded the sketch, reset and reset again with the GPIO left high.....

Reset results after a “apparently successful flash upload” (Arduino IDE output):
Sketch uses 206,664 bytes (39%) of program storage space. Maximum is 524,288 bytes.
warning: unsupported baud rate: 115200, using 115200
Uploading 38224 bytes from C:\Users\timmeyer\AppData\Local\Temp\build2395631062301061534.tmp/Kegerator-v4.3.cpp_00000.bin to flash at 0x00000000
......................................
Uploading 168488 bytes from C:\Users\timmeyer\AppData\Local\Temp\build2395631062301061534.tmp/Kegerator-v4.3.cpp_40000.bin to flash at 0x00040000
.....................................................................................................................................................................

/* connect PUTTY set for 76800bps, reset while still in “upload” mode w GPIO0 pulled low */
 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
/*  as expected, "I'm cool, waiting for code delivered over the Serial port" */

/*  Now reboot/power cycle with the GPIO left to flow high – which normally SHOULD execute the previously uploaded code*/
load 0x40100000, len 30216, room 16
tail 8
chksum 0x2f
load 0x3ffe8000, len 2860, room 0
tail 12
chksum 0x21
ho 0 tail 12 room 4
load 0x3ffe8b30, len 4988, room 12
tail 0
chksum 0x8a
csum 0x8a

/*  This is repeatable – power cycle 5X and see the same exact output 5X.
I’ve tried it on three machines now (all Win7 with similar IDE downloads), no change in results.

Previously (last week) this EXACT process worked more than half the time – my code booted up and performed great.
The only changes that I am aware of is my code grew modestly, I’ve written a bit more to the EEPROM (nowhere near what I’d expect to create wear), and the original module was transported from work & back (bit of mechanical jostling).
*/

Any other ideas?  When life (work, wife, kids, etc) permits, I'm going to try to use a Linux box as the IDE host and uploader instead of the WIndows boxes used to date, and I'm planning on putting a pair of caps right on the PCB to make sure power is clean.  I may also wire other GPIO leads high as a hedge.

I did verify with a 'sillyscope that my power is solid and clean - at least at the power supply (rock solid 3.3v, no noticeable ripple).  I don't think that this is power issue now, but I'll throw some caps on to be sure.

Again - my environment was always a little flakey (maybe one upload in 5-10 failed), but it degraded over time to now be unusable - on any of the machines I'm using.

It FEELS like I hit some threshold where the upload process is breaking - setting the wrong jump vectors, putting the code in the wrong place.....  dunno.
Based on this info, any more thoughts, suggestions or observations?
thanks all, I really appreciate the positive feedback-
tim

tim-in-oakton, 2015-04-23:
Had time on a break to set up on Ubuntu - compiles and uploads without complaints but STILL doesn't run - After uploading using Linux Arduino IDE and rebooting without  GPIO held low I see (at 76800bps):

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 31024, room 16
tail 0
chksum 0x12
load 0x3ffe8000, len 2844, room 8
tail 4
chksum 0x56
load 0x3ffe8b20, len 4956, room 4
tail 8
chksum 0x66
csum 0x66

It seems a little notable that it compiles to different sizes, but I don't expect that is the root issue - I may try with a minimal sketch as well, but this is replicated behavior - two modules, four computers, two different IDE host OS's (Win7&Ubuntu 14something).

Next up is caps & terminating ALL inputs....  still think this is an upload tool issue though.

duncan-a, 2015-04-23:
*codewise-nicolas*: Things are getting a bit too technical for me but I'll
try to survive as long as possible...

My setup' is Win XP Pro with the ESP8266/Arduino .zip 1.6.1-p1 version -
Board: Generic ESP8266 board, Programmer: esptools.

It certainly appears to upload in two blocks, as shown in *tim-in-oakton*'s
latest post (thanks for that Tim - I'd still like to have understood enough
to have done that!) and it terminates with a clean "Done uploading" most of
the time - there do seem to be random errors but nothing particularly
consistent. I don't think it can be verifying as it goes  - despite taking
an incredibly long time from hitting upload to it finishing (admittedly my
laptop is incredibly ancient and slow, but it takes several minutes - I can
make a cup of coffee easily).

I think it's probably an issue with esptools but the occasional
inconsistency makes me wonder if it it one single cause...

Thanks to all for the inputs so far.

duncan-a, 2015-04-23:
*Tim...*

*Alexander* wrote in his first post in this thread:

"
*My advice would be to grab PuTTY, monitor the serial port, and try to
restart the module in to flash mode. **If you see a boot mode of anything
but (1,7) in the debug output, something's wrong. First number is the boot
type - 1 means boot from UART (aka flash over serial), 3 means boot from
SPI flash (aka normal startup), 5 means boot from SDIO (aka...nobody's
doing this yet.) The second number appears to be related to whether or not
the chip started cleanly - 7 is good, anything less and you might not be
able to flash.*"

You're seeing 3.7, which would (I think) imply that it's not reacting yo
GPIO0 correctly and is remaining in "*normal startup*" mode - if I'm
understanding correctly...

Yes? No? Maybe? Rapidly losing the will to live?

abl, 2015-04-24:
@duncan-a something else probably has your serial port captured - I find it helps to unplug, reboot, start PuTTY, and then plug in. Windows doesn't make that easy. I'm happy to set up a Teamviewer session or something to help debug, if you'd like!

@tim-in-oakton yep, boot mode (3,7) is a normal boot, so your GPIOs are not being pulled the way they need to be pulled. https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process details this a bit - you want UART mode, which is GPIO15 low, GPIO0 low, and GPIO2 high.

Links2004, 2015-04-24:
don´t connect GPIO2 to VCC, in boot-loader mode its an alternative TX connect it to VCC can damage the chip! 
see #92

igrr, 2015-04-24:
Still, a 10k pull-up on GPIO2 has improved stability (both upload time and run time) in my case.

igrr, 2015-04-24:
Also, regarding the degraded upload performance — could this be related to increased flash erase time?
If you could grab the behaviour of TX/RX lines during upload using a logic analyser or a scope, we could see if the flash erase time is an issue here.

tim-in-oakton, 2015-04-24:
I was unclear - let me try again......
When I pull GPIO0 down and reset while looking at the serial port with PUTTY, I get:
ets Jan 8 2013,rst cause:1, boot mode:(1,7)

After I upload (see above details, looks fine and it HAS worked maybe 500+ times before), I let GPIO0 float and reset the board.  Using PUTTY to listen, the board then says:
ets Jan 8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 31024, room 16
tail 0
chksum 0x12
load 0x3ffe8000, len 2844, room 8
tail 4
chksum 0x56
load 0x3ffe8b20, len 4956, room 4
tail 8
chksum 0x66
csum 0x66

I will try a local cap and pulling GPIO2 high with a 10K resistor & report back.

tim-in-oakton, 2015-04-24:
OK, I'm feeling more & more confident this is an uploader issue - I put a 470uF cap between power & gnd, pulled up GPIO0&2 w 10K resistors, and zero change in behavior. 
Using a "minimum" sketch that just opens the port & prints out "waiting" in a loop, it uploads "normally", I release GPIO0 and boot it and it displays (@76800):
 
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 26644, room 16
tail 4
chksum 0xf3
load 0x3ffe8000, len 2532, room 4
tail 0
chksum 0x3b
load 0x3ffe89f0, len 1308, room 8
tail 4
chksum 0x4a
csum 0x4a

And no longer runs or loops.

My thesis is that at a certain point/code size the uploader/bootloader steps on something on the module that breaks the normal bootup process, and as it is written to flash it gets "stuck".

Ivan - I don't have a logic analyzer handy, I do have a scope and an extra FTDI module that I *could* use to watch the interaction during an upload (now that I know the bitrate and that PUTTY speaks it)...

Could I suggest that someone on the ESP team look into the uploader Python and bootloader code to see if there is an obvious failure to 
1) erase all the flash (<-- what i think)
2) properly set the jump vector after upload to point to the uploaded code IF something unexpected was written to the flash

I have a couple more modules on order to see if I can reliably flash smaller sketches before flashing my project (larger) sketch which seems to have killed three modules now (in the same manner).

I'll bet a beer that it's something in the upload process at this point - here is a wall of text from "uploading" in verbose mode.....  The only thing that looks like a warning is the line "serialport_receive_C0: 1C instead of C0"....  any idea if that is normal or a clue?  The upload takes just under 12s at 115kbps, and the "erasing flash" line takes a surprisingly SHORT amount of time.

-----------------------------------------------------------------------

Sketch uses 178,876 bytes (34%) of program storage space. Maximum is 524,288 bytes.
F:\Users\timm\Documents\Arduino\hardware\arduino-esp8266-master/tools//esptool.exe -vv -cd none -cb 115200 -cp COM5 -ca 0x00000 -cf C:\Users\timm\AppData\Local\Temp\build3702862278599651419.tmp/BareMinimumtim_.cpp_00000.bin -ca 0x40000 -cf C:\Users\timm\AppData\Local\Temp\build3702862278599651419.tmp/BareMinimumtim_.cpp_40000.bin 
esptool v0.4.1-3-g9ab93f4 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 230400 to 115200
	setting port from COM1 to COM5
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\Users\timm\AppData\Local\Temp\build3702862278599651419.tmp/BareMinimumtim_.cpp_00000.bin success
warning: unsupported baud rate: 115200, using 115200
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	serialport_receive_C0: 1C instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_open
Uploading 30528 bytes from C:\Users\timm\AppData\Local\Temp\build3702862278599651419.tmp/BareMinimumtim_.cpp_00000.bin to flash at 0x00000000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting serial port timeouts to 5000 ms
	setting serial port timeouts to 1000 ms
	espcomm cmd: receiving 2 bytes of data
	writing flash
..............................
	setting address from 0x00000000 to 0x00040000
	espcomm_upload_file
	stat C:\Users\timm\AppData\Local\Temp\build3702862278599651419.tmp/BareMinimumtim_.cpp_40000.bin success
	espcomm_open
Uploading 148392 bytes from C:\Users\timm\AppData\Local\Temp\build3702862278599651419.tmp/BareMinimumtim_.cpp_40000.bin to flash at 0x00040000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting serial port timeouts to 5000 ms
	setting serial port timeouts to 1000 ms
	espcomm cmd: receiving 2 bytes of data
	writing flash
.................................................................................................................................................
starting app without reboot
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
-----------------------------------------------------------------------

AndyKirby, 2015-04-24:
There are so many different ideas on how the pins go.

The following works every time for me and has been used by folk from our
hackspace with equally reliable results.

http://www.sheffieldhardwarehackers.org.uk/wordpress/

Pull up = 10K resistor to Vcc (3v3)
Pull down = 10K resistor to Gnd (0v)

ESP8266-01

Using 10K resistors

Pull up, GPIO0
Pull up, GPIO2
Pull up, Rst (Reset)
Pull up, CH_PD

To program using the esp bootloader (like you are doing with the arduino
IDE)

Short GPIO0 to Gnd for the duration of loading the code.
Momentarily short RST to Gnd to trigger a reset or alternatively
powercycle the board.

This puts the bootloader into listening mode and you can then load your
code onto the board.

When loading is complete

Remove the Gnd short on GPIO0 and do a reset again.

Board comes up and runs your app fine.

This also works fine on other of the variants of the ESP8266 like 03 and
07. except ones with GPIO15, this should be pulled down to gnd via a 10K
resistor.

https://hackaday.io/project/3253-esp8266-native/log/11726-deep-sleep

10K is a weak pull up/down value that will not do the device any harm if
a pin is brought up in a output mode (Spare serial TX for instance). the
current is about .33 mA so tiny. But will ensure that if in an input
mode it has the default level set so it will always be known what state
it is in. The states as noted above work.

Warning if you leave a pin floating (Unconnected) it can be in any of
the two states or oscillating rapidly between the two depending on how
long a bit of wire is connected to it and how effective it is at being
an antenna. As the pin is in a random state, the results of any action
that is dependant on the state of the pin (Like bootloading) will be
just as random.

Do not put capacitors unnecessarily across pins, they are not pull ups.
Only use resistors for pulling pins up or down.

The little bit of effort taken to pull the important pins reliably into
their necessary default configurations is time and effort well spent.

Dunno if this is useful or already known but thought it worth repeating
just in case. Context and understanding can sometimes be assumed that
isn’t there.

Cheers

Andy Kirby

duncan-a, 2015-04-24:
*Tim*

I'm close to being convinced irs an uploader issue, but I don't have the
ability to suss what it might be.

It seems to me that it's only affecting those who are working from the
ESP8266/Arduino/Releases .zip file version (or we're the only ones who
don't know how to integrate a 'build' or whatever) and I can't help feeling
that there may be some significance in that, but tomorrow I will lay the
whole thing out on the breadboard with the shortest possible wiring between
point*s *in an attenpt to reduce the possibility of extraneous pickup - I
don't think it's a factor but as Sherlock Holmes said "*When you have
eliminated the impossible, whatever remains, however improbable, must be
the truth*"...

*Andy*

Why did you, and the people at your hackspace, reject the idea of
'automating' the upload sequence using DTR/RTS and CTS combination? I'm
just curious as I would have thought the brains, experience and test
equipment would have been in plentiful supply. I've seen conflicting
methods being suggested, so I've avoided it so far.

The capacitor being referred to, of which you disapprove, was a 'chunky
electrolytic' across the 3.3V rail to reduce the likelihood of PSU Current
issues so nothing rounding-off signals. I agree completely about pullups
and you were right to point it out .

The Olimex modules come with 2Ks built-in (default high with a soldered
jumper) and this failure to upload applied to that board immediately after
an 01 (with 10Ks) failed, so, in my particular case, that can't be part of
the problem...

This is definitely a three pipe problem..........!

duncan-a, 2015-04-24:
Although this thread <https://github.com/esp8266/Arduino/issues/101> isn't
(I don't think?) directly related to the upload problem, I was struck by
the line "*This should have been fixed in db86766
<https://github.com/esp8266/Arduino/commit/db86766b366310c3c4b9df919bcf8acf7f11cf33>,
discussed in #98 <https://github.com/esp8266/Arduino/issues/98>. Are you
using the latest version from git?*". It seems to reinforce my thinking
that the code is continually being updated, improved and bug-fixed (with
the odd hiccup - see here <https://github.com/esp8266/Arduino/issues/104>)
- but only for those who are able to understand and implement "*the last
commit on the repo*" (whatever that means), and they're not migrating to
the .zip file on the /Releases/ page...

I may be totally wrong about this so feel free to correct me - or better
still, can somebody explain what those of us who are not 'hyper-techie'
need to do to maintain what we have at the current (believed correct so
far) status...

codewise-nicolas, 2015-04-24:
Iam using the latest version from git, and have the problem.
Its not too complicated to update the "release" version with the git files.
Simply do the "Download Zip" button, and grab the folders inside the "Hardware" folder and replace completely your local ones.

For those having the problem intermitently, try archiving the build bins, so that when you find a compile/upload that has problems you can compare with the same code in a working .bin.
That is, compile upload, fail, copy the .bin, re-compile, re-upload, success, compare this bin to bad bin (should be the same.

This will help determine if its a compile thing or an upload thing.

duncan-a, 2015-04-25:
That sounds easy - I'll try that tomorrow (it's 2am now and I was up at 6am
after 4 hours sleep, and that's not good at my age).

I've been wallowing in a swamp of jargon in various posts - pull requests
and various other things - maybe you should get your simple explanation
added to the  ReadMe file in the GitHub page...?

The problem (in all cases, I think) appears to be uploading as what did
work now won't.  Fore some people, it seemed to go through an intermittent
phase before failing completely, whereby it compiles without problems (but
it could be that the compiler is missing something) and it appears to
upload correctly, but it just does nothing.

As it's completely dead with me now, I'll need to leave it to one of the
others to try comparing .bins...

duncan-a, 2015-04-25:
Could this be relevant - it relates to compiling...?

I'd be quite happy to have an ESP-Only Arduino IDE (and call it ESP-Ard or
something) if I could get working ESPs again.......................

duncan-a, 2015-04-25:
Stupid boy, Pike.......

Forgot the link: https://github.com/esp8266/Arduino/issues/110

tim-in-oakton, 2015-04-25:
Update, no progress but have eliminated possible sources - this is an upload issue I am pretty sure (or a compiler/upload issue) - things I've tried in the last day:
* Wiring cleaned up, short runs, validated power, big cap across power & gnd - no impact.
* tried using Linux build from our github - no change.  Program compiled and uploaded fine (no errors or warnings), still would not longer run.  Same behavior using WIn7 and Linux.
* After uploading and I release GPIO0 from ground, it boots and reports checksums it seems happy with - but doesn't ACTUALLY start running my code (which is now a minimal sketch which opens serial port and Serial.printrf's a message - doesn't run.

So - every hardware recommendation made here has been tried (I am an EE if that's worth much credibility), multiple modules have been used to reproduce the reported issue, multiple PS, FTDI modules have been swapped with zero impact to symptoms.

This was working for me for weeks (albeit with some flakiness where I had to reboot the host PC or exit/reenter Arduino to get an upload to work occasionally), but as of last week my original module refused to execute code that "allegedly" uploaded fine.  Replacing the module, I was able to upload the minimal sketch (worked first time), my large beer-keg sketch (worked the first time), but the SECOND upload of the same compiled code appears to have "broken" the module - since then it has not been able to execute any uploaded code.  I repeated this with a THIRD module with the exact same results.

Is there a tested approach for clearing out ALL the module flash?  
How does one clear ALL flash with certainty?
thanks for suggestions-
tim

duncan-a, 2015-04-25:
Tim

Do the numbers change within the bootloader messages to PuTTY after each
upload and run attempt? Does anything else change from each (possible)
upload attempt?

I think I can see where you're going with the 'clear ALL flash' approach,
but I'm not able to get that 'first upload works, everything else fails to
run condition' with a fresh module.

I've done everything you've done, plus tried other USB cables - all that's
left is the compile/upload sequence as far as I can see.

Do any of the more techie readers of this long running saga, have an easy
way to determine if the 'esptools' contained within the Releases/ .zip file
have changed subsequently? Here I'm assuming that's the only part of the
'composite IDE' that's involved with the compile/upload - but I could be
way off beam with that conclusion...

The .bin comparison as suggested by Nicolas is probably exactly what I'm
talking about (?) but doing that manually would be near impossible I
suspect, so it would need some sort of comparison program - but how the
heck a bit difference, somewhere in the middle, would be traced back to a
particular statement, is in a different dimension to me...!!!

The same sort of minimum sketch uploads and runs fine on a 3.3V Pro Mini,
so it relates to the 'spliced-in' ESP parts without a doubt in my book.

tim-in-oakton, 2015-04-26:
Using a checksum app, the following minimal sketch compiles to the same binaries (bleh.cpp_00000.bin, bleh.cpp_40000.bin) on two PC's and is consistently returning the same checksums from the ESP when it boots (but the code itself is NOT executed, just a boot message that it's in mode 3,7 and it sees code with checksum :

void setup() {
  delay(1000);
  //  set up the UART for 76800 (yes, I tried other speeds as well, this is handy to let PUTTY watch for boot msg & output)
  Serial.begin(76800);
  delay(200);
  Serial.println("booting ESP8266 - ");
}

void loop() {
  delay(1000);
  Serial.println("doing nothing, noisily.....");
}

uploads with no errors observed (although "erasing flash" goes by instantly) - on boot I see:
ets Jan  8 2013,rst cause:1, boot mode:(3,7)
load 0x40100000, len 26644, room 16
tail 4
chksum 0x42
load 0x3ffe8000, len 2532, room 4
tail 0
chksum 0xee
load 0x3ffe89f0, len 1324, room 8
tail 4
chksum 0x45
csum 0x45

So, same bins, same upload behavior/messages, same boot messages repeatedly, but still NOT WORKING consistently.
I'm going to have a look at the flash erase process....  it feels like the problem is in ESPTools getting into a corner case where is writes something wrong then cannot "overwrite" the wrongness...

I have a bunch more modules on order, that will let me recreate.

IF anyone wants to try the sketch that has broken my modules, I'll send it by email to them - 
thanks -
tim

tim-in-oakton, 2015-04-26:
***  ROOT ISSUE - ESPTool.exe is NOT correctly erasing flash, allows module to get "borked" and is then "unborkable".
Above wall of text can be summarized as "sometimes on windows modules get to a state where uploads stop working", and many people suggest many different hardware-related fixes.  They didn't work for me, and simply grabbing the esptool.py script and running:
F:\Users\timm\Documents\Arduino\PythonESPTool>python esptool.py --port COM5 --baud 76800  erase_flash
put the module back into an uploadable state.
Manually asking esptool.exe to erase the module did not work....  so esptool.exe as packaged here has a bug wrt erasing that esptool.py doesn't.

So - for all other frustrated windows users, if you bork your module and it stops working, do the following until this bug is fixed:
* install Python2.7 (google for instructions)
* download esptool.py from this git distro into a local directory (hope this link keeps working - https://github.com/themadinventor/esptool, I could not find in this GIT repo)
* open a command window in the same directory and run "python esptool.py --port COM5 --baud 76800  erase_flash"
* use your IDE to download your prev sketch, should now work again.

I'm going to try to diddle the platform.txt file to use this instead of the esptool.exe, if it is easy I'll report back.
This may be hard to reproduce, if I find a sketch that consistently breaks esptool.exe I will share, but please comment IF this approach helps you out of a "stuck" module so the smart guys (Ivan, looking at you) can prioritize and look into whether it's worth fixing/replacing esptool.exe.

Cheers all, HTH someone else!
tim (going back to putting his beer taps online and talking to AWS SNS service)

GerryKeely, 2015-04-26:
Hi
 just add the following at bottom of  platform.txt

tools.pyesptool.cmd=esptool
tools.pyesptool.cmd.windows=esptool.exe
tools.pyesptool.path={runtime.ide.path}\hardware\tools\esp8266\pyesptool
tools.pyesptool.program.params.verbose=""
tools.pyesptool.program.params.quiet="" 
tools.pyesptool.program.pattern=python "{path}\{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}\{build.project_name}_00000.bin"  0x40000 "{build.path}\{build.project_name}_40000.bin"

Gerry

duncan-a, 2015-04-26:
Tim

Is there any way to ensure that this gets picked up by those who are
developing the 'product' to ensure a fix is incorporated?

I don't follow your method completely (but I'll have a go) - thank you for
all your hard work in finding the solution - your sainthood will be in the
post............!!!

Duncan

igrr, 2015-04-26:
@tim-in-oakton Great find! Do you have the compiled sketch that causes the module to get into a "failed" state?
If you enable verbose upload logs in Arduino preferences you will see it showing the path to the binaries before the upload happens. Can you post those two binaries somewhere?

duncan-a, 2015-04-26:
*tim-in-oakton*

Have been thinking (marginally easier than trying to work out how to
implement the fix)...

New module out of the packet - upload a few times and it fails - module's
flash memory fills up because it's not being cleared - yes?

How does that explain the situation of...

New module out of the packet - fails to upload first time, once the problem
has occurred on the previous module? In this case, we're back to a state
that matches the first pass on the first module - are we not? In which
case, it *should* work - no?

I'm not doubting, or playing 'devil's advocate', but what am I
misunderstanding?

*GerryKeely*

"just add the following at bottom of platform.txt" - At what step in the
process outlined by tim-in-oakton? Which of the "platform.txt" files?
According to a search, I have "Platform.txt" in:

   - \hardware\arduno\avr (6KB)
   - \hardware\arduno\sam (6KB)
   - \hardware\esp8266com\esp8266 (6KB)

Does this provide a 'permanent workaround' (until whatever needs to be done
is done)?

Sorry if I'm seeming dense, but this is all over my head - I had a module
that worked fine, then 'the problem' occurred and I haven't been able to
load (and run) to any module since...

duncan-a, 2015-04-26:
Whoops... Need a sample from Tim's beer taps....

Should have read:

   - \hardware\arduno\avr (6KB)
   - \hardware\arduno\sam (*5*KB)
   - \hardware\esp8266com\esp8266 (*5*KB)

tim-in-oakton, 2015-04-26:
Apparently it isn't a "killer sketch" -  there must be some other cause for the state, as I tested/modified/uploaded/tested the last version of my sketch without ANY issues.

Possibly it is some accumulation of factors, as the problem of flakey uploads did get get worse over time before it finally stopped for good (until I used the python tool to properly erase the flash.

The next time I see this state I will definitely capture the bins - but intuition is telling me that this is an issue of accumulated errors/baggage

If anyone else finds this helpful, please post your details of how you got into the state.

tim-in-oakton, 2015-04-26:
Duncan - try to follow these steps & report back:

    install Python2.7 (google for instructions)
    download esptool.py from this git distro into a local directory (hope this link keeps working - https://github.com/themadinventor/esptool, I could not find in this GIT repo)
    open a command window in the same directory and run "python esptool.py --port COM5 --baud 76800 erase_flash"

This is another tool used to upload to the ESP8266, and in my case it could properly erase the Flash memory.  I don't think it is a case of the flash "filling up", but of some incorrect or troublesome data being written (for some reason - a software bug, me interrupting the upload, alpha particle, electical noise, poor power...  dunno & doesn't matter).  In my case, this troublesome data prevented booting with subsequently uploaded sketches.  Properly erasing the sketch with the above tool fixed it.

HTH
tim

PS - we'll be publishing the keg sketch shortly, will post a link here -

GerryKeely, 2015-04-26:
Hi
The additions I mentioned are to be included in the files located in the esp8266 directory.
I too had problems with the esptool.exe but since changing to the esptool.py I have had no problems.
Gerry

duncan-a, 2015-04-26:
OK, I'll do my best to try it tomorrow when I get back from the hospital
(if I'm feeling up to it).

I used to work in satellites (not physically 'in' them, obviously) so I
understood the alpha particle reference (is marriage a 'single event upset'
that can devastate the norm?) - what I'm not understanding is why
subsequent uploads don't overwrite the "incorrect or troublesome data", and
why the problem should replicate to a brand new module at the first try...

If you want to send a full beer keg and ESP8266, I'm happy to beta test
your project!!!

duncan-a, 2015-04-27:
*Tim*

With incompetence coursing through my bloodstream (and a slowly diminishing
volume of left-over sedative from earlier), I embarked on the process:

   - Downloaded and reinstalled Python 2.7.
   - Downloaded the zip file containing esptool.py, unzipped it, copied the
   folder 'esptool-master' to program files (ok?) and renamed it "esptool".
   - Ground to a halt...

How do I "open a command window in the same directory"? Presumably you're
meaning the newly renamed "esptool" for 'same directory' in my case?

Assuming it works (I have no doubts about your competence, it's my own I
doubt!), I'm proposing to use the following as the 'Minimum Expectation of
Success Sketch" (MESS)...

#include <ESP8266WiFi.h>
void setup()
{
  Serial.begin(115200);
  delay(10);
  Serial.println("Setup");
  Serial.println();
}
void loop()
{
  Serial.println("Loop");
  delay(1000);
}

Does that seem reasonable?

I realise that the #include isn't necessary, but I'll build this sketch
back up to what I need, when things work again, and it prevents me
forgetting it later!

I gave "the problem" and "the solution" a lot of thought at the hospital
(interrupted by nurses in 'scrubs', who always get my full attention!) and
I'm still not sure what's going on that would prevent it uploading to a new
module that hasn't had the problem - unless, somehow, when something goes
wrong with the upload, something becomes set in the 'uploader' that doesn't
get reset (but which* is* being reset by the 'erase_flash' method.

From what I can work out, it's only Windows that is affected - the Mac
users don't seem to have this problem and the Linux users are seemingly
below the radar. Is that a fair conclusion?

If that *is* the case, then there's something different in the basic
process between the two versions - obviously there will be different code
for the two OSs, but I'm thinking of the process steps themselves, not how
they are being implemented...

Duncan

tim-in-oakton, 2015-04-27:
Duncan - hope your recovery is uneventful and restful!
Your sketch is reasonable - same basic sketch as I chose.
Where you ground to a halt is easy to get past-
"Open a command window there" - I would have said "open a dos prompt" but apparently it's not called DOS anymore.  WHat I mean is "open a DOS command shell, and then cd to the directory where the python script (esptool.py) is located.  Being lazy, I navigate there in windows, hold down SHIFT and then right click in that windows that affords me the choice to "Open a command windo here".  I *think* XP did the same thing, otherwise put the python file (esptool.py) somewhere like C:/dumperoo/esptool.py then open a dos shell and type "cd c:/dumperoo" then the command to run the tool:
python esptool.py --port COM5 --baud 76800 erase_flash

That should give the ESP8266 a high colonic and allow normal operation to resume - it sure did for me!
Cheers-
tim

abl, 2015-04-28:
I've been in and out of the hospital myself. Fun times!

Boot mode (3,7) means a hardware problem. In order for the ESP8266 to be flashed, it needs to be in boot mode (1,7). If it enters (3,7), your GPIOs were not set correctly - it's entirely possible this is due to esptool.exe (or another esptool binary) vs esptool.py.

There's no "standard" way to reset/prep-to-flash the ESP8266 via serial - there's a semistandard way, but it's not part of the spec in the way it is with Arduino.

When you use esptool.py to erase the flash, it's running its own reset-to-bootloader code and then performing an erase. When it's done, it leaves the module in flash mode (i.e. (1,7)).

So I don't think this has anything to do with erasing the flash - when you upload a new flash you're overwriting everything relevant anyways - but it does mean someone could extract the ~10 lines or so of Python that esptool.py uses to put the chip in flash mode and save you the time it takes to run an erase. :)

Oh - the way the reset-to-bootloader works is by toggling signals on the COM port in a way that's not really part of the original intent of said signals. Windows serial is...quirky, and the PySerial library is pretty mature and is well aware of those quirks.

duncan-a, 2015-04-28:
*Tim*

If the high colonic doesn't fix the problem, then it's going to be "Pass
the camera on a wobbly stick - Nurse, the screens please..."- then we'll
stand a chance of extracting the Icelandic songstress (you said the modules
were becoming "borked").

I fell asleep in the chair last night - I'll let you have feedback later...

*Alexander*

Forgive my ignorance, but are you saying that the problem lies within
esptool as implemented within the Windows 'package' on GitHub?

I'm still confused about the sequence of events as I experienced them:
Worked fine with an Olimex module (several times)
Worked fine with an ESP-01 (several times)
Failed to work with an ESP-01 (naive conclusion that module had failed)
Tried another (previously working) ESP-01 - nothing
Tried the (previously working) Olimex module - nothing
Tried a new (fresh out of the bag) ESP-01 - nothing

I'm not seeing how Tim's solution of the Flash erase with esptool.py
manages to then 'unlock' esptool.exe (if I'm understanding correctly) so
that it then begins functioning correctly again for uploading.

If esptool.exe was leaving a module in the wrong state after an upload
(either in 1,7 on exit or 3.7 preventing further uploads) then I can see
that a pass with esptool.py would (from what you describe) get the module
back in the correct state - but the subsequent code upload with esptool.exe
would then recreate the problem wouldn't it?

I'm not doubting anybody on this, I'm just trying to understand it...

What's the long term solution - a rewrite of esptool.exe, an automatic link
to use esptool.py at the critical moment, get a mortgage and buy a Mac???

duncan-a, 2015-04-28:
*Tim*

All I'm getting is* 'python' is not recognised as an internal or external
command, operable program or batch file.*

I did it as per instructions, even down to the folder name of "Dumperoo"
(back to the high colonic irrigation again!) and I tried running the setup
file within that, and repeating...

Any ideas???

It's probably me failing to do something that, to you, was an obvious step.

oshlab, 2015-04-28:
Duncan, It sounds like you do not have python installed. https://www.python.org/downloads/ be sure to get version 2.7.9. That script didn't work for me with 3+. It didn't work for me at all, my esp-03 is still behaving radically, but I got the script to run at least. You also have to add python to your home path. I believe the installer has an option to do it at the bottom of packages to install, if not, you can do it as described here http://superuser.com/questions/143119/how-to-add-python-to-the-windows-path.

duncan-a, 2015-04-28:
*OSHLab*

Thanks for that - I've got 2.7 but not sure which flavour...

*Tim* (et al)

After the failed Flash erase attempt (my last post - and it feels like I'm
hearing that bugle!), I took a time-out for a walk around the village.
Believing, as I do, that all programming is some form of magic, I had the
bright idea that trying to use esptool.py might have snapped esptool.exe
out of its lethargy at the thought of impending unemployment. So...

I came back, tried the 'MESS' (see earlier in the saga) and, lo and behold, *it
worked*, tried another 01 module, worked again - !well, whoop my pee!, I
thought and got bold...

I uploaded the real sketch (that had worked before this saga began) and it
uploaded *and* ran but wouldn't connect (those dots get hypnotic after a
while) so I tried the other module and got an error - my fault, connected
GPIO0 to the wrong GND row on the breadboard (floating, so nothing serious).

Tried again. It appeared to upload OK but the problem revealed itself again
- all modules are now back where they were...

It may be that the attempt to upload with GPIO0 high is locking something
up (that could be helpful to know) but why the failed attempt to erase
Flash should have affected a cure (however temporary) is confirming my
belief that the dark arts are real and 'software people' inhabit the inner
circle.

The failure to connect to the WiFi LAN may be an actual board failure, I
have no way of knowing - or of testing another at the moment.

Between the original 'not working' and the temporary 'working' the only
possibly related change was the attempt to run esptool.py - the machine
wasn't switched off, rebooted or anything else that might be relevant in my
'hardware of the valve (tube) era' thinking.

I may repeat the Flash erase attempt under the same conditions (python not
installed), just to see if the 'fix' happens again, but I suspect it was a
one-off that was actually unconnected (might it be time related - millis
counter rolling over?).

I'm baffled for now - need chamomile tea and a lay down in a dark
room..................

duncan-a, 2015-04-28:
"The problem" seems to be cropping up elsewhere now - see here
<https://github.com/esp8266/Arduino/issues/130> and here
<https://github.com/esp8266/Arduino/issues/141> (scroll down to the third
post)...

Maybe ESP is for *E*xtra *S*ensory *P*erception and our systems are being
controlled by the brainwaves of a 12 year old savant in Minsk?!?!?

duncan-a, 2015-04-29:
*Tim* and* OSHLab*

I couldn't determine which version of Python 2.7 I had (it was installed),
so I downloaded and installed 2.7.9 and selected the option to add it to
the path (deselected by default).

I tried again but got the same message *'python' is not recognised as an
internal or external command, operable program or batch file*, so I thought
I'd try rebooting the laptop.

That seems to have got Python 'into the system' but running the command to
erase flash is now giving a different error:

Traceback <most recent call last>:
  File "esptool.py", line 22, in <module>
    import serial
Import Error: No module named serial

Any ideas???

I seem to be closer, but still no cigar...

On 28 April 2015 at 21:41, Duncan Amos <duncan.oatlands@gmail.com> wrote:

> "The problem" seems to be cropping up elsewhere now - see here
> <https://github.com/esp8266/Arduino/issues/130> and here
> <https://github.com/esp8266/Arduino/issues/141> (scroll down to the third [...]

abl, 2015-04-29:
When your path gets updated, you have to restart programs before they see the new path. Rebooting works but is usually overkill :)

You need PySerial for the "serial" module: https://pypi.python.org/pypi/pyserial

Specifically you want the 2.7-win32 installer: https://pypi.python.org/packages/any/p/pyserial/pyserial-2.7.win32.exe#md5=21555387937eeb79126cde25abee4b35

duncan-a, 2015-04-29:
*Alexander* and *Tim*

Thanks for that. Made a difference but I still can't erase Flash

Still hasn't worked though. I used Tim's line "python esptool.py --port
COM5 --baud 76800 erase_flash" and changed COM5 to COM8, which resulted in
an error message:

Traceback (most recent call last):
  File "esptool.py", line 472, in <module>
    esp = ESPROM(args.port, args.baud)
  File "esptool.py", line 66, in __init__
    self._port = serial.Serial(port, baud)
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 38, in
__init
__
    SerialBase.__init__(self, *args, **kwargs)
  File "C:\Python27\lib\site-packages\serial\serialutil.py", line 282, in
__init
__
    self.open()
  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 66, in
open
    raise SerialException("could not open port %r: %r" % (self.portstr,
ctypes.W
inError()))
serial.serialutil.SerialException: could not open port 'COM8':
WindowsError(5, '
Access is denied.')

The will to live is ebbing rapidly...!!!

abl, 2015-04-29:
Something else has COM8 open or COM8 is an invalid port. Unfortunately Windows is kind of awful with serial ports - it's really hard to figure out what has a port stuck.

Does this happen after a reboot (and before running anything else)? Does COM8 show up in device manager?

duncan-a, 2015-04-29:
Whoops I had the IDE open in the background... Duh...

That seems to have worked for the erasing (there's no confirmation, but no
error either) - once I'd realised that the module has to be in 'upload
mode' (GPIO0 Low, toggle Reset).

Is there a way to 'cut and paste' into the command line? I've got a lot of
these to do and if they 'lock up' in future it's a lot of typing when you
have cataracts. For some reason, 'batch files' is in my head but I didn't
understand those when I was younger.

I'll try uploading a sketch after I've had celebratory baked beans on
toast...

abl, 2015-04-29:
Yes - right-click and there's a paste option (there's no keyboard command for it IIRC) - you can also right-click and choose mark, select text, and press enter to copy it.

Batch files are just text files with a list of commands and a ".bat" extension; you could create eraseflash.bat, add the commands you want, and then type "eraseflash.bat" (in the same directory where you created the batch file) to run those commands.

duncan-a, 2015-04-29:
*Alexander*
Thanks that's going to save a lot of time.

*Tim (et al)*
Partial success - the modules are accepting uploads again but things aren't
quite right...

I'll investigate more tomorrow, I'm exhausted...

duncan-a, 2015-04-29:
It's now 01:30 and I got out of bed to try something...

I didn't erase the flash memory on the Olimex module, only the 01s.

I just uploaded to the Olimex and it's working again ot was locked up like
the 01s...

Conclusion? Well, whatever running esptool.py and its erase_flash is doing,
it's a) also able to erase the flash on unconnected, unpowered, modules on
an adjacent breadboard, or b) it's unlocking something within the
IDE/esptool combination, or c) there's something else going on...

Over to those who understand much more than I do..............

This may be relevant - I found it at
https://github.com/esp8266/Arduino/issues/130 down towards the bottom.

*Short summary of the bug in SPIEraseArea (thanks to SpLab from
esp8266.ru/forum <http://esp8266.ru/forum>): SPIEraseArea function in the
esp8266 ROM has a bug which causes extra area to be erased. If the address
range to be erased crosses the block boundary, then extra head_sector_count
sectors are erased. If the address range doesn't cross the block boundary,
then extra total_sector_count sectors are erased. Where head_sector_count
is the number of sectors from the start of erased area to the block
boundary, and total_sector_count is the total number of sectors to be
erased. Sector size is 4096 bytes and block is 16 sectors.*

olivernz, 2015-04-30:
Hey Duncan,

I did notice something odd. Did the esptool erase and it worked fine for a
while and then it always got errors half way through. Nothing I did would
work. So walked away to get a coffee, came back, erased, all working again
(ESP-01).

I'm wondering if there is a heat/static/whatever issue that builds up if
you recycle the flash too often. Once it has some breathing time they come
back.

Just an observation (still waiting on my ESP-12's so can't confirm for
them).

Cheers Oliver

[35 more replies]

closed 2017-09-26 by devyte