esp8266/Arduino #96: Question: is mDNS suposed to work in AP mode ?

luc-github, 2015-04-22:
Hi I have tried mDNS sample in Station Mode, and it works very well - excellent!
But I have issue in AP mode:
I have modified the sample code to start in AP mode without password
```
  // Connect to WiFi network
  //WiFi.begin(ssid, password);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid);
```
and started the  using AP IP
``` if (!mdns.begin("esp8266", WiFi.softAPIP())) {```

webserver answer as expected on browser if using ip address (normal),  but no answer when using http://esp8266.local - seems mDNS started without issue and I do not see any error on serial.

Is that normal ? or do I miss to setup something ?

Thanks,

Luc

igrr, 2015-04-27:
I have tried this and it seems that multicast doesn't work over softAP interface.
For multicast destinations udp_send returns error code 1, which is not one of lwIP error codes (which are all negative), so it must be coming from the lower level (netif->output).
This looks like an issue with Espressif's softAP interface.

luc-github, 2015-04-28:
Ok thanks a lot for your answer 
so I close the issue or you want to keep a tracker ?

luc-github, 2015-09-07:
Verified fixed - sorry for the delay thanks again for your great work

gregslomin, 2015-10-07:
I cannot seem to make this work. Using the current staging board manager build .begin("esp8266", WiFi.softAPIP()) and .begin("esp8266") both return false.

luc-github, 2015-10-08:
I have same issue -  seems something is broken since last test
No answer on STA neither AP on OSX/Windows/Android

luc-github, 2015-10-16:
Seems using a String and c_str() is not working well and even can lead to exception 28
like:
```
String shost = "esp8266";
.....
if (!mdns.begin(shost.c_str())) ...
```
but using an array of char is ok,  like : 
```
char hostname [MAX_HOSTNAME_LENGTH+1];
...
strcpy (hostname,"ESP8266_008261");
...
if (!mdns.begin(hostname)) ...
```
now both AP and STA are answering

closed 2015-10-16 by luc-github

gregslomin, 2015-10-16:
Is this an issue with the String class or an issue with const char pointers? Does the ESP8266 attempt to append a .local on the end of the given hostname and reuse the same memory?

luc-github, 2015-10-16:
looking at the mDNS code, the content is copied so should not have issue 
 ```
// Copy in domain characters as lowercase
  for (int i = 0; i < n; ++i)
    _hostName[i] = tolower(domain[i]);
  _hostName[n] = '\0';
```

karlbackstrom, 2015-10-20:
I noticed I needed "WiFi.mode(WIFI_AP);" before "WiFi.softAP(ssid);"
Without setting mode  .begin("esp8266") returns false.

This worked both with a String and the strcpy above.

teprom, 2017-07-20:
Hey there,

I'm sorry for reopening this old thread after almost 2 years, but I have the same issue.
Does someone have the solution for that ?

Kind regards.

devyte, 2017-07-20:
It works for me

creatormir, 2020-02-05:
Use [Captive Portal ](https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino) for redirect. Perfect!

papadenis, 2020-08-07:
[I think I went through this problem](https://github.com/esp8266/Arduino/issues/2602#issuecomment-670414843)
#2602