esp8266/Arduino #97: Unable to use upload speed greater than 115200

einsteinx2, 2015-04-22:
The 256000 gives the following error:

    Uploading 37456 bytes from /var/folders/qz/6zrpphk11_ddxstvt5tk21740000gn/T/build394505067532361022.tmp/ESP_i2c_OLED_Test2.cpp_00000.bin to flash at 0x00000000
    warning: espcomm cmd: didn't receive command response
    warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed

The 921600 gives the following error:

    Uploading 37456 bytes from /var/folders/qz/6zrpphk11_ddxstvt5tk21740000gn/T/build394505067532361022.tmp/ESP_i2c_OLED_Test2.cpp_00000.bin to flash at 0x00000000
    warning: espcomm cmd: wrong direction/command: 0x01 0x08, expected 0x01 0x02
    warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed

igrr, 2015-04-23:
Are you able to use 256000 and 921600 using esptool.py, or any other upload tool?
Having long wires between the usb-uart converter and ESP will likely make uploads at 921600 impossible.

duncan-a, 2015-04-23:
Can you still upload at slower baudrates?

I only ask because several people are having problems with uploading - did
work, now doesn't (see here <https://github.com/esp8266/Arduino/issues/95>)
- and, somewhere along the process of trying, I have a feeling that I've
seen both of those errors

igrr, 2015-04-24:
The first error,
```
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
```
means that esptool has not received the response from flash erase command. Currently the timeout is 5 seconds (and it's 10 seconds in esptool.py)

The second error,
```
warning: espcomm cmd: wrong direction/command: 0x01 0x08, expected 0x01 0x02
```
means that there was some garbage on the serial line from previous communication, which was not flushed for some reason.
I could look at those issues, but the problem is that I can't reproduce those upload errors. Everything, unfortunately, works smoothly for me.

einsteinx2, 2015-04-25:
@duncan-a I'm able to flash fine almost every try by using the 115200 speed. Sometimes it doesn't work the first time (seems like it's not waiting long enough for it to reset maybe), but always works the second time. The error I get sometimes with 115200 is different than these errors.

 I've never gotten the other speeds to work. I haven't tried directly with esptool though.

@igrr Hmm maybe it's a difference with my FTDI cable? What FTDI cable/programmer are you using?

duncan-a, 2015-04-25:
I'm confused - how are you setting the upload speed there's no option for
that in the menus?

What's the default upload speed in the version within the Windows version
of the .zip file on the Releases page, do you know?

Given that I'm running an ancient laptop, speed could be an issue - though
it did work...

einsteinx2, 2015-04-25:
@duncan-a I'm on OS X, and in the Arduino IDE compiled from this repo, I have 256000 and 921600 in the upload speed menu...

duncan-a, 2015-04-25:
Hmmm - the upload speed menu doesn't exist in the Windows version.

Can you quickly explain what you mean by "Arduino IDE compiled from this
repo"? I'm not super-techie, so I understand "Arduino IDE" (it's what I
write sketches with) but the rest is a mystery to me - though I have seen
the phrase used elsewhere as I've been searching everything I can find that
might hold a clue to the problem...

Thanks

einsteinx2, 2015-04-25:
@duncan-a There's the official Arduino IDE distributed by the guys at http://arduino.cc. This GitHub repository is for the modified version of the Arduino IDE made to work with the ESP8266 wifi chip. I meant that I'm using this modified version which I compiled myself, so it should have all of the latest fixes and modifications made by @igrr and company.

At least in this modified version of the IDE, there is a submenu under the "Tools" menu called "Upload Speed", which has options: 115200, 9600, 57600, 256000, and 921600. I can successfully compile and run code on my ESP8266 using the 115200 speed (which is the default), but not the higher speeds.

duncan-a, 2015-04-26:
Thanks Ben

By "compiled myself", do you mean 'downloaded the latest .zip from
github.com/esp8266/Arduino/ (currently it says it's latest commit eb2bbfa8f6
<https://github.com/esp8266/Arduino/commit/eb2bbfa8f68555c3b00c2cffd45e78360284bc2e>
), unzip it and overwrite the existing install'?

If so, how?

In my case, the existing install is '1.6.1-p1' from the .zip file of the
/Releases page - the Windows version.

Toshik, 2015-04-26:
@duncan-a 
Here you can get latest precompiled for windows: https://ci.appveyor.com/api/buildjobs/8s54dteakqvt3e3q/artifacts/build/windows/arduino-1.6.1-windows.zip

einsteinx2, 2015-04-26:
@duncan-a By compiled myself, I mean I cloned this repository to my machine (so I always have the latest updates) and then used the "ant" tool to compile it (ant; ant build; ant run). I'm on OS X, but it's basically the same on Windows. It sounds like you're new to git/programming, so probably best to use the bre-build binaries that @Toshik linked to.

einsteinx2, 2015-04-26:
@igrr I just tried using the latest esptool.py from https://github.com/themadinventor/esptool and was able to successfully flash at 921600, so this seems to be an issue with the esptool included in this repo.

einsteinx2, 2015-04-26:
Right now I'm having trouble installing at all from the Arduino IDE, but I can use the bin files created by it with esptool.py successfully even at 921600 speed. Would it be possible to replace the bundled esptool binary with esptool.py?

hallard, 2015-05-02:
Just tried with latest repo compiled from scratch, I'm now able to upload also at 921600 on Windows 8.1 X64 with esptool.exe

igrr, 2015-05-04:
@einsteinx2 I use an FTDI as well. I also can't upload at 921600, neither with esptool.py or esptool-ck. But i do have quite long wires from the FTDI to the ESP.

igrr, 2015-05-04:
By the way, esptool was updated to 0.4.3 — I fixed one of the causes of "wrong direction/command" error.

duncan-a, 2015-05-04:
I downloaded and un-zipped the file suggested by @Toshik but the .exe file
won't run.

It flashes up a *very* short-lived error message that, I think, says that
it's not a valid Win-32 file...........

With a laptop that's more than ten years old, running XP, I don't have a
lot of options at this point The 'Release' that I was running was fine...

einsteinx2, 2015-05-26:
Just pulled the latest and built and now I can use 921600 transfers! It usually fails the first time, but then I hit run again and it works. I used to have that same issue when flashing using 115200 before, though now I can flash at 115200 or 230400 without hitting run twice. So things have definitely improved!

Closing this issue as I'm now able to use faster speeds. Thanks!

einsteinx2, 2015-05-26:
Actually spoke a bit too soon. Sometimes it works the first time. Sometimes the second. Or sometimes I try over and over and get this error:

Uploading 34944 bytes from /var/folders/qz/6zrpphk11_ddxstvt5tk21740000gn/T/build1153384146308317912.tmp/ESP_i2c_OLED_Test2.cpp_00000.bin to flash at 0x00000000
...................................
Uploading 212952 bytes from /var/folders/qz/6zrpphk11_ddxstvt5tk21740000gn/T/build1153384146308317912.tmp/ESP_i2c_OLED_Test2.cpp_10000.bin to flash at 0x00010000
...................................................warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04

pfalcon, 2015-05-30:
Did someone measured actual upload speed? esp8266 has auto baud detection, and if it is mis-detected, you can't run it at a decent speed at all, regardless of what you say (USB) port to use. I regularly get uploads of 9600 real speed.

einsteinx2, 2015-05-31:
I haven't measured per se, but it definitely seems to be using the correct speed. It's significantly faster when uploading at 921600 vs 115200 or 230400. And 230400 does seem to be about twice as fast as 115200.

einsteinx2, 2015-05-31:
Regarding my issues with 921600 speed, it appears to be related to the reset mechanism. I'm using an FTDI cable which automatically resets the ESP module. I noticed that on the first run, it works every time at 921600 speed. However subsequent runs fail more often than not. However, if I unplug the ESP and plug it back in before running, it works every time. So likely it's not resetting to 921600 speed after running the program.

pfalcon, 2015-05-31:
Thanks. My culprit was CH341 USB-UART adapter under Linux, more details are at https://github.com/themadinventor/esptool/issues/44

jrobeson, 2016-10-30:
is there anything actionable in this ticket left?

einsteinx2, 2016-10-31:
I'm no longer working with ESP chips at this time, so unless other's are experiencing this issue, feel free to close it.

joeybab3, 2016-12-27:
I've found the upload speeds past 256000 to be unreliable unless I'm specifically working with a wemosd1mini or nodemcu, Even with 256000 I occasionally have to revert to 115200 for a temp fix.

devyte, 2017-10-20:
The previous comment is correct. There are a lot of chinese clones which don't work well at high speeds.
Also, the specific speeds depend on the USB-TTL chip (FTDI, CH340, PL2302, etc), the driver, and the OS.
Closing, as nothing further to do in this repo.

closed 2017-10-20 by devyte