esp8266/Arduino #101: The Serial.setDebugOutput does not disable wifi messages

av1024, 2015-04-24:
Looks like some pre-initialized variables issue on HardwareSerial

(Using git version):
```
Serial.begin(115200);
Serial.setDebugOutput(false);
```
// still send debug from wifi

SOULTION:
```
Serial.begin(115200);
Serial.setDebugOutput(true);
Serial.setDebugOutput(false);
```

igrr, 2015-04-24:
This should have been fixed in db86766, discussed in #98. Are you using the latest version from git?

av1024, 2015-04-24:
Wow! I missed this fix because look only for "open" issues...
I use downloaded ZIP with 20150419 05:23 timestamps (1dd897 ?) monkey-patched for str_tok.

upd: looks like latest (just now) downloaded zip give me #88 issue :(

closed 2015-04-24 by av1024