esp8266/Arduino #104: Error when compiling any of the examples

marcoschwartz, 2015-04-24:
Hello, since the last commit on the repo (ad74beec4d76f95fdfd58eafee2b6a9ca72e0756) I have the following error with all my sketches using the ESP8266:

/Users/marco/Downloads/Arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /var/folders/cy/_hsnm_s10gz1lkx2yyrlmg4w0000gn/T/build6868186071784743931.tmp/HelloServer.cpp.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Error compiling.

Anybody else has that?

av1024, 2015-04-24:
Confirm.
Falling back to pre- strtok_r version with a bit monkey-patching

natetrue, 2015-04-24:
I'm getting the error too with that commit.  Looks like the linker is trying to put too much stuff into iram and not irom - which is odd, the linker script has not changed.

natetrue, 2015-04-24:
I tried moving libm.a to the irom section and it appears to work:

diff --git a/hardware/tools/esp8266/sdk/ld/eagle.app.v6.ld b/hardware/tools/esp8266/sdk/ld/eagle.app.v6.ld
index 4bafb69..a72e7d0 100644
--- a/hardware/tools/esp8266/sdk/ld/eagle.app.v6.ld
+++ b/hardware/tools/esp8266/sdk/ld/eagle.app.v6.ld
@@ -153,6 +153,7 @@ SECTIONS
     _irom0_text_start = ABSOLUTE(.);
     *core_esp8266_*.o(.literal*, .text*)
     *.cpp.o(.literal*, .text*)
+    *libm.a:(.literal .text .literal.* .text.*)
     *(.irom0.literal .irom.literal .irom.text.literal .irom0.text .irom.text)
     _irom0_text_end = ABSOLUTE(.);
   } >irom0_0_seg :irom0_0_phdr

libm seems to have a lot of the mathy functions (log, sin, etc), though, so it may not be the best choice to move out of iram.

Links2004, 2015-04-24:
good idea have changed it

thewiep, 2015-04-24:
I had the same issue with ad74bee and adding the line from abdbdb1 fixes it.
Thanks!

igrr, 2015-04-26:
fixed in abdbdb1

closed 2015-04-26 by igrr