esp8266/Arduino #105: Can't create a UDP Socket and Multicast Socket

Makuna, 2015-04-24:
I need to have two listening sockets (Contexts in Esp8266/Arduiino speak) for my CoapServer. One for the Multicast and one for a unicast.  They both use the same port by CoAp standards.

Which ever one is created first succeeds, but the second one created fails in UdpContext::Listen.  The call to udp_bind within UdpContext::Listen returns error code is ERR_USE; meaning the address is in use?

Coap Port is 5684
Coap Multicast Address is 224.0.1.187

By normal socket concepts this should be doable, not so sure by Esp8266 context concepts.

Further, if I just create a multicast socket, while it will receive multicast packets, it will not receive packets sent directly to the ipaddress of the device.  This works as I would expect IF I can create the two functional sockets.  

here are snippits of the code, just for completeness, they work independently but when called back to back.

multicast
```C++
    multicastSocket = new WiFiUDP();

        if (!multicastSocket ->beginMulticast(WiFi.localIP(),
            IPAddress(CoapMultiCastAddress) ,
            CoapPort))
        {
            // error
        }
```
unicast
```
    unicastSocket = new WiFiUDP();

    // unicast
    if (!unicastSocket->begin(CoapPort))
    {
        // error
    }
```

igrr, 2015-04-27:
With lwIP network stack you can only bind one PCB to a given port. 
However, joining a multicast group is not a property of PCB, it is a property of an interface. This means that when you create a PCB and bind it to a port, it will receive both multicast and unicast packets targeting this port, provided you have joined the multicast group.

I've added ```WiFiUDP::destinationIP``` method that you may use to distinguish the packets sent to a multicast and unicast destinations.

Here's my test sketch:
```
void loop() {
  int cb = Udp.parsePacket();
  if (!cb)
    return;
    
  Serial.print("got udp packet from ");
  Serial.print(Udp.remoteIP());
  Serial.print(" to ");
  Serial.println(Udp.destinationIP());
  delay(1000);
}
```
I send some packets using nc:
```
$ date | nc -u -n -w 1 224.1.1.1 5322
$ date | nc -u -n -w 1 192.168.1.26 5322
```

And here's the output:
```
got udp packet from 192.168.1.23 to 224.1.1.1
got udp packet from 192.168.1.23 to 192.168.1.26
```

Makuna, 2015-04-27:
But I was not receiving a unicast packet to the "context" that was created with the Multicast Join.  Could you include your setup for that here so I can confirm I wasn't doing something wrong.

Makuna, 2015-04-30:
Confirmed I am receiving both types with correct destination addresses.

closed 2015-04-30 by Makuna