esp8266/Arduino #112: Wdt reset

plinioseniore, 2015-04-25:
Hi,

I'm porting an AVR based framework on ESP8266, actually I'm getting the watchdog reset while running my application.

Just to be sure, I've used this code and even this fire the watchdog.

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
Serial.println("ready");
}
void loop() {
  // put your main code here, to run repeatedly
while(1);
} 

So, how can I reset the watchdog timer before it reset the microcontroller? In the examples that are coming with the IDE there is no explicit call to a watchdog reset, and I guess that is done in the background while not working in the loop.

Thanks,
Dario.

Links2004, 2015-04-25:
you can do it like in the AVR
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/Esp.h#L43

but better use ```delay(0);``` the delay allow the wifi and tcp functions to run,
if you block them to long in you application,
you lost your connections.

some note:
on the AVR arduino the watchdog is disabled by default.
on the ESP8266 it is enabled.

I wood recommend to keep it enabled and use ```delay(0);``` in your code.

plinioseniore, 2015-04-25:
Hi,

thanks for the fast answer. Just got that the pre-compiled IDE has older
cores, I will update and have a try.

Thanks,
Dario.

igrr, 2015-04-25:
Better yet use yield(); -- it has the same effect as delay(0), but it is
also semantically compatible with other Arduino boards that have support
for Scheduler library.

plinioseniore, 2015-04-25:
Thanks. 

I will use the latest cores with the prebuild IDE, and extend the wdt timer. What is surprising is that the code looks faster on an AVR than on the ESP, probably all the 8bit data manipulating aren't native for this micro.

closed 2015-04-25 by plinioseniore