esp8266/Arduino #118: DHT11 readings are wildly inaccurate

nsgn, 2015-04-26:
I've got a DHT11 sensor that reads correctly on an arduino nano but when used with the esp8266 arduino code it gives wildly inaccurate, rapidly changing values. Such as 260000 instead of 78*F. Upon each receive of data from the DHT11 the values change drastically, by thousands. ESP model is ESP-01.

I've attempted on two different GPIO pins. When using GPIO 2 the reading is inaccurate for both temp and humidity. When using GPIO 3 (stealing from serial) the humidity is correct but the temperature is no different - still nothing close to a temperature.

Forgive any lack of necessary detail here. It's my first issue report on GitHub. I will gladly recreate and further test the issue or provide any additional detail that might help with resolution. Just let me know what will be most helpful. I'm also entirely open to the option that I might be doing something stupid and take no offense to being shown where I'm in error.

Thanks for the great project. Besides this odd issue with DHT sensors it works flawlessly. Uploads go perfectly every time from this Windows pc and other sketches I've loaded up just as I would expect in every way. Fantastic work.

igrr, 2015-04-26:
Do you have a pull up resistor on GPIO2? What code do you use to initialize DHT library?
Are you providing the third argument as mentioned in the Readme?
DHT dht(DHTPIN, DHTTYPE, 15);
Are you running the sketch at default 80MHz or higher at 160MHz?

duncan-a, 2015-04-26:
Just to jump in...

If I ever get the upload problems sorted, I'm intending to deploy some
DHT22 sensors with ESP-01s - how do you set the sketch speed? That's not
something that I've ever seen with 'straight Arduino'.

What's best for the DHT sensors?

Thanks

igrr, 2015-04-26:
There's a menu that allows you to set either 80MHz or 160MHz. That said, I haven't tried DHT library at 160MHz. The third parameter (15 in ```DHT dht(DHTPIN, DHTTYPE, 15);```) likely needs to be tuned.
[image]

duncan-a, 2015-04-26:
Ah, it seems this is another 'Mac Only' option - though I fail to see why,
if upload speed and CPU speed are useful for the modules, they aren't
implemented in the Windows version too...

Thanks for solving that part of the mystery that is ESP8266/Arduino!

I've yet to understand that third parameter in the DHT implementation (I
didn't realise it was a variable, as such). I have no idea if '15' will
work for me until I can get the upload problem sorted out...

Toshik, 2015-04-26:
@duncan-a Nope, it is latest build option :) Other issue #97 I answered you, where to get it.

igrr, 2015-04-26:
This isn't a Mac-only option. It was added about three weeks ago in 3bde54d, so if you are using the release from the "releases" page, you won't yet see it.

Regarding the third option for the DHT library... Well, DHT line of sensors use a one-wire protocol which encodes "0" and "1" using different pulse lengths. The library we are talking about, https://github.com/adafruit/DHT-sensor-library, doesn't have any fancy auto-calibration built in. You have to tell it upfront how many cycles to wait until the pulse can be considered a "1". Here's a snippet from the sample sketch included with the DHT library:
```
// NOTE: For working with a faster chip, like an Arduino Due or Teensy, you
// might need to increase the threshold for cycle counts considered a 1 or 0.
// You can do this by passing a 3rd parameter for this threshold.  It's a bit
// of fiddling to find the right value, but in general the faster the CPU the
// higher the value.  The default for a 16mhz AVR is a value of 6.  For an
// Arduino Due that runs at 84mhz a value of 30 works.
// Example to initialize DHT sensor for Arduino Due:
//DHT dht(DHTPIN, DHTTYPE, 30);
```
The value of 15 seems to work for the ESP at 80 MHz.

duncan-a, 2015-04-26:
OK, thanks for your help - I'll try and make sense of all this and ensure
I'm running the latest build option...

I really am out of my comfort zone with this now. I plugged them in,
uploaded and they worked - great!!! Then they stopped working, though
uploading appeared to go smoothly (if, as before, *incredibly* slowly)...

I can't even reinstall the original AT command firmware and hang them of
Arduino Pro Minis - all very frustrating...

hallard, 2015-04-26:
@duncan-a ,
you're right CPU Frequency is not available on windows, the work around is to add this definition to your boards.txt

```
##############################################################
esp01_160.name=Generic ESP8266 board (160 MHz)
esp01_160.upload.tool=esptool
esp01_160.upload.speed=115200
esp01_160.upload.resetmethod=none
esp01_160.upload.maximum_size=524288
esp01_160.upload.wait_for_upload_port=true
esp01_160.build.mcu=esp8266
esp01_160.build.f_cpu=160000000L
esp01_160.build.board=ESP8266_ESP01
esp01_160.build.core=esp8266
esp01_160.build.variant=esp01
```
That should do the job for 160MHz;-)

igrr, 2015-05-04:
@nsgn Any update? Did you get a chance to try this again following the suggestions?

closed 2015-05-14 by igrr

rdibened, 2016-10-17:
You may have a bad DHT22 sensor.  Let me tell you my story.  I purchased two DHT22 sensors.  One was connect to my Raspberry Pi-2 and the other to an Arduino Uno.  The one on the Raspberry Pi reads fine but the one on the Ardunio reads 19% Humidity while the one on the Raspberry Pi and another Hydrometer read 48% and 51% respectively.  I switched the DHT22 sensors (put the Raspberry Pi sensor on the Arduino and visa versa). Now the one on the Ardunio reads correctly and the on on the Raspberry Pi reads 19%.  Proving I have a bad sensor.  This may be your problem.   If you have a second DHT22 you might want to see if switching them fixes the problem.  I have read a lot about these sensors and I think they may have a high failure rate at manufacture.  I just purchase four more and will test them all when they arrive for accuracy if off they are going back.