esp8266/Arduino #119: Wifi.status

kdschlosser, 2015-04-26:
not sure if this is a problem per say but the use of

WiFi.status() != WL_CONNECTED

is not a check to see if you are connected to the Wireless AP

it looks to see if it has an IP address

they are 2 completely different things.

maybe i am just misinterpreting the WL_CONNECTED but i would imagine that means Wireless LAN Connection

I have a server that is DHCP that server can go down and and not affect the AP connection. granted it can't talk to the Network. but I am trying to make a script that if the connection actually goes down it will reconnect. and if the server goes down it will just keep on broadcasting DHCP requests until it gets an IP

not actually have to reset the connection.

script is for redundancy. it's supposed to fire up other backup equipment if there is a problem. and then send me an e-mail and text message letting me know something's broke and what it is that's broken. I don't know if this is incorrect coding on my part or it's the coding in the IDE and how it deals with checking a connection. but an IP address is not a wifi connection.

if there is an alternate way someone please let me know. I have gone digging through the wificlient library and i am not seeming to be able to find what i am looking for.

I don't want this firing up equipment that doesn't need to be and not telling me what is exactly wrong.

thanks again for the assistance. and i do appoligize if this is my dumbass not reading the code correctly. lol

ajitam, 2015-06-04:
hi kdschlosser I'm interested in the same thing. Any progress?

Links2004, 2015-06-04:
```WL_CONNECTED``` is only returned if the wifi interface has connected an has IP address
in DHCP mode this means you got the IP form the DHCP server.
in Static mode you have set the IP yourself.

github has a nice search:
https://github.com/esp8266/Arduino/blob/7dc997b4a451e7ef0823be07ca379c31f805639a/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L460

kdschlosser, 2015-06-05:
that didn't exactly answer my question or statement. the problem is the ESP reboots because of not having a "connection" which is determined by the IP and an IP address is not a wifi connection, so there is no cause for the module to keep on boot cycling, there are other functions that the ESP is able to perform without actually having an IP Address. is there a way to have the ESP opperate without an IP address?

Links2004, 2015-06-05:
you can move the WiFi waiting to the loop like this:
```
void setup() {
    Serial1.begin(921600);
    delay(10);

    // We start by connecting to a WiFi network
    Serial1.println();
    Serial1.println();
    Serial1.print("Connecting to ");
    Serial1.println(ssid);

    WiFi.begin(ssid, password);
}

void loop() {
    if(WiFi.status() != WL_CONNECTED) {
        Serial1.println("[loop] no wifi");
        delay(500);
    } else {
        Serial1.println("[loop] WiFi connected");
        Serial1.print("[loop] IP address: ");
        Serial1.println(WiFi.localIP());
        delay(500);
    }
}
```

```
Connecting to Testwifi1
[loop] no wifi 
[loop] no wifi 
[loop] no wifi 
[loop] no wifi 
[loop] no wifi 
[loop] no wifi 
[loop] no wifi 
[loop] WiFi connected 
[loop] IP address:  11.11.1.2 
[loop] WiFi connected 
[loop] IP address:  11.11.1.2 
```

but mixed mode (dhcp / static) is not possible.
the SDK from espressif only give the status of the Wifi like this:
```
enum {
    STATION_IDLE = 0,
    STATION_CONNECTING,
    STATION_WRONG_PASSWORD,
    STATION_NO_AP_FOUND,
    STATION_CONNECT_FAIL,
    STATION_GOT_IP
};
```
so you have no change to see if you are connected but not get an answer from the DHCP.
you can only try to setting it static and then call ```wifi_station_dhcpc_start();```

kdschlosser, 2015-06-05:
it restarts, reboots like a computer, any time that W_CONNECTED returns false in any kind of a loop with it, sees it as an infinite loop so the ESP restarts i think. have put it in it's own function outside of loop and setup i have had it loop inside setup, have had it loop inside loop and use loop for the loop, it always resets the ESP. no matter what i do. it would have been nice to use it to jump to a different wifi netowrk (don't even know if that is possible) but that was the hope for it.

Links2004, 2015-06-05:
if you need WiFi connection management for multiple APs you can use WiFiMulti
https://github.com/esp8266/Arduino/blob/esp8266-sdk-1.0/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/examples/WiFiMulti/WiFiMulti.ino

a "loop" in setup is possible you only need to add a delay(0); 
in there or the WDT will reset your ESP.

```serial.setDebugOutput(true);```
gives you more info about reboots/crash.

tecan, 2015-06-14:
the access point example in the arduino IDE should be fixed / updated. when connected all it does is spit out periods over console like nothings connected even when it is connected.

wizard23, 2015-06-16:
I have the same problem as @tecan. The AP example only spits out dots....
The AP exists and I can connect to it with my laptop. But WiFi.status() never gets set to WL_CONNECTED. It has the value 6 (WL_DISCONNECTED) for ever.

WiFi.localIP() returns 0.0.0.0 so I cant start a webserver.

far5893, 2015-07-06:
I'm in the same situation of @wizard23    some help ?

igrr, 2015-07-06:
@far5893 @wizard23 @tecan 
Please remove
```C++
while (WiFi.status() != WL_CONNECTED) {
  delay(500);
  Serial.print(".");
}
```
from your sketch when running in SoftAP mode.

avarachanvc, 2015-11-11:
Is the softAP resetted after every attempt to connect to a router?

I just happened to find a particular behavior on arduino core. 
I start in softAP mode. Then I submit the ssid and password through a web form, via ajax (important because the ajax connection is is also key to my finding).

My code

<code>

	WiFi.begin(ssid,Wrong_password);
	unsigned long startMillis = millis();
	unsigned long currentMillis = millis();
	while (WiFi.status() != WL_CONNECTED) {
	  delay(500);// refer->A
	  currentMillis = millis();
	  if ((currentMillis - startMillis) > 20000) {
		//return error code for not connecting even after 20 seconds.
		break;            
	  }
	}

	//Handle Successfull connection.
	if (WiFi.status() == WL_CONNECTED) {
		// Send valid ip reponse to AJAX call. 
	}
	else {
	  // Send wifi.status() code as response to ajax.
	  //WiFi.disconnect();
	}
</code>
If I submit the wrong ssid and pw, around 6 seconds, the ajax connection is aborted. Chrome shows <code>net::ERR_CONNECTION_RESET.</code>

Is there an internal function that resets/ restarts the softAP while waiting to connect to router? The softAP seems to reestablish the AP. I tried changing the 500ms (refer->A) to 2 seconds to see if the number of times attempting to check

Links2004, 2015-11-11:
the ESP need to switch to all channels for scanning and if its try to connect to some AP that Wifi channel of the softAP is changed.
the limitation comes from the Hardware, there is only one RF fronted. 

try to use a WiFi scanner app or tool and look for your soft AP, then you shut see the channel changes.
on every change you will lose the connection to the softAP.

rogerclarkmelbourne, 2015-11-17:
@Links2004 

I don't think your explanation of this is correct.

I'm trying to do the same thing as @avarachanvc is doing, in fact my code looks very similar.

Anyway. I ran "Acrylic Wifi" which is a wifi scanner for Windows and the ESP8266 always seems to be on Channel 1, where as my target network is on channel 9.

Anyway, I can get an ajax response from the ESP8266 back to my PC (which is connected to the ESP8266 as an AP).

This means that the ESP8266 is operating on both channel 1 and also channel 9.

I think the issue is that @avarachanvc is calling things like wifi.disconnect() - although they are commented out in the code that is posted.

I think calling WiFi.disconnect() or  WiFi.mode(WIFI_STA); at any time near where you send back the response to the ajax, is seriously messing things up.

I suspect the only way to be able to use WiFi.disconnect() is after a time delay, but this would need to be done on a timer callback I think rather than a delay, as delays are not good in this sort of system, as most things are asynchronous.

Although I wrote my code the same way, I also suspect that doing this

```
    while (WiFi.status() != WL_CONNECTED && millis()<timeOut) 
    {
      delay(500);
      DBG_OUTPUT_PORT.print(".");
    }
```

is not a good idea either, I've noticed that Windows sporadically sends requests for a file named ncsi.txt - which from what I've read is a file that sits on a windows server and Microsoft use as a flag to indicate if you are really connected to the internet or not.
Annoyingly windows keeps requesting this file quite often, including when the ESP8266 is trying to connect to another AP.

I've not quite worked out what's going on (and I still may need to use WiFi.disconnect(), but I find if I deliberately send a wrong password. Then even if I resent the ajax request a while later, with the correct password, that the ESP8266 fails to connect to the AP.

 
Anyway, I think there is more to this than something to do with the RF section of the chip

Links2004, 2015-11-17:
the softAP is clearly moved to the channel of the AP.

some test code for the channel switching of the ESP:

```cpp
    // clear all configs
    WiFi.disconnect();
    WiFi.softAPdisconnect();

    // open soft AP
    WiFi.softAP("softAP","123456789000", 1, false);

    // wait some time for making screenshot
      for(uint8_t t = 20; t > 0; t--) {
          Serial1.printf("[SETUP] WAIT %d...\n", t);
          Serial1.flush();
          delay(1000);
      }
   // connect to AP
      WiFi.begin("133744445678785154IoT", "pass here");
```

before ESP bower on:
[image]

ESP has opened softAP on channel 1
[image]

ESP connect to AP on channel 9 (soft AP is moved there too)
[image]

of course the web server will not response when there is a  ```while (WiFi.status() != WL_CONNECTED)```.

but the switch of the channel, will lead too an disconnect of the WiFi.
normally the Computer scans again and auto connect to softAP at channel 9.
but the disconnect will end in a close of all tcp connections.

rogerclarkmelbourne, 2015-11-17:
@Links2004

OK. 
I think the Wndows 7 TCP stack must somehow be able to cope with the wifi switching channels, as occasionally, I get a perfectly valid response from the ajax request.
However this only happens on random occasions.

As this is random, I don't know if other OSs do the same.

Also, I do occasionally get disconnection errors from the ajax request.

I wonder if there is a workaround for this at all.
e.g on the ajaz request to connect to the AP, change the config of the Soft AP to change channel to the channel of the target AP.

Wait a few secs for the client OS to notice the Soft AP has moved

Then send an ajax request from the client to tell the ESP to connect to the AP ( which will now be on the same channel as the Soft AP)

However, It looks like its highly dependent on which OS is connecting to the AP.

Links2004, 2015-11-17:
there is a command ```wifi_set_channel``` in the SDK.

[image]

may it work to move the AP before the connect.
but you need the can before to know the right one, hope this will not disconnect the client too.

here is an example how to scan for APs in a async way:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L41

rogerclarkmelbourne, 2015-11-17:
Thanks

Actually, for some reason, scanning doesn't seem to drop the soft AP.

I'm using an ajax request to do the scan, just using 

WiFi.scanNetworks();

I seem to have a very high success rate with this, and don't seem to have any problems on Windows or iOS with Wifi disconnection of an open ajax request.

Re: wifi_station_disconnect

As I have debug turned on, I noticed that if that the SDK kept trying to reconnect if the password etc was incorrect. 

So I had already tried calling Wifi.disconnect() which basically calls, wifi_station_disconnect(), however this didnt seem to help.

I think the main problem is when the client OS connecting to the softAP decides to rescan.

Also part of the problem could be the way that we are detecting connection. i.e it looks like the presumption with WiFi.begin() is that the correct username and password will always be supplied.

I think I will investigate using  wifi_station_set_reconnect_policy(false) to see if this improves things

Humancell, 2016-01-04:
I'm working on some code to detect different Wifi failures, and am trying to understand the best way to do this.  From reading this thread, it seems that WiFi.status() == WL_CONNECTED will tell me "if I have an IP Address" ... but will that change if - for example - I turn off the AP?  Or if I move out of range of the AP?

I'll do more testing myself, but am very curious what the "recommended" way is to handle Wifi failures and re-connection attempts?

Thanks!

Links2004, 2016-01-04:
the latest version has a event callback for wifi changes.
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino#L12
possible events:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiType.h#L47-L57

Humancell, 2016-01-04:
Nice!  Thank you for the references!

Just as an update of my own testing ...

1. If I turn off my AP after the ESP is connected, and I poll the WiFi.status(), it returns a WL_DISCONNECTED (6) most of the time, and WL_NO_SSID_AVAIL (1) sometimes ... I was polling at 500ms.  It just bounces between these two.

2. If I turn the AP back on, the ESP auto-reconnects!  I didn't realize it would do this without code ... nice feature!

Now I'll add these events into my code ... thanks again!

Humancell, 2016-01-04:
@Links2004 is this new event callback in "stable" yet?

Links2004, 2016-01-05:
no only straring

cyberreefguru, 2016-03-05:
I think I have the opposite problem of everyone else.  I have not narrowed it down yet, but when the board is powered up, I'm getting a connection status of WL_CONNECTED before I configure the node.  Because of that, the software attempts to disconnect first and then segfaults.  Some how wifi_station_get_connect_status is returning STATION_GOT_IP before I call begin.  Any suggestions?

maxdd, 2017-04-11:
I leave these here since they are really similar problems

https://github.com/esp8266/Arduino/issues/3132
https://github.com/esp8266/Arduino/issues/3128

devyte, 2017-10-20:
No complete sketch, basic info.
I think the original post should be addressed by the current event system plus a bit of support code.
Closing due to age and lack of info.
If you think there is still a problem, please open a new issue, and fill in the requested info.

closed 2017-10-20 by devyte

Anatoli113, 2017-11-29:
Hi,
I want to know that how to detect that station go out of esp8266 softAP range.
I am going to make simple sketch code.
This code is to confirm that the mobile phone certified to esp8266 softAP is outside the wifi range.
In order to do, I used WiFi.softAPgetStationNum().
But once mobile certified to softAP, even though the mobile goes out of wifi range, this function always returns 1.
The esp82266 always has to be softAP.

I hope somebody to help me. Nothing is a correct way to confirm?
Thanks in advance.

devyte, 2017-11-29:
@Anatoli113 like I said in the previous comment, look into the wifi event system.

alimorawej, 2018-03-24:
Hi,
I have problem with WiFi.mode(WiFi_AP_STA), i get in WiFi.localIP() returns 0.0.0.0 so I cant start a webserver, but WiFi.softAPIP() are okay.

Connecting to SmartHome2
WiFi connected
 IP address: 0.0.0.0
Mode: STA+AP
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (10): SmartHome2
Passphrase (10): 1234567890
BSSID set: 0

Anyone can help?

barsznica, 2018-09-15:
https://github.com/esp8266/Arduino/issues/2593#issuecomment-323801447

> 

A workaround is to replacewhile ( status != WL_CONNECTED) {withwhile (WiFi.localIP().toString() == "0.0.0.0") {
--

This worked for me!

riotgibbon, 2021-01-28:
> [#2593 (comment)](https://github.com/esp8266/Arduino/issues/2593#issuecomment-323801447)
> 
> P.S. Well it didn't last overnight, so I'll try the bottom approach.
> 
> > A workaround is to replace `while ( status != WL_CONNECTED) {` with `while (WiFi.localIP().toString() == "0.0.0.0") {`
> 
> P.S. I, also, found that in the BasicOTA example, it uses `while (WiFi.waitForConnectResult() != WL_CONNECTED)`.
> This works too (for now).

thank you, this worked for me as well