esp8266/Arduino #120: ESP-07 or ESP-12 ili9341 2.4" tft Color problem

tontito, 2015-04-26:
http://www.esp8266.com/viewtopic.php?f=32&t=2625&start=10

background not black allways white colors incorrect 

RED = CYAN 
CYAN = RED
YELLOW = BLUE

All colors incorrect and background allways white.

igrr, 2015-04-27:
@Sermus any idea what could be wrong here?

tontito, 2015-04-27:
i don`t know.

maybe 2.2" ili9341 different 2.4" ili9341.

tontito, 2015-05-02:
hi again.

02.05.2015 windows build
(https://ci.appveyor.com/api/buildjobs/dbj2egv85gam6s8t/artifacts/build/windows/arduino-1.6.1-windows.zip)
download and copile ili9341 example no error compiling but not working screen always white screen.

stonemull, 2015-06-19:
I had the same with an ILI9341 2.4 TFT (different library though so your code may be different)  and changed the driver to initialise with inversion turned on.

//    writeRegister8(ILI9341_INVERTOFF, 0);		// original
    writeRegister8(ILI9341_INVERTON, 0);	//set by col

addresses are ..

#define ILI9341_INVERTOFF          0x21 //
#define ILI9341_INVERTON           0x20 //

if you want to check the ILI9341 datasheet

tontito, 2015-07-12:
New screen receive. No problem good color

closed 2015-07-12 by tontito