esp8266/Arduino #121: Need help with ESP8266WebServer.h

dogrocker, 2015-04-26:
Hello, I'm working on ESP8266WebServer.h
try to save config data to EEPROM and load when ESP is restart

The idea by esp-arduino-apboot
on Chriscook8 http://www.esp8266.com/viewtopic.php?f=29&t=2520

I modified to use ESP8266WebServer.h for handle RestAPI easily 

But now it have some error I couldn't fix it (Don't know what I'm wrong with ESP8266WebServer.h).

Need some one to help to fix this.

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>
#include <EEPROM.h>

MDNSResponder mdns;
ESP8266WebServer server(80);

const char* ssid = "BUBBLES";
const char* passphrase = "BUBBLES";
String st;
String content;

void setup() {
  Serial.begin(115200);
  EEPROM.begin(512);
  delay(10);
  Serial.println();
  Serial.println();
  Serial.println("Startup");
  // read eeprom for ssid and pass
  Serial.println("Reading EEPROM ssid");
  String esid;
  for (int i = 0; i < 32; ++i)
    {
      esid += char(EEPROM.read(i));
    }
  Serial.print("SSID: ");
  Serial.println(esid);
  Serial.println("Reading EEPROM pass");
  String epass = "";
  for (int i = 32; i < 96; ++i)
    {
      epass += char(EEPROM.read(i));
    }
  Serial.print("PASS: ");
  Serial.println(epass);  
  if ( esid.length() > 1 ) {
      // test esid 
      WiFi.begin(esid.c_str(), epass.c_str());
      if (testWifi()) { 
          launchWeb(0);
          return;
      }
  }
  setupAP(); 
}

bool testWifi(void) {
  int c = 0;
  Serial.println("Waiting for Wifi to connect");  
  while ( c < 20 ) {
    if (WiFi.status() == WL_CONNECTED) { return true; } 
    delay(500);
    Serial.print(WiFi.status());    
    c++;
  }
  Serial.println("");
  Serial.println("Connect timed out, opening AP");
  return false;
} 

void launchWeb(int webtype) {
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("Local IP: ");
  Serial.println(WiFi.localIP());
  Serial.print("SoftAP IP: ");
  Serial.println(WiFi.softAPIP());
  if (!mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("Error setting up MDNS responder!");
    while(1) { 
      delay(1000);
    }
  }
  Serial.println("mDNS responder started"); 
  createWebServer(webtype);
  // Start the server
  server.begin();
  Serial.println("Server started"); 
}

void setupAP(void) {
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
     {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
     }
  }
  Serial.println(""); 
  st = "<ol>";
  for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      st += "<li>";
      st += WiFi.SSID(i);
      st += " (";
      st += WiFi.RSSI(i);
      st += ")";
      st += (WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*";
      st += "</li>";
    }
  st += "</ol>";
  delay(100);
  WiFi.softAP(ssid);
  Serial.println("softap");
  launchWeb(1);
  Serial.println("over");
}

void createWebServer(int webtype)
{
  // Check for any mDNS queries and send responses
  mdns.update();
  
  if ( webtype == 1 ) {
    server.on("/", []() {
        IPAddress ip = WiFi.softAPIP();
        String ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
        content = "<!DOCTYPE HTML>\r\n<html>Hello from ESP8266 at ";
        content += ipStr;
        content += "<p>";
        content += st;
        content += "</p><form method='get' action='setting'><label>SSID: </label><input name='ssid' length=32><input name='pass' length=64><input type='submit'></form>";
        content += "</html>";
        server.send(200, "text/html", content);  
    });
    server.on("/setting", []() {
        String qsid = server.arg("ssid");
        String qpass = server.arg("pass");
        if (qsid.length() > 0 && qpass.length() > 0) {
          Serial.println("clearing eeprom");
          for (int i = 0; i < 96; ++i) { EEPROM.write(i, 0); }
          Serial.println(qsid);
          Serial.println("");
          Serial.println(qpass);
          Serial.println("");
            
          Serial.println("writing eeprom ssid:");
          for (int i = 0; i < qsid.length(); ++i)
            {
              EEPROM.write(i, qsid[i]);
              Serial.print("Wrote: ");
              Serial.println(qsid[i]); 
            }
          Serial.println("writing eeprom pass:"); 
          for (int i = 0; i < qpass.length(); ++i)
            {
              EEPROM.write(32+i, qpass[i]);
              Serial.print("Wrote: ");
              Serial.println(qpass[i]); 
            }    
          EEPROM.commit();
          content = "<!DOCTYPE HTML>\r\n<html>";
          content += "<p>saved to eeprom... reset to boot into new wifi</p></html>";
        } else {
          content = "Error";
          Serial.println("Sending 404");
        }
        server.send(200, "text/html", content);
    });
  } else {
    server.on("/", []() {
      server.send(200, "text/plain", "this works as well");
    });
    server.on("/setting", []() {
      server.send(200, "text/plain", "setting.");
    });
    server.on("/cleareeprom", []() {
      content = "<!DOCTYPE HTML>\r\n<html>";
      content += "<p>Clearing the EEPROM</p></html>";
      server.send(200, "text/html", content);
      Serial.println("clearing eeprom");
      for (int i = 0; i < 96; ++i) { EEPROM.write(i, 0); }
      EEPROM.commit();
    });
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  server.handleClient();
}
```

igrr, 2015-04-27:
>But now it have some error I couldn't fix it
So what exactly is the error you are getting?

dogrocker, 2015-04-27:
it no error but the server can't run and sometime it can run but just one request and I got no response.

maybe I'm doing a wrong loop I don't know what the real problem are.

cause old code he use loop to create the server, but with ESP8266WebServer.h the loop is on the main program loop so I create the server two times and don't know how to stop the first one.

igrr, 2015-04-27:
Okay, at least please provide the output you are getting in the serial window.

dogrocker, 2015-05-04:
```
Startup
Reading EEPROM ssid
SSID: !
              empty otp
      
Reading EEPROM pass
PASS: %s %u
          
SDK ver: %s compiled @ %s %s
  phy ver: %d, pp 
Waiting for Wifi to connect
66666666666666661111
Connect timed out, opening AP
scan done
35 networks found
1: .@  TRUEWIFI (-68) 
2: RADIUS-TESTING (-91) 
3: RADIUS-TESTING (-64) 
....
softap

WiFi connected
Local IP: 

mDNS responder started
Server started
```

but I when I connect to my 8266wifi to config at 192.168.1.4 it  can't connect.

biohazardxxx, 2015-05-04:
I have the same issue with this example code:
```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
 
String form = "<form action='led'><input type='radio' name='state' value='1' checked>On<input type='radio' name='state' value='0'>Off<input type='submit' value='Submit'></form>";
 
ESP8266WebServer server(80);

const int led = 13;
 
void handle_led() {
  int state = server.arg("state").toInt();
  
  digitalWrite(led, state);
  server.send(200, "text/plain", String("LED is now ") + ((state)?"on":"off"));
}

void setup(void) {
  Serial.begin(115200);
  Serial.println("");
  pinMode(led, OUTPUT);
  
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  WiFi.softAP("ESP_WiFiSwitch");
    Serial.println(WiFi.softAPIP());
   
   server.on("/", [](){
   server.send(200, "text/html", form);
  });
  
  server.on("/led", handle_led);
  server.begin();
  Serial.println("HTTP server started");
}
 
void loop(void) {
  server.handleClient();
}
```

gerardwr, 2015-05-04:
@biohazardxxx

The wifimode you use is not correct, WIFI_STA is for Station Mode.

Changing WIFI_STA to WIFI_AP and it works (for me).

dogrocker, 2015-05-04:
This is my final code

```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <EEPROM.h>

ESP8266WebServer server(80);

const char* ssid = "BUBBLES";
const char* passphrase = "BUBBLES";
String st;
String content;

void setup() {
  Serial.begin(115200);
  EEPROM.begin(512);
  delay(10);
  Serial.println();
  Serial.println();
  Serial.println("Startup");
  // read eeprom for ssid and pass
  Serial.println("Reading EEPROM ssid");
  String esid;
  for (int i = 0; i < 32; ++i)
    {
      esid += char(EEPROM.read(i));
    }
  Serial.print("SSID: ");
  Serial.println(esid);
  Serial.println("Reading EEPROM pass");
  String epass = "";
  for (int i = 32; i < 96; ++i)
    {
      epass += char(EEPROM.read(i));
    }
  Serial.print("PASS: ");
  Serial.println(epass);  
  if ( esid.length() > 1 ) {
      // test esid 
      //WiFi.mode(WIFI_AP_STA);
      //WiFi.disconnect();
      //delay(100);
      WiFi.begin(esid.c_str(), epass.c_str());
      if (testWifi()) {
        launchWeb(0);
        return;
      }
  }
  setupAP();
}

bool testWifi(void) {
  int c = 0;
  Serial.println("Waiting for Wifi to connect");  
  while ( c < 20 ) {
    if (WiFi.status() == WL_CONNECTED) { return true; } 
    delay(500);
    Serial.print(WiFi.status());    
    c++;
  }
  Serial.println("");
  Serial.println("Connect timed out, opening AP");
  return false;
} 

void launchWeb(int webtype) {
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("Local IP: ");
  Serial.println(WiFi.localIP());
  Serial.print("SoftAP IP: ");
  Serial.println(WiFi.softAPIP());
  createWebServer(webtype);
  // Start the server
  server.begin();
  Serial.println("Server started"); 
}

void setupAP(void) {
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
     {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
     }
  }
  Serial.println(""); 
  st = "<ol>";
  for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      st += "<li>";
      st += WiFi.SSID(i);
      st += " (";
      st += WiFi.RSSI(i);
      st += ")";
      st += (WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*";
      st += "</li>";
    }
  st += "</ol>";
  delay(100);
  WiFi.softAP(ssid, passphrase, 6);
  Serial.println("softap");
  launchWeb(1);
  Serial.println("over");
}

void createWebServer(int webtype)
{
  
  if ( webtype == 1 ) {
    server.on("/", []() {
        IPAddress ip = WiFi.softAPIP();
        String ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
        content = "<!DOCTYPE HTML>\r\n<html>Hello from ESP8266 at ";
        content += ipStr;
        content += "<p>";
        content += st;
        content += "</p><form method='get' action='setting'><label>SSID: </label><input name='ssid' length=32><input name='pass' length=64><input type='submit'></form>";
        content += "</html>";
        server.send(200, "text/html", content);  
    });
    server.on("/setting", []() {
        String qsid = server.arg("ssid");
        String qpass = server.arg("pass");
        if (qsid.length() > 0 && qpass.length() > 0) {
          Serial.println("clearing eeprom");
          for (int i = 0; i < 96; ++i) { EEPROM.write(i, 0); }
          Serial.println(qsid);
          Serial.println("");
          Serial.println(qpass);
          Serial.println("");
            
          Serial.println("writing eeprom ssid:");
          for (int i = 0; i < qsid.length(); ++i)
            {
              EEPROM.write(i, qsid[i]);
              Serial.print("Wrote: ");
              Serial.println(qsid[i]); 
            }
          Serial.println("writing eeprom pass:"); 
          for (int i = 0; i < qpass.length(); ++i)
            {
              EEPROM.write(32+i, qpass[i]);
              Serial.print("Wrote: ");
              Serial.println(qpass[i]); 
            }    
          EEPROM.commit();
          content = "<!DOCTYPE HTML>\r\n<html>";
          content += "<p>saved to eeprom... reset to boot into new wifi</p></html>";
        } else {
          content = "Error";
          Serial.println("Sending 404");
        }
        server.send(200, "text/html", content);
    });
  } else if (webtype == 0) {
    server.on("/", []() {
      server.send(200, "text/plain", "this works as well");
    });
    server.on("/setting", []() {
      server.send(200, "text/plain", "setting.");
    });
    server.on("/cleareeprom", []() {
      content = "<!DOCTYPE HTML>\r\n<html>";
      content += "<p>Clearing the EEPROM</p></html>";
      server.send(200, "text/html", content);
      Serial.println("clearing eeprom");
      for (int i = 0; i < 96; ++i) { EEPROM.write(i, 0); }
      EEPROM.commit();
    });
  }
}

void loop() {
  // put your main code here, to run repeatedly:
  server.handleClient();
}
```

It work now but main problem is when I connect to my SSID I got my IP address but can't close the soft  AP

this is serial monitor
```
Startup
Reading EEPROM ssid
SSID: xxx                             
Reading EEPROM pass
PASS: xxx                                                       
Waiting for Wifi to connect
66666663
WiFi connected
Local IP: 172.20.10.3
SoftAP IP: 192.168.4.1
Server started
```

I can join my `softap` open web browser and go to 
`http://192.168.4.1/` and it return `this works as well`

and I can join my real ap open web browser and go to 
`http://172.20.10.3/` and it return `this works as well`

How can I close the softAP?

igrr, 2015-05-05:
```WiFi.mode(WIFI_STA);```

closed 2015-05-14 by igrr

stozk, 2015-05-24:
Is it normal that the wifi credentials which I write to the eeprom are gone after I disconnect the power from the ESP?

stozk, 2015-05-25:
Ok seems like it was related to this error, I've updated my IDE:

https://github.com/esp8266/Arduino/issues/240

dogrocker, 2015-05-26:
No I use EEPROM to save wifi credentials, and I can disconnect the power off and on it can save the wifi credentials.

Santosh1971, 2015-11-19:
It doesnot support password entered as special character like @,# etc.. it stores it as %40 ,%23 etc.. and doesnot get connected to wifi. What to do? 
Currently I am managing by not having special character in my Password.
pl. advise.

luc-github, 2015-11-19:
@Santosh1971 you have to handle this with decode function: https://github.com/esp8266/Arduino/issues/454

Santosh1971, 2015-11-19:
Thanks Luc for your immediate response.. I checked it.. It is too complicated for me to understand. I will try searching if somebody has implemented it. it will be a great help if you update your example with this correction.