esp8266/Arduino #122: tmElements_t has not been declared

onkelfunny, 2015-04-26:
I try to Compile the DS3232RTC library and receive this error:

Arduino/libraries/DS3232RTC/DS3232RTC.h:137:26: error: 'tmElements_t' has not been declared
         static byte read(tmElements_t &tm);
                          ^
Arduino/libraries/DS3232RTC/DS3232RTC.h:138:20: error: 'tmElements_t' has not been declared
         byte write(tmElements_t &tm);

What can i do to fix this?

closed 2015-04-26 by onkelfunny

kqorany, 2016-02-11:
I am facing the same problem. Pls, tell me what have you did to fix this ?

reePIman, 2016-04-30:
I had this problem as well but resolved it by removing the #include <Time.h> and then re-including the library.  After re-including the library it included both Time and Timelib header files and viola it worked.  In the process of debugging this I also upgraded to Arduino IDE 1.6.8.

GanjaFarmer, 2016-08-11:
Add at the top of the sketch:
```#include <Time.h>
#include <TimeLib.h>
#include <DS3232RTC.h>```

KudosAbhay, 2017-02-24:
I was facing this error too, when I had used PubSubClient library with sim800 library. 
So, I included 
#include <Time.h>
#include <TimeLib.h>

in sim800.h file, and it started working.

Excalibur83, 2017-06-26:
Thanks a lot for your support.
Replacing <Time.h> with <TimeLib> is enough (works for me at least).
Time library call TimeLib library. But you can't delete Time.h files from your Library folder. Other Library in folder Time-Master call Time.h only.