esp8266/Arduino #123: digitalRead on NodeMCU user switch

hallard, 2015-04-26:
Hi there,

I've got a NodeMCU board with a flash switch and a user switch. The user switch is set as follow

[image]

R3 is not populated and here the full [schematic](https://raw.githubusercontent.com/nodemcu/nodemcu-devkit/master/Documents/NODEMCU_DEVKIT_SCH_BIG.png)

My sketch is as follow

```
#include "ESP8266WiFi.h"

void setup() 
{
  Serial.begin(115200);
   pinMode(16, INPUT);
}
void loop() 
{
  Serial.print("GPIO16=");
  Serial.println(digitalRead(16));
}
```

As soon I launch it, GPIO16 is displayed as 1 on serial then I push the user switch, then GPIO16 is displayed as 0, and then GPIO16 never get back to 1 when I release the switch. Is there something I missed or it is related to internal hardware (I could not find a descent datasheet) ? Of course I also tried with `INPUT_PULLUP` with same results

Thank you for your help

GerryKeely, 2015-04-26:
Hi
I think it's to do with the do with the way digitalRead for pin 16 is handled.if you look at the file "core_esp8266_wiring_digital.c" you will see that digitalRead for pin16 is different (around line 113)
Gerry

hallard, 2015-04-26:
Gerry,

you're right, this pin have special function (wake) 
[image]

But this pinout says I should be able to use it as normal I/O pin (no interrupt ok) I'd like to find a datasheet as good as the ATMel ones which tell us what we can do on each pin and what we can't ;-)

hallard, 2015-04-27:
Ok, @GerryKeely pointed me on the right way, I investigated on nodemcu source, key switch is mulplipexed with onboard led, this is really well done, just one pin for switch and LED. The trick is to do things as follow (set gpio16 to input to read then go back pin to output)
the following function does all  the stuff, you call it with parameter to  (LOW) to light the LED and HIGH to set the LED off, in return you have 0 if switch is pressed else 1
```
// KEY_LED functions
uint8_t key_led( uint8_t level)
{
  uint8_t temp;
  digitalWrite(16, 1);
  pinMode(16, INPUT);
  temp = digitalRead(16);
  pinMode(16, OUTPUT);
  digitalWrite(16, level);
  return temp;
}
```

following my original sketch is now
```
#include "ESP8266WiFi.h"

// KEY_LED functions
uint8_t key_led( uint8_t level)
{
  uint8_t state;
  digitalWrite(16, 1);
  pinMode(16, INPUT);
  state = digitalRead(16);
  pinMode(16, OUTPUT);
  digitalWrite(16, level);
  return state;
}

void setup() 
{
  Serial.begin(115200);
}

void loop() 
{
  // blink NodeMCU onboard led 
  uint8_t led = ((millis() % 1000) < 500);

  // Set the LED and read switch
  uint8_t gpio16 = key_led(led);

  Serial.print("GPIO16=");
  Serial.println(gpio16);
}
```

works like a charm

linagee, 2015-04-29:
Today I learned: NodeMCU has a way to tie GPIO16 and nRST together. (You need this to get out of ESP.deepSleep.) Just need to add that non-populated resistor R3!

hallard, 2015-04-29:
@linagee 
glad you learned, I think we all learn everyday with these devices ;-)
A quick question for you, If I populate R3 does this have any incidence on program if I do not use deep sleep ?

linagee, 2015-04-30:
If you populate it, you should not really be using GPIO16 for anything else unless you want it to reboot every time.

hallard, 2015-05-02:
@linagee,

Thanks, looking at the schematics, makes sense, except if I want to do a soft reset or use the user switch as real reset :-)

closed 2015-05-02 by hallard