esp8266/Arduino #124: Linux 32xBit error

sigrokBlack, 2015-04-26:
After successfully building the IDE on a 32Bit machine I get these error message, when I want to compile something:
Cannot run program "/home/eagle/Downloads/Arduino-esp8266/build/linux/arduino-1.6.1/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++": error=2, File or Path not found

What can I do?

Greetings from Germany
Joshua

SamuelGold, 2015-04-26:
I solve this problem, downolad zip from here and unpack https://github.com/jcmvbkbc/crosstool-NG
After do this:

cd crosstool-NG-lx106-g-/
./bootstrap
./configure --prefix=`pwd` && make
make
make install
./ct-ng xtensa-lx106-elf
./ct-ng build
cp -r ./builds/xtensa-lx106-elf/  /YOUR_PATH/arduino-1.6.1/hardware/tools/esp8266/

Hope it helps!

sigrokBlack, 2015-04-27:
Hi, i did it (compiling was fine)and i get this messages over and over:
Arduino: 1.6.1 (Linux), Platine: "Generic ESP8266 Module, 80 MHz, 115200"

Cannot run program "/home/eagle/Dokumente/arduino-1.6.1/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++": error=2, File or Path not found

The file xtensa-lx106-elf-g++ isn in the folder as it should be... 
What can i do?
joshua

SamuelGold, 2015-04-27:
Hello!
Go to the folder crosstool-NG-lx106-g-/
cd crosstool-NG-lx106-g-/
Then run this:
cp -r ./builds/xtensa-lx106-elf/ /home/eagle/Dokumente/arduino-1.6.1/hardware/tools/esp8266/

sigrokBlack, 2015-04-27:
I did it, but the file isn't there:
ls in /Dokumente/arduino-1.6.1/hardware/tools/esp8266/xtensa-lx106-elf/bin
xtensa-lx106-elf-addr2line     xtensa-lx106-elf-elfedit     xtensa-lx106-elf-gdb      xtensa-lx106-elf-ranlib
xtensa-lx106-elf-ar            xtensa-lx106-elf-gcc         xtensa-lx106-elf-gprof    xtensa-lx106-elf-readelf
xtensa-lx106-elf-as            xtensa-lx106-elf-gcc-4.8.2   xtensa-lx106-elf-ld       xtensa-lx106-elf-size
xtensa-lx106-elf-cc            xtensa-lx106-elf-gcc-ar      xtensa-lx106-elf-ld.bfd   xtensa-lx106-elf-strings
xtensa-lx106-elf-c++filt       xtensa-lx106-elf-gcc-nm      xtensa-lx106-elf-nm       xtensa-lx106-elf-strip
xtensa-lx106-elf-cpp           xtensa-lx106-elf-gcc-ranlib  xtensa-lx106-elf-objcopy
xtensa-lx106-elf-ct-ng.config  xtensa-lx106-elf-gcov        xtensa-lx106-elf-objdump

SamuelGold, 2015-04-27:
Did you have g++ installed in your system?
Try command:
g++

sigrokBlack, 2015-04-27:
Yes it's installed. What else can i do?

igrr, 2015-04-27:
The issue is that you are not using the lx106-g++ branch of crosstool-NG.
https://github.com/jcmvbkbc/crosstool-NG/archive/lx106-g++.zip

Or just clone the git repository and then ```checkout -b lx106-g++```.

sigrokBlack, 2015-04-28:
Thank you it works!!
But now esptool is missing ;) 
Cannot run program "/home/eagle/Dokumente/arduino-1.6.1/hardware/tools/esp8266/esptool
Where do i have to download esptool?

igrr, 2015-04-28:
Here's the github page: https://github.com/igrr/esptool-ck
I don't have a build for linux x86, but if you make one, I can post it on the releases page for others to use.
You can run ```make dist``` to build the release archive.
Same goes for the linux x86 toolchain — if you ```tar czvf``` it and upload somewhere, that would also be cool.

tytower, 2015-04-30:
Man this is a serious error issue . Its all over the net . People can't get past the compile error and this explanation is just dangerous . It needs to be fixed properly

tytower, 2015-05-01:
I don't seem to be able to edit my comment above
What I mean is getting the crosstool as suggested above is not a good thing to tell a casual user of the Arduino IDE.
"The issue is that you are not using the lx106-g++ branch of crosstool-NG."
So how does it get fixed in the ESP2966 hardware library that is being used by normal people?
I don't want to be downloading /compiling and installing programs cause I'll end up with a bloody great mess!

igrr, 2015-05-01:
The question was how to make this thing work on 32-bit Linux machine. The
answer is outlined above. I didn't mean to suggest that this is the way
that regular users should follow.

Well, the reason we don't have a version for Linux x86 is that I don't have
access to 32-bit Linux server that would allow me to build the toolchain
and esptool.
So when someone who has access to the right system builds the toolchain and
uploads it somewhere, I'll create a version of arduino for 32-bit Linux.

At this point you can either contribute to the project by building the
toolchain and sharing it, or wait until someone else does that.

igrr, 2015-05-01:
Clarification: I use Travis-CI for Arduino builds for Linux. They have a
64-bit environment. For the toolchain builds I use a vps from DigitalOcean.
These are also 64-bit.
I will try setting up a 32-bit Vagrant environment to build the toolchain.
The reason I haven't done this before is that no one actually came asking
for a 32-bit Linux version until last week :)

tytower, 2015-05-01:
Good stuff - I have such a machine and can compile on it from the usual ./configure , make ,make install. Buts thats about the depth of my knowledge. I am happy to do so if you could lead me by the nose

tytower, 2015-05-01:
tytower@yahoo.com

tytower, 2015-05-01:
I ran the configure ,make make install on my machine and got this directory . Is that of any use ?

tytower, 2015-05-01:
I hav'nt got this system worked out yet to edit posts . Edit doesn't work with my browser. I attached a file to the email I sent you but I don't see it here so I don't know how to get the compiled directory you seek to you ? Hang on I see something below,nope tried dragging a tar.gz file but no go. Selected it but no go. Edit in full screen didn't work either

igrr, 2015-05-01:
You need to run:

./bootstrap
./configure --prefix=pwd && make
make
make install
./ct-ng xtensa-lx106-elf
./ct-ng build
tar czvf archive.tgz ./builds/xtensa-lx106-elf/

and then upload archive.tgz somewhere, wetransfer.com is fine.

tytower, 2015-05-01:
Got to ./ct-ng then errors from there like so 

# ./ct-ng xtensa-lx106-elf
bash: ./ct-ng: make: bad interpreter: No such file or directory
# ./ct-ng build
bash: ./ct-ng: make: bad interpreter: No such file or directory
# tar czvf archive.tgz ./builds/xtensa-lx106-elf/
tar: ./builds/xtensa-lx106-elf: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
# 

So I'll try to put the previous gz file where you suggest . I attached it to the previous email. Where did that go ?

tytower, 2015-05-01:
Well its gone up to wetransfer too but I don't see it here anywhere

tytower, 2015-05-02:
So did what I sent to you  have any use ?

igrr, 2015-05-02:
You didn't send me the link to the uploaded file, did you?

tytower, 2015-05-02:
Mate sorry . I didn't get a link from WeTransfer It sent it to the reply email address which seems to be here . I also cloned and "make dist" as per instructions for esptool and sent it , so I have both zip files and will put them on a file holder someplace shortly. When you say "send them to me " how can I do that when you don't publish your email address or any way to get you except here?
You must have set it up that way I guess

igrr, 2015-05-02:
ivan@esp8266.com -- it's actually written in the Readme file. otherwise just
copy&paste the link you got from wetransfer.

tytower, 2015-05-02:
Didn't see it there -Here is possibly accessible but I'll email you directly
http://www.filedropper.com/esptool32tar
http://www.filedropper.com/toolchaintrytar
IIt would be good that I get the email with your picture page - That was everywhere Hi Hi

tytower, 2015-05-04:
So were either of them of any use ?

tytower, 2015-05-05:
Here is where I am atm
http://www.esp8266.com/viewtopic.php?f=6&t=2872&p=16517#p16517
Sitting waiting . Any progress?

igrr, 2015-05-05:
@tytower I checked and it looks like Travis doesn't have 32-bit build agents currently. So we need to provision a 32-bit vagrant env for travis builds — but I haven't yet had time to do that.

tytower, 2015-05-05:
OK Ivan Happy to help if I can

--------------------------------------------------

tytower, 2015-05-09:
Seeing as there was nothing happening here any time soon I decided to fire up a 64 bit version of Puppy Linux called "Fatdog" and install the java , and ArduinoIDE as needed. Different processor too -Intel rather than AMD.  So with other problems that took two days to get up and running and what do you think happened when compiling the ESP8266 WiFi sketch ??
Exactly the same bloody error as before !!!

I think you want to have another good hard look at this as something in your library is not right and its not just a 32 bit IDE Problem as you say it is .

tytower, 2015-05-09:
I might also add that I commented on Sandeeps git hub page that some entries in the directory listings of his library did not look right and his response has not arrived yet and he says he got his library as a fork off yours so the same errors are probably here too

consolacion, 2015-05-12:
got the same error on ubuntu32 bit, eventhough the file it says it is missing is clearly present. Tried it on  Linux 64 as well... indeed, same problem

igrr, 2015-05-12:
@consolacion Could you please try to run this from command line?
```xtensa-lx106-elf-g++ --version```
(appending the correct path before xtensa-lx106-elf-g++)
and see if it runs from the command line?

consolacion, 2015-05-12:
it says 'command not found, eventhough I see the file right in front of me

igrr, 2015-05-12:
can you run ```readelf -l xtensa-lx106-elf-g++``` to see which library is missing?

consolacion, 2015-05-12:
Do not really see anything missing, but i do get a bunch of output
Entry point 0x4027a0
There are 8 program headers, starting at offset 64

Program Headers:
  Type           Offset             VirtAddr           PhysAddr
                 FileSiz            MemSiz              Flags  Align
  PHDR           0x0000000000000040 0x0000000000400040 0x0000000000400040
                 0x00000000000001c0 0x00000000000001c0  R E    8
  INTERP         0x0000000000000200 0x0000000000400200 0x0000000000400200
                 0x000000000000001c 0x000000000000001c  R      1
      [Requesting program interpreter: /lib64/ld-linux-x86-64.so.2]
  LOAD           0x0000000000000000 0x0000000000400000 0x0000000000400000
                 0x00000000000a4fc4 0x00000000000a4fc4  R E    200000
  LOAD           0x00000000000a5000 0x00000000006a5000 0x00000000006a5000
                 0x0000000000000f48 0x0000000000002be8  RW     200000
  DYNAMIC        0x00000000000a5028 0x00000000006a5028 0x00000000006a5028
                 0x00000000000001c0 0x00000000000001c0  RW     8
  NOTE           0x000000000000021c 0x000000000040021c 0x000000000040021c
                 0x0000000000000044 0x0000000000000044  R      4
  GNU_EH_FRAME   0x0000000000099c20 0x0000000000499c20 0x0000000000499c20
                 0x0000000000001f0c 0x0000000000001f0c  R      4
  GNU_STACK      0x0000000000000000 0x0000000000000000 0x0000000000000000
                 0x0000000000000000 0x0000000000000000  RW     8

 Section to Segment mapping:
  Segment Sections...
   00     
   01     .interp 
   02     .interp .note.ABI-tag .note.gnu.build-id .gnu.hash .dynsym .dynstr .gnu.version .gnu.version_r .rela.dyn .rela.plt .init .plt .text .fini .rodata .eh_frame_hdr .eh_frame 
   03     .ctors .dtors .jcr .dynamic .got .got.plt .data .bss 
   04     .dynamic 
   05     .note.ABI-tag .note.gnu.build-id 
   06     .eh_frame_hdr 
   07

consolacion, 2015-05-14:
friend had the same problem on windows 7, 32 bits

tytower, 2015-05-16:
Round and round skirting the issue . You need to look again . Here is the full error message trying to run the Arduino IDE 1.6.4 on 32 bit version

# ./arduino
Java HotSpot(TM) Server VM warning: You have loaded library /opt/arduino-1.6.4/lib/libastylej.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
java.lang.UnsatisfiedLinkError: /opt/arduino-1.6.4/lib/libastylej.so: /opt/arduino-1.6.4/lib/libastylej.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch)
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(Unknown Source)
	at java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.lang.Runtime.load0(Unknown Source)
	at java.lang.System.load(Unknown Source)
	at cc.arduino.packages.formatter.AStyleInterface.<clinit>(AStyleInterface.java:42)
	at cc.arduino.packages.formatter.AStyle.<init>(AStyle.java:52)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(Unknown Source)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(Unknown Source)
	at java.lang.reflect.Constructor.newInstance(Unknown Source)
	at java.lang.Class.newInstance(Unknown Source)
	at processing.app.Editor.createToolMenuItem(Editor.java:917)
	at processing.app.Editor.addInternalTools(Editor.java:940)
	at processing.app.Editor.buildToolsMenu(Editor.java:690)
	at processing.app.Editor.buildMenuBar(Editor.java:494)
	at processing.app.Editor.<init>(Editor.java:221)
	at processing.app.Base.handleOpen(Base.java:889)
	at processing.app.Base.handleOpen(Base.java:854)
	at processing.app.Base.handleNew(Base.java:745)
	at processing.app.Base.<init>(Base.java:444)
	at processing.app.Base.guardedMain(Base.java:215)
	at processing.app.Base.main(Base.java:128)
/opt/arduino-1.6.4/lib/libastylej.so: /opt/arduino-1.6.4/lib/libastylej.so: wrong ELF class: ELFCLASS64 (Possible cause: architecture word width mismatch)
Cannot load native library /opt/arduino-1.6.4/lib/libastylej.so
The program has terminated!
# 
Now this may be your library or not . I don't have the experience to tell but I want to get on and run this properly  using your library . Can you help at all ?

igrr, 2015-05-17:
1.6.4? Where did you download it from — Arduino.cc website?
This fork is still 1.6.1.

tytower, 2015-05-18:
Of course

----------------------------------------------------------------------------On Sun, 17 May 2015 02:22:32 -0700

igrr, 2015-05-19:
The latest board manager package has 32-bit Linux toolchain included (Readme has been updated with installation instructions).

consolacion, 2015-05-19:
That seems to be working Ivan, at least it compiles without errors :-)  Great. Thanks. Quite some work you have done.
Installation through the boardmanager was a breeze! Well done

consolacion, 2015-05-19:
btw... it was running smooth already on my 64 bits windows 8.1

igrr, 2015-05-19:
Kudos to @majenkotech for building the 32-bit linux toolchain! :+1:

closed 2015-05-19 by igrr

consolacion, 2015-05-19:
kudo's to both :-)

tytower, 2015-05-24:
Hi Ivan , I am having trouble getting realistic readings from a pressure sensor BMP180 on my ESP8266 ESP12 boards.

I get this 

temperature: 116.81 deg C, 242.25 deg F
absolute pressure: -1429.59 mb, -42.22 inHg
relative (sea-level) pressure: -1563.70 mb, -46.18 inHg
computed altitude: 750 meters, 2461 feet

The temp should be about 21.5 C and the pressure about 950 mb.
The same device gives good readings on an Arduino ATMega328P.
I have played with the Wire.setClock() function settings tried 
40000 , 100000,400000. All without difference. Here is my code if needed .

Can you suggest a possible fix?
[code]
//#include <SPI.h>

#include <DHT.h>
#include <stdio.h>
#include <SFE_BMP180.h>
#include <Wire.h>
#include <ESP8266WiFi.h>

#define DHT_PIN 2    //GPIO2  DHT22 for my sensorRHT03
#define TIMEOUT 3000 //3 second timout on Sparkfun Server response
#define SPARKFUN_UPDATE_TIME 20000         //Update SparkFun data server every 60000 ms (1 minute).
#define debug
#define ALTITUDE 750.0 // Altitude of My HQ in  meters
unsigned long timer1 = 0;
unsigned long timer2 = 0;
char server[] = "data.sparkfun.com";    // name address for data.sparkFun (using DNS)
// Set the static IP address to use if the DHCP fails to assign

const String publicKey = "RMzp9ANyqzfjw4drzNnZ";
const String privateKey = "--------------------";
const byte NUM_FIELDS = 4;
const String fieldNames[NUM_FIELDS] = {"humidity" , "pressure2" , "temp" , "temp2"};
String fieldData[NUM_FIELDS];

//Character array to hold results
char results[4];

long lastAttemptTime=0; 
float humidity=0;
float temp=0;
float temp2=0;
long pressure2=0;

const int dataSDA = 5; //BMP180 SDA 
const int clockSCL = 4;//BMP180 SCL 

double T,P,p0,a;

long lastReadingTime = 0;

SFE_BMP180 pressure;

DHT dht(DHT_PIN,DHT22,15);
WiFiClient client;
// Time to sleep (in seconds):
const int sleepTimeS = 30;

//////////////////////////////////////////////////////////////////////////////////

void setup(){ 

const char* ssid = "ME-HOME";
const char* password = "";

[... 201 lines cut ...]
    Serial.println(fieldData[2] );
    Serial.println(fieldData[3] );
    delay(500);

    
    // deepSleep time is defined in microseconds. Multiply
    // seconds by 1e6 
    Serial.println(ESP.getChipId());
    Serial.println(ESP.getFlashChipId());
    Serial.println(ESP.getFlashChipSpeed());
    Serial.println(ESP.getFlashChipSize());
  //Serial.println(" Going to Sleep !" );
  #endif
  
  //ESP.deepSleep(sleepTimeS * 1000000);
  
  }
 }

[/code]
-- 
Yahoo <tytower@yahoo.com>

-- 
Yahoo <tytower@yahoo.com>