esp8266/Arduino #128: Wifi.config not working?

ArnieO, 2015-04-26:
I am trying to set static IP address, and tried this:
    IPAddress ip(192,168,0,124);
    (...)
    WiFi.config(ip);
    WiFi.begin(ssid, password);
Compiler error: It wants three arguments: IP, gateway, subnet.

New attempt:
    IPAddress ip(192,168,0,124);
    IPAddress gateway(192,168,0,1);
    IPAddress subnet(255,255,255,0);
    (...)
    WiFi.config(ip, gateway, subnet);
    WiFi.begin(ssid, password);
This compiles, but the IP address does not change. 

Please advise.

Imroy, 2015-04-27:
You need to do config() after begin().

ArnieO, 2015-04-27:
Thank you Imroy - with config() after begin() it works!
Lacking documentation, I followed a Wifi shield example on the Arduino web page that uses begin() before config(). 
Apparently, the ESP needs it the other way around.

closed 2015-04-27 by ArnieO

dilushan, 2015-08-08:
Thanks this solved my problem also. In my case that is for the WiFi.softAPConfig

DLavin, 2015-09-22:
This also works for WiFiMulti.run(), where there is no begin().

daviesrv, 2016-01-29:
Thanks for the info. Cleared my issue.

alaintanguay, 2016-03-09:
Hello 
I try to change the Mask on softAPConfig , to 255.0.0.0 , but still 255.255.255.0
The IP have change , but not the mask

this is my code

IPAddress ip(10,0,0,101);
WiFi.softAP(apssid,appassword);
WiFi.softAPConfig(ip, ip, IPAddress(255,0,0,0));

Thanks for your support

pit001, 2016-05-08:
same for me - SGW isn't working... 
WiFi.config(IPAddress(192, 168, 4, 23), IPAddress(192, 168, 0, 5), IPAddress(255, 255, 240, 0));

thorathome, 2016-06-04:
I have found WiFi.config issue, for some reason, is particularly sensitive to serial upload speeds.  
Using Arduino IDE 1.6.9, uploading at 115200, for example, I ran into intermittent problems with WiFi.config on my Chinese v1 ESP8266 board.  Ratcheting down to 57600, however, cured it. 
(Tools -> Upload Speed -> 57600)

Go figure.  But slowing down the serial upload got me back on track.   Maybe it'll help you.

drmpf, 2016-09-22:
Rolling back to V2.2.0 fixed all my static IP connection problems

flagtrax, 2016-09-25:
Using NodeMCU/lua, I've found several issues trying to connect after using config and changing the password or auth parameters for AP mode or stationap . Using NodeMCU, according to the doc's the config statement for AP mode is in the form wifi.ap.config(cfg) Where:
ssid SSID chars 1-32
pwd password chars 8-64
auth authentication method, one of wifi.OPEN (default), wifi.WPA_PSK, wifi.WPA2_PSK, wifi.WPA_WPA2_PSK
channel channel number 1-14 default = 6
hidden 0 = not hidden, 1 = hidden, default 0
max maximal number of connections 1-4 default=4
beacon beacon interval time in range 100-60000, default = 100
I'm loading via ESPlorer IDE. If I omit password all together as well as auth parameters, I can connect (sometimes) but my android phone shows a secure wpa/psk connection no matter how I set the auth parameter. According to what I read (i think) I think it says (using lua ) the parameter "wifi.OPEN" should set it to unprotected access. My laptop wont connect at all (using windows 10) simply stating "cannot connect to this network". When I enter the auth parameter (auth=wifi.open) I get an error at compile time that says it was expecting a numeric (not string) entry. So I entered a 0 (zero), and the compiler took it but nothing changed. Similarly If I enter a password, sometimes it works sometime it doesn't. For the NodeMCU community, the documentation on this is very sketchy. As far as upload speed goes, ESPlorer uart speed is 9600 by default. (yeah, pretty slow). This has all been using a generic ESP12E, flashed with NodeMCU, using ESPlorer as an IDE.

palani12, 2017-06-20:
Iam using wifi shield Esp8266  is connected the my laptop but wifi shield getting data not produce in laptop. my wifi shield port is  showing in my laptop network settings .i was click the  showing wifi shield port is open then show  NO internet open. what is the problem showing NO internet open. then i was do the trouble shooting in my network settings after trouble shooting  problem found  --The DNS Server is not responding.

tfoutfou, 2017-12-17:
I was having the same problem and i thing i found the solution .

the order on the manual is WiFi.config(ip, dns, gateway, subnet)

but after inspection of the library it seem the the CORRECT order have to be
IP , GATEWAY , SUBNET , DNS 

From a computer guy point of view it is logical .

So for example if you want to declare a DNS wich is mandatory to resolve names
you have to declare   IP, gateway , subnet , then DNS 
you CANT declare DNS without filling the first three parameters

akhil432, 2018-01-16:
I have downloaded tcp client application from playstore.i have to connect it to my wifi module esp8266.my ip address is 192.168.1.128 then what is my port number..pls help me

justin8, 2018-03-24:
Just to add to this. As @thorathome said, slowing down to 57600 baud uploads completely fixed this issue for me. Nothing else I tried would work.

andrewradke, 2018-04-09:
I've run into this issue primarily after the I've just uploaded a new firmware to the ESP8266 directly from the Arduino IDE.

At first it assigns itself a DHCP address even with a call to `WiFi.config(ip, gateway, subnet, dns_ip);`
A reset, or two, later it correctly assigns itself an IP address.

My work around, which so far works, is to send some data through the connection and then check whether the IP address is as expected. In my case I'm sending a syslog packet and a DNS query to get enough data through (delays and Serial.print() don't seem to affect it).

Below is my currently working code:
```
  if (use_staticip) {
    WiFi.config(ip, gateway, subnet, dns_ip);
  }

  if (use_syslog) {
    setupSyslog();  // Sets the syslog server, hostname and priority
  }

  if (use_staticip) {
    logString = str_ipv4 + str_space + cfg_configured;
    logMessage(app_name_sys, logString, true);
    IPAddress tmpip;
    tmpString = IPtoString(dns_ip);
    WiFi.hostByName(tmpString.c_str(), tmpip);
    while (IPtoString(WiFi.localIP()) != IPtoString(ip) ) {
      Serial.println(str_dot);
      WiFi.config(ip, gateway, subnet, dns_ip);
    }
  }
```

andrewradke, 2018-04-20:
What I've determined by watching network traffic and logs is that after uploading a new firmware directly from the Arduino IDE the ESP8266 will try to get it's IP via DHCP even if you give it a static address via `WiFi.config(ip, gateway, subnet, dns_ip);` It will also continue to try to get a DHCP address periodically as well. Resetting the ESP8266 doesn't change the behaviour.

The ESP8266 will behave properly once power has been removed and restarted.

As such my solution above won't always catch the problem depending on the timing of any DHCP reply from the DHCP server after setting the static IP. Adding the check inside your loop() will catch things eventually if the address does get set via DHCP but that is a poor workaround that only makes sense if the underlying bug can't be fixed.

d-a-v, 2018-04-20:
Thanks for having opened a new issue in #4656

jstuewe, 2019-09-30:
Many older examples left off the DHCP server and it used to work fine with just WiFi.config(staticIP, gateway, subnet). After updating to a recent (ie. 2.5.0) library the WiFi.hostByName function in my NTP routine stopped working and just returned "(IP unset)".

Apparently the older libraries didn't need the DHCP server in WiFi.config but the newer libraries do.
I fixed it by adding the 'gateway' at the end as the DHCP server.  ie. WiFi.config(staticIP, gateway, subnet, **gateway**)

Pablo2048, 2019-09-30:
??? DHCP server address is not used in hostByName() function - actually you don't have to set DHCP server address, because that address is contained in DHCP response packet. Did you mean DNS server?

plrfishing, 2020-02-22:
I agree with jstuewe that the 4th parameter DNS server is required when I started using the NTPClient library. I fought this issue for 2 hours tonight after adding the NTPClient library and code to a previously working version of code. Google search found this github posting and his reply solved my problem.

DeepakAgr1411, 2021-03-29:
Posting from jstuewe is a lifesaver. I was stuck since last 2 weeks with not being able to access internet through my esp8266. Adding gateway as the 4th parameter fixes the issue. Funny that devs made a breaking change without documenting it. The official documentation for WiFi.config() is obsolete and useless.

jstuewe, 2021-03-29:
> Posting from jstuewe is a lifesaver.....
The official documentation for WiFi.config() is obsolete, incomplete and useless.

Glad my comment helped. They definitely need to update the documentation.

d-a-v, 2021-03-29:
Arduino legacy API: https://www.arduino.cc/en/Reference/EthernetBegin
Our API as stated in documentation: [v2.7.4](https://arduino-esp8266.readthedocs.io/en/2.7.4_a/esp8266wifi/station-class.html?highlight=WiFi.config#config)
Both orders are [acceptable and managed](https://github.com/esp8266/Arduino/blob/2843a5ace10eb696f19c9d754be7bbf6331046b5/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L278L279)

note: in current master, this code is split between [there](https://github.com/esp8266/Arduino/blob/5ac64ffa27461161401971796150ef7f2b26a77d/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L273L274) and [there](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/LwipIntf.cpp#L27).

If one of them is not working, please open a new issue for us to keep track of it so we can solve them in the next versions.

jstuewe, 2021-05-23:
The issue in the [v2.7.4 API](https://arduino-esp8266.readthedocs.io/en/2.7.4_a/esp8266wifi/station-class.html?highlight=WiFi.config#config) document is that the DNS parameter is listed as optional.  Using a static IP address worked without a DNS entry in older libraries but in 2.5.0 and above it can't resolve DNS entries if you don't include the DNS parameter.  I don't know if it's an issue with the code or the documentation but one or the other needs to be updated. 

`dns1, dns2 - **optional** parameters that define IP addresses of Domain Name Servers (DNS) that maintain a directory of domain names (like e.g. www.google.co.uk) and translate them for us to IP addresses
`

This is also discussed in https://github.com/esp8266/Arduino/issues/2371

d-a-v, 2021-05-23:
Do I understand that you can resolve addresses with a static IP setup without giving a DNS parameter ?
That'd be odd. You'd have received a DNS entry from the DHCP server before you configure your static address, or in older versions of the core (lwIP1.4?) DNS entries like google's were hardcoded.
You can see that by enabling the serial console.
It is anyway mandatory to use a dns server for address resolution in any case, static or dhcp. And such DNS server addresses cannot popout from nothing.

justin8, 2021-05-23:
Kind of? Most DHCP setups will provide the DNS servers to their clients. It would be unusual to need a manually defined DNS server together with DHCP

d-a-v, 2021-05-23:
I was answering to that:
> Using a static IP address worked without a DNS entry

justin8, 2021-05-23:
Right, yeah. That would only work as you described. Name resolution needs something to provide that data, it won’t “just work”

jstuewe, 2021-05-24:
It did 'Just work' before 2.5.0 so older code and examples stop working after updating. Probably due to the hard coded DNS serves d-a-v mentioned above.  The issue is that the documentation lists DNS as optional where it is required for most use cases.  It would be nice to include that in the documentation.  

I don't know if it would be appropriate, but it it would work in many cases to include the gateway as a default DNS server if one isn't provided. My code uses WiFi.config(staticIP, gateway, subnet, gateway)