esp8266/Arduino #129: Mac os file upload problems

racingpassion, 2015-04-27:
Hi let me start by saying it's great the work you're doing for the community keep up the good work thanks. My problem is when uploading sketch to the ESP I get the following errors,

Sketch uses 196,196 bytes (37%) of program storage space. Maximum is 524,288 bytes.
Uploading 35920 bytes from /var/folders/x5/6zgc42x17w9_qnl9g4m9bf_h0000gn/T/build254825791600570695.tmp/WiFiWebServer2.cpp_00000.bin to flash at 0x00000000
warning: espcomm cmd: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed

Is there a straight forward remedy to fix this problem on a Mac OS system .
Kind regards thanks in advance
Andy.

circuitFlower, 2015-04-27:
I was getting this. I couldn't get around it. So I installed a virtual linux machine and tried that way. with the virtual linux machine (Ubuntu), I'm getting null pointer exceptions on the code. I'm still stuck.

racingpassion, 2015-04-27:
Thanks for your reply I will try your solution until A permanent fix has been found, once again thanks for your reply kind regards Andy.

igrr, 2015-05-04:
@racingpassion Could you please try the latest version of esptool (0.4.3)? I fixed a bug which could cause failed uploads.
Also what is the usb-serial adapter that you are using?

gerardwr, 2015-05-04:
@racingpassion

I can confirm that the latest software solves similar upload problems I had with previous software verslons.

I'm on Mac 10.10 and installed acc. the sandeepmistry github.

closed 2016-02-29 by igrr