esp8266/Arduino #130: Possibly broken generated sketch

av1024, 2015-04-27:
The latest (f1e6a72d8c) commit seems broken for me... Looks like uploaded code does not start. After reboot standard text at 74480bps displayed, ends with "main.c" and neither Serial nor digitalWrite don't work.

The total sketch size is 19kb bigger comparing to old (pre str_tok fix) commit (257308 vs 230776 bytes

The full erase (write blank512k) does not solves issue

Links2004, 2015-04-27:
can confirm this,
latest working commit is 6a37b4461caa8bbff1ba473ec55f248cb5db3522 looks like the smartconfig merge makes problems (commit 196992a300846413bf576ee506f98899906ac5af).

igrr, 2015-04-27:
Just built WiFIScan and mDNS sample sketches and they run fine.
Could you please check the md5 and size of the binaries generated from the Blink sample sketch (without any modifications)?
Here's what I get:
```
-rw-r--r--  1 igrokhotkov  staff   31760 Apr 28 00:34 Blink.cpp_00000.bin
-rw-r--r--  1 igrokhotkov  staff  165880 Apr 28 00:34 Blink.cpp_40000.bin
MD5 (Blink.cpp_00000.bin) = 68c35701d1ae5e19beaa7a56c65a827d
MD5 (Blink.cpp_40000.bin) = 3f4bfe65afbcd79be762d6975fd44f87
```

Links2004, 2015-04-27:
if i only add:
```-lsmartconfig```
```*libsmartconfig.a:(.literal .text .literal.* .text.*)```
it is still working

if i use the changes in the ```ESP8266WiFiClass``` class it stops working

ok i build the Blink

Links2004, 2015-04-27:
blink @ 80Mhz
Blink.cpp_00000.bin AECFA03303248482E6BBF1B317B906E9
Blink.cpp_40000.bin 4DB0716064326BB10A280ABD60CEB973
compiled on Windows

igrr, 2015-04-27:
@av1024 are you compiling on Windows as well?

Links2004, 2015-04-27:
@igrr witch version of the xtensa-lx106-elf compiler you use on linux?

xtensa-lx106-elf-gcc.exe --version
xtensa-lx106-elf-gcc.exe (GCC) 4.8.2

i test one from the 17.02.2015 (from the 1.6.1 zip) and a version from 25.02.2015.
both are not working and i get different md5 for it.

igrr, 2015-04-27:
I'm on a Mac.
xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2

Links2004, 2015-04-27:
this version: https://github.com/jcmvbkbc/crosstool-NG/tree/crosstool-ng-1.20.0 ?

igrr, 2015-04-27:
yes, branch lx106-g++, commit 7c6bc14.

Links2004, 2015-04-27:
i dont find the commit sha1 i try to build b404fb9ec3a5c59e6b4d90741d39f7d1937431cd for windows hope this will work in mingw

igrr, 2015-04-27:
I'm not sure this is a toolchain issue. Why would windows toolchain stop working after some minor changes to the code?

Actually OP mentioned that the bootloader output ends with *main.c*. If I do full erase then I do get this:
```
ets Jan  8 2013,rst cause:1, boot mode:(3,7)

ets_main.c 
```
Perhaps something is wrong with the binary image generation?

Links2004, 2015-04-27:
yes i get the same bootloader messages

av1024, 2015-04-27:
@igrr, Linux Mint x64.
Checked out and build at 11:45 MSK
Unfortunately I have only ssh access for this PC for at least two weeks and I don't know how to invoke arduino buildsystem via terminal ((

Links2004, 2015-04-27:
if you have ssh you can use X11-Forwarding  ```ssh -X```

av1024, 2015-04-27:
I did it! )) Google and @Links2004 helped me )

$ git log
commit f1e6a72d8cb628cd8fa82d6ee11e97acef469b17
Author: Ivan Grokhotkov <igrr@users.noreply.github.com>
Date:   Mon Apr 27 05:49:02 2015 +0800

Merge pull request #127 from Links2004/esp8266
ram macro, fix warnings, #125

=== generic 80MHz 8266:
.../arduino --verify Bilnk.ino

-rw-rw-r-- 1 av av  31776 апр.  27 21:57 Blink.cpp_00000.bin
-rw-rw-r-- 1 av av 165880 апр.  27 21:57 Blink.cpp_40000.bin

$ md5sum *.bin
2ec8e8fa6f9c31febf45c083c8639940  Blink.cpp_00000.bin
b083e11acefbb4de5fd301088ece12a1  Blink.cpp_40000.bin

igrr, 2015-04-27:
@Links2004 can you please upload the Blink.cpp.elf and two .bin files somewhere? (wetransfer.com should be good)

chris153002, 2015-04-27:
i also get the same message, when i upload using the arduino ide and when i upload using an external app (running on windows)

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

ets_main.c

igrr, 2015-04-28:
Unfortunately I can't reproduce the issue myself.
I can try to check how the binaries are generated from your elf. Perhaps this is some corner case of incorrect checksum calculation in esptool.

chris153002, 2015-04-28:
here is the elf file as well as both bin files compiled from the "HelloServer" example in the ide

http://we.tl/aYy4QtBeUR

igrr, 2015-04-28:
Thanks for the files!
Downloaded, unpacked:
```
-rw-r--r--@ 1 igrokhotkov  wheel  548341 Apr 28 08:18 HelloServer.cpp.elf
-rw-r--r--@ 1 igrokhotkov  wheel   33264 Apr 28 08:18 HelloServer.cpp_00000.bin
-rw-r--r--@ 1 igrokhotkov  wheel  203704 Apr 28 08:18 HelloServer.cpp_40000.bin
```
Verified that esptool produces identical binaries from your elf file:
```
$  ~/e/esptool-ck/esptool -eo HelloServer.cpp.elf -bo mHelloServer.cpp_00000.bin -bs .text -bs .data -bs .rodata -bc -ec -eo HelloServer.cpp.elf -es .irom0.text mHelloServer.cpp_40000.bin -ec
$ md5 *.bin
MD5 (HelloServer.cpp_00000.bin) = 2a67292e548e1d9dd89839f372ffbd85
MD5 (HelloServer.cpp_40000.bin) = b6ff635acd8bbc4b817a7a9dd77876c2
MD5 (mHelloServer.cpp_00000.bin) = 2a67292e548e1d9dd89839f372ffbd85
MD5 (mHelloServer.cpp_40000.bin) = b6ff635acd8bbc4b817a7a9dd77876c2
```
So at least esptool output doesn't depend on the platform.
Next I flashed your binaries to the esp:
```
$ ~/e/esptool-ck/esptool -cp /dev/tty.usbserial-A9G7ZHP5 -cf HelloServer.cpp_00000.bin -ca 0x40000 -cf HelloServer.cpp_40000.bin 
Uploading 33264 bytes from HelloServer.cpp_00000.bin to flash at 0x00000000
.................................
Uploading 203704 bytes from HelloServer.cpp_40000.bin to flash at 0x00040000
.......................................................................................................................................................................................................
```
The bootloader output is as follows:
```
ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 28896, room 16 
tail 0
chksum 0xb2
load 0x3ffe8000, len 1548, room 8 
tail 4
chksum 0x7f
load 0x3ffe8610, len 2776, room 4 
tail 4
chksum 0x48
csum 0x48
```
And then the sketch runs okay, trying to connect to some WiFi...
I'm stuck, really.

Links2004, 2015-04-28:
it is strange i test bb082c9879311aa0da6fe4b22cf3dafbb1dbe613 and the blink is working but if i compile code who is working in #links2004/esp8266_dev1 is broken.

tested on a ESP-7 and ESP-12 module @80Mhz

Links2004, 2015-04-28:
at the moment i think the problem comes form the last SDK merge, they increase the possible size
https://github.com/esp8266/Arduino/blob/esp8266/hardware/tools/esp8266/sdk/ld/eagle.app.v6.ld#L8

i try to write a sketch to test the real flash size of the hardware. May we have different Flash chips on our boards.
[image]

Links2004, 2015-04-28:
```cpp
#include <Arduino.h>

extern "C" {
#include "user_interface.h"
}

#define SERIAL_OUT Serial

bool test_flash_ll(uint32_t pAddr) {
    uint32_t tmp;
    if(spi_flash_read(pAddr, &tmp, 1) == SPI_FLASH_RESULT_OK) {
        return true;
    }
    return false;
}

bool test_flash(uint32_t pAddr) {

    SERIAL_OUT.print(F("[Flash] addr: 0x"));
    SERIAL_OUT.print(pAddr, HEX);
    SERIAL_OUT.print(" - ");
    SERIAL_OUT.print(pAddr, DEC);
    SERIAL_OUT.print(" ");

    if(test_flash_ll(pAddr)) {
        SERIAL_OUT.println("OK");
        return true;
    }
    SERIAL_OUT.println("FAIL");
    return false;

}

//The setup function is called once at startup of the sketch
void setup()
{
    Serial.begin(115200);
    Serial1.begin(921600);
    Serial1.setDebugOutput(true);

    SERIAL_OUT.println();

    for(uint8_t t = 4; t > 0; t--) {
        SERIAL_OUT.print(F("[SETUP] BOOT WAIT "));
        SERIAL_OUT.print(t);
        SERIAL_OUT.println(F("..."));
        SERIAL_OUT.flush();
        delay(1000);
    }

    SERIAL_OUT.print(F("[SETUP] Flash ID: 0x"));
    SERIAL_OUT.println(spi_flash_get_id(), HEX);

    #define Mbit ((1024*1024)/8)

    uint32_t step = Mbit; //1 Mbit
    uint32_t addr = Mbit;
    while(1) {
        if(test_flash(addr)) {
            addr += step;
        } else {
            step /=2;
            addr -= step;
        }

        if(step <= 1) {
            test_flash(addr);
            break;
        }
    }
    SERIAL_OUT.println("----------------------");
    SERIAL_OUT.print(F("[Flash] Size: 0x"));
    SERIAL_OUT.print(addr, HEX);
    SERIAL_OUT.print(" - ");
    SERIAL_OUT.println(addr, DEC);

    SERIAL_OUT.println("Fin.");

}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
}
```

my output:

```
------------------ ESP8266 BOOTLOADER ------------------
 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6) 
 
load 0x40100000, len 28896, room 16  
tail 0 
chksum 0x1b 
load 0x3ffe8000, len 1332, room 8  
tail 12 
chksum 0xeb 
ho 0 tail 12 room 4 
load 0x3ffe8540, len 1796, room 12  
tail 8 
chksum 0xea 
csum 0xea 
Rüþ
------------------ ESP8266 APP ------------------
 
[SETUP] BOOT WAIT 4... 
[SETUP] BOOT WAIT 3... 
[SETUP] BOOT WAIT 2... 
[SETUP] BOOT WAIT 1... 
[SETUP] Flash ID: 0x1340C8 
[Flash] addr: 0x20000 - 131072 OK 
[Flash] addr: 0x40000 - 262144 OK 
[Flash] addr: 0x60000 - 393216 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x70000 - 458752 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x78000 - 491520 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7C000 - 507904 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7E000 - 516096 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7F000 - 520192 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7F800 - 522240 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FC00 - 523264 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FE00 - 523776 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FF00 - 524032 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FF80 - 524160 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FFC0 - 524224 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FFE0 - 524256 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FFF0 - 524272 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FFF8 - 524280 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FFFC - 524284 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FFFE - 524286 OK 
[Flash] addr: 0x80000 - 524288 FAIL 
[Flash] addr: 0x7FFFF - 524287 OK 
---------------------- 
[Flash] Size: 0x7FFFF - 524287 
Fin. 
```

av1024, 2015-04-28:
@Links2004, confirmed.

Have the same result on both "valid" and "non-valid (aedf97abf)" builds using ESP-07 module

Links2004, 2015-04-28:
if i use the official flash (http://bbs.espressif.com/viewtopic.php?f=7&t=25) tool i get 

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)                                   
                                                                                
load 0x40100000, len 28896, room 16                                             
tail 0                                                                          
chksum 0xcc                                                                     
load 0x3ffe8000, len 1548, room 8                                               
tail 4                                                                          
chksum 0xa0                                                                     
load 0x3ffe8610, len 4884, room 4                                               
tail 0                                                                          
chksum 0x2d                                                                     
csum 0x2d                                                                       
R                                  
```
but the application dont run :(

esptool.py does the ```ets_main.c``` stuff

espressif says 
```
ets Jan  8 2013,rst cause:1, boot mode:(3,2)

ets_main.c
```
is ```0x0 address bin read error.```

but at the moment i out of ideas.

Links2004, 2015-04-28:
ok i try to use the boot.bin from to sdk in between the bootloader and the app.

good case: 
```
------------------ ESP8266 BOOTLOADER ------------------
 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6) 
 
load 0x40100000, len 1320, room 16  
tail 8 
chksum 0xb8 
load 0x3ffe8000, len 776, room 0  
tail 8 
chksum 0xd9 
load 0x3ffe8308, len 412, room 0  
tail 12 
chksum 0xb9 
csum 0xb9 
 
2nd boot version : 1.3(b3) 
  SPI Speed      : 40MHz 
  SPI Mode       : QIO 
  SPI Flash Size : 4Mbit 
jump to run user1 
 
Rüþ
------------------ ESP8266 APP ------------------
BOOTUP 
``` 

bad one:
```
------------------ ESP8266 BOOTLOADER ------------------
 
 ets Jan  8 2013,rst cause:2, boot mode:(3,7) 
 
load 0x40100000, len 1320, room 16  
tail 8 
chksum 0xb8 
load 0x3ffe8000, len 776, room 0  
tail 8 
chksum 0xd9 
load 0x3ffe8308, len 412, room 0  
tail 12 
chksum 0xb9 
csum 0xb9 
 
2nd boot version : 1.3(b3) 
  SPI Speed      : 40MHz 
  SPI Mode       : QIO 
  SPI Flash Size : 4Mbit 
jump to run user1 
 
error magic! 
backup boot failed. 
 
user code done 

------------------ ESP8266 APP -----------------
```

may it deepens on the size of the code.

good:
```
section                                  size         addr
.data                                    1332   1073643520
.rodata                                  1684   1073644864
.bss                                    41104   1073646552
.irom0.text                            169624   1076101120
.text                                   28896   1074790400
```

bad:
```
section                                                                                           size         addr
.data                                                                                             1548   1073643520
.rodata                                                                                           4884   1073645072
.bss                                                                                             41808   1073649960
.irom0.text                                                                                     216968   1076101120
.text                                                                                            28896   1074790400
```

igrr, 2015-04-28:
I ran your sketch on my module and the size is the same:
```
[SETUP] Flash ID: 0x1640C8
...
[Flash] Size: 0x7FFFF - 524287
```

Links2004, 2015-04-28:
the flash ID reference to this chip:  F25L32QA from ESMT http://www.esmt.com.tw/DB/manager/upload/F25L32Q.pdf
that say it is an 32 Mbit? but we only get access to 4Mbit? may limit in SDK.
but this is a other problem.

igrr, 2015-04-28:
SDK reads the flash chip size from the 0x00000 binary header (it's in the first dword). So since esptool generates binaries which have "4Mbit" hard-coded, SDK won't try to access flash above that.

igrr, 2015-04-28:
@Links2004 can you revert the latest SDK update to check if it is responsible for this issue? We need to narrow down on the exact change that leads to this error.

Links2004, 2015-04-28:
ok then i get ```section `.irom0.text' will not fit in region `irom0_0_seg'```
link: https://github.com/Links2004/Arduino/tree/esp8266_dev2

igrr, 2015-04-28:
Hmm, did you do a full revert?
I was at bb082c9 and did ```git revert 951ab8e```.
Compiling works okay.
Can you please try the following binaries? http://we.tl/dL6iz1S9Ye

Links2004, 2015-04-28:
i do a revert and then cherry pick all others. (result in my esp8266_dev2 branch)

files get also the ```ets_main.c```

cmd used:
```
esptool.py -p COM3 erase_flash
esptool.py -p COM3 -b 460800 write_flash 0x00000 HelloServer.cpp_00000.bin 0x40000 HelloServer.cpp_40000.bin
```
history:
working: 6a37b44
not working: 196992a

igrr, 2015-04-28:
Try reverting my smartconfig merge as well then.

Links2004, 2015-04-28:
it test different versions and remove different code. when i get the size from:

```
section                                                                                          size         addr
.data                                                                                            1340   1073643520
.rodata                                                                                          4552   1073644864
.bss                                                                                            41632   1073649416
.irom0.text                                                                                    197592   1076101120
.text                                                                                           28864   1074790400
```
to 
```
section                                                                                          size         addr
.data                                                                                            1500   1073643520
.rodata                                                                                          4420   1073645024
.bss                                                                                            40240   1073649448
.irom0.text                                                                                    192792   1076101120
.text                                                                                           28600   1074790400
```
for example all is working.

it look for me like it is not a code problem in first case.
it has more to do with the memory management.

to get this size info run: 
```
xtensa-lx106-elf-size -A <name_here>.elf
```

igrr, 2015-04-28:
I have removed my smartconfig change, this should also reduce the size of .irom0.text — does the latest trunk work for you?

Links2004, 2015-04-29:
no last trunk is not working.
```
.data                                                                                            1340   1073643520
.rodata                                                                                          4556   1073644864
.bss                                                                                            41624   1073649424
.irom0.text                                                                                    197720   1076101120
.text                                                                                           28896   1074790400
```

i try to create some simple dummy code to trigger the bug.
so we get an easy way to test it.

Links2004, 2015-04-29:
ok her is the bug trigger:

```cpp
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>

#include <OneWire.h>
#include <Wire.h>

extern "C" {
#include "user_interface.h"
}

const char* ssid     = "your-ssid";
const char* password = "your-password";

const char const_text[] = "STILL WORKING! const";
const char text[] = "STILL WORKING! ram";

#define SERIAL_OUT Serial1

//macro to create easy "functions" to fill the space costs 304Byte of .irom0.text per macro call
#define TEST_FUNCTION(a)                \
uint32_t test_##a##_dummy(void) {       \
    uint32_t test = 1;                  \
    test += millis();                   \
    test %= 123335;                     \
    if(test == 1) test *= 2;            \
    if(test == 3) test = 2;             \
    if(test == 3) test /= 2;            \
    if(test == 5) test += 2;            \
    if(test == 6) test -= a;            \
    if(test == 7) test %= 2;            \
    if(test == 9) test = 442;           \
    if(test == 51) test += 65462;       \
    if(test == 32) test = 2;            \
    if(test == 23) test /= 2;           \
    if(test == 35) test += 2;           \
    if(test == 56) test -= 2;           \
    if(test == 77) test %= 2;           \
    if(test == 99) test = 442;          \
    if(test == 521) test += 65462;      \
    test *= 2;                          \
    test += 2;                          \
    test -= 2;                          \
    test /= 2;                          \
    if(test == 35) test += 2;           \
    if(test == 56) test -= 2;           \
    if(test == 77) test %= a;           \
    if(test == 99) test = 442;          \
    if(test == a) test += 65462;        \
    test *= 2;                          \
    test += 2;                          \
    test -= 2;                          \
    test /= 2;                          \
    if(test == 35) test += 2;           \
    if(test == 56) test -= 2;           \
    if(test == 77) test %= 2;           \
    if(test == 99) test = a;            \
    if(test == 521) test += 65462;      \
    return test;                        \
}

TEST_FUNCTION(1)
TEST_FUNCTION(2)
TEST_FUNCTION(3)
TEST_FUNCTION(4)
TEST_FUNCTION(5)
TEST_FUNCTION(6)
TEST_FUNCTION(7)
TEST_FUNCTION(8)
TEST_FUNCTION(9)
TEST_FUNCTION(10)
TEST_FUNCTION(11)
TEST_FUNCTION(12)
TEST_FUNCTION(13)
TEST_FUNCTION(14)
TEST_FUNCTION(15)
TEST_FUNCTION(16)
TEST_FUNCTION(17)
TEST_FUNCTION(18)
TEST_FUNCTION(19)
TEST_FUNCTION(20)
TEST_FUNCTION(21)
TEST_FUNCTION(22)
TEST_FUNCTION(23)
TEST_FUNCTION(24)
TEST_FUNCTION(25)
TEST_FUNCTION(26)
TEST_FUNCTION(27)
TEST_FUNCTION(28)
TEST_FUNCTION(29)
TEST_FUNCTION(30)
TEST_FUNCTION(31)
TEST_FUNCTION(32)
TEST_FUNCTION(33)
TEST_FUNCTION(34)
TEST_FUNCTION(35)
TEST_FUNCTION(36)
TEST_FUNCTION(37)
TEST_FUNCTION(38)
TEST_FUNCTION(39)
TEST_FUNCTION(40)
TEST_FUNCTION(41)
TEST_FUNCTION(42)
TEST_FUNCTION(43)
TEST_FUNCTION(44)
TEST_FUNCTION(45)
TEST_FUNCTION(46)
TEST_FUNCTION(47)
TEST_FUNCTION(48)
TEST_FUNCTION(49)
TEST_FUNCTION(50)
TEST_FUNCTION(51)
TEST_FUNCTION(52)
TEST_FUNCTION(53)
TEST_FUNCTION(54)
TEST_FUNCTION(55)
TEST_FUNCTION(56)
TEST_FUNCTION(57)
TEST_FUNCTION(58)
//TEST_FUNCTION(59) //commend this in an you get "ets_main.c"

void setup()
{

    Serial.begin(115200);
    Serial1.begin(921600);
    Serial1.setDebugOutput(true);

    SERIAL_OUT.println();

    for(uint8_t t = 4; t > 0; t--) {
        SERIAL_OUT.print(F("[SETUP] BOOT WAIT "));
        SERIAL_OUT.print(t);
        SERIAL_OUT.println(F("..."));
        SERIAL_OUT.flush();
        delay(1000);
    }

    SERIAL_OUT.print(F("[SETUP] Flash ID: 0x"));
    SERIAL_OUT.println(spi_flash_get_id(), HEX);

    WiFi.softAP("ESP8266_BOOTED");
    WiFi.begin(ssid, password);

    SERIAL_OUT.println(text);
    SERIAL_OUT.println(const_text);

    for(uint16_t i = 1; i < 128; i++) {
        SERIAL_OUT.print("TEST_FUNCTION(");
        SERIAL_OUT.print(i);
        SERIAL_OUT.println(")");
    }

}

void loop()
{
}
```

MEMORY USAGE BAD CASE:
 ```
section                                                          size         addr
.data                                                            1340   1073643520
.rodata                                                          2616   1073644864
.bss                                                            41520   1073647480
.irom0.text                                                    196872   1076101120
.text                                                           28896   1074790400
 ```
WORKING CASE:
 ```
.data                                                            1340   1073643520
.rodata                                                          2616   1073644864
.bss                                                            41520   1073647480
.irom0.text                                                    196568   1076101120
.text                                                           28896   1074790400
 ```

tested with 69eb31a2cf8de218bbdfe1efebf3613efe507c7a on windows.

may you need to add on more of the macro calls, will depend on your compiler does other optimizations then the windows one.

Makuna, 2015-04-29:
I believe I am hitting the same issue.  Tried it with the latest bits and still repos.

(this requires https://github.com/Makuna/Task, into Documents\Arduino\libraries)

uncomment the define WONT_WORK to get it to fail.

```C++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WifiUdp.h>
#include <Task.h>

// #define WONT_WORK

WiFiUDP udpServer;

#define WLAN_SSID        "<SSID>"       
#define WLAN_PASS        "<PASSWORD>"
#define CoapMulticastIPv4Address 0xBB0100e0 // 224.0.1.187 reversed byte order
#define CoapPort 5683

void printUdpData(uint8_t* data, uint16_t count)
{
    Serial.print(F("Data: ("));
    Serial.print(count);
    Serial.print(F(")"));
    for (int i = 0; i < count; i++)
    {
        Serial.print(F(" 0x"));
        Serial.print(data[i], HEX);
    }
    Serial.println();
}

#if defined(WONT_WORK)
  void checkConnection(uint32_t deltaTime)
  {
      static wl_status_t s_lastStatus = WL_CONNECTED;
  
      wl_status_t status;
  
      status = (wl_status_t)WiFi.status();
      if (status != s_lastStatus)
      {
          if (status == WL_CONNECTED)
          {
              Serial.println(F("Regained Connection!"));
          }
          else
          {
              Serial.println(F("Lost Connection!"));
          }
          s_lastStatus = status;
      }
  }
  
  FunctionTask checkConnectionTask(checkConnection, MsToTaskTime(20000)); // every 20 seconds

  void ServiceUdp(uint32_t deltaTime)
  {
    int16_t result = 0;
  
    result = udpServer.parsePacket();
  
    if (result != 0)
    {
        Serial.println("\r\nReceived:\r\n");
        Serial.print("got udp packet from "); 
        Serial.print(udpServer.remoteIP()); 
        Serial.print(" to "); 
        Serial.println(udpServer.destinationIP());
        
        uint8_t data[512];
        if (udpServer.read(data, min(result, sizeof(data))) != result)
        {
          Serial.println("(not enought data)?");
        }
        else
        {
          printUdpData(data, result);
        }
    }
  }
  
  FunctionTask serviceUdpTask(ServiceUdp, MsToTaskTime(2)); // every 2 ms
#endif  // WONT_WORK

TaskManager taskManager;

void setup() {
  
  // give the human time to connect the serial monitor
  delay(10000);
  
  Serial.begin(115200);
  Serial.println("Starting things off");
  Serial.flush();
  
  wl_status_t status;
  WiFi.mode(WIFI_STA);
  status = (wl_status_t)WiFi.begin(WLAN_SSID, WLAN_PASS);

  if (status == WL_NO_SHIELD)
  {
      Serial.println(F("Radio Dead!"));
      return;
  }

  Serial.println(F("Getting Dhcp address..."));
  {
      uint8_t dhcpWait = 250; // 25 seconds

      while (dhcpWait > 0)
      {
          status = (wl_status_t)WiFi.status();
          if (status == WL_CONNECTED)
          {
              IPAddress address = WiFi.localIP();

              if (INADDR_NONE != address)
              {
                  Serial.println("ip !");

                  dhcpWait = 1;
              }
              else
              {
                  status = WL_DISCONNECTED;
              }
          }
          delay(100); 
          dhcpWait--;
      }
  }        
  if (status != WL_CONNECTED)
  {
      Serial.println(F("Failed to get DHCP address!"));
      return;
  }
  
  Serial.println(F("Got Address!"));
  
  // multicast connect
  if (!udpServer.beginMulticast(WiFi.localIP(),
      IPAddress(CoapMulticastIPv4Address),
      CoapPort))
  {
    Serial.println(F("No server Udp started"));
    return;
  }
  
  Serial.println(F("Server Udp started"));
  

    taskManager.Setup();
    
    
    
#if defined(WONT_WORK)      
    taskManager.StartTask(&checkConnectionTask);
    taskManager.StartTask(&serviceUdpTask);
#endif // WONT_WORK
    
    Serial.println(F("Running!"));
}

void loop() {
  taskManager.Loop();
}
```

Links2004, 2015-04-29:
can you run ```xtensa-lx106-elf-size -A <name_here>.elf``` for the the good and the bad case?

Links2004, 2015-04-29:
may it is an problem of the bootloader / flash tools.

if the last address of the flash is below 0x00070000 all is working fin.
if the we need to write to flash >= 0x00070000 it is broken.

may the memory is mapped like this for some strange reason:
0x00000000 - Flash start
0x0006FFFF - Flash end
0x00070000 - Flash start
0x000DFFFF - Flash end

this only an idea but it would explain the probe well.
because we override the magic bytes on beginning in this case.

looking for a way to do a full dump over 0x00000000 to 0x000DFFFF to confirm this.

Links2004, 2015-04-29:
ok no memory mapping problem.

but the address 0x00000 - 0x00FFF goes to value 0xFF if i flash the bad case files. 

dump cmd for esptool.py

```
esptool.py -p COM1 read_flash 0x00000 10240 dump.bin
```

if i change the flash order of the in the esptool.py from
```
esptool.py -p COM1 write_flash 0x00000 00000.bin 0x40000 40000.bin
```
to
```
esptool.py -p COM1 write_flash 0x40000 40000.bin 0x00000 00000.bin
```

all is working. 
really strange, can some one confirm this?

igrr, 2015-04-29:
Guys, please use xtensa-lx106-elf-size -A -x otherwise it's damn hard to
analyze those decimal sizes :)

igrr, 2015-04-29:
Ooh, I've read somewhere on the russian forum about a bug in the ROM function that erases (incorrect amount of) the flash prior to upload. This looks terribly like it. Let me find the reference...

igrr, 2015-04-29:
Here is the relevant part of esptool.py which supposedly works around the flash erase bug:
```
    """ Start downloading to Flash (performs an erase) """
    def flash_begin(self, _size, offset):
        old_tmo = self._port.timeout
        self._port.timeout = 10

        area_len = int(_size)
        sector_no = offset/4096;
        sector_num_per_block = 16;
        #total_sector_num = (0== (area_len%4096))? area_len/4096 :  1+(area_len/4096);
        if 0== (area_len%4096):
            total_sector_num = area_len/4096
        else:
            total_sector_num = 1+(area_len/4096)
        #check if erase area reach over block boundary
        head_sector_num = sector_num_per_block - (sector_no%sector_num_per_block);
        #head_sector_num = (head_sector_num>=total_sector_num)? total_sector_num : head_sector_num;
        if head_sector_num>=total_sector_num :
            head_sector_num = total_sector_num
        else:
            head_sector_num = head_sector_num

        if (total_sector_num - 2 * head_sector_num)> 0:
            size = (total_sector_num-head_sector_num)*4096
            print "head: ",head_sector_num,";total:",total_sector_num
            print "erase size : ",size
        else:
            size = int( math.ceil( total_sector_num/2.0) * 4096 )
            print "head:",head_sector_num,";total:",total_sector_num
            print "erase size :",size

        if self.command(ESPROM.ESP_FLASH_BEGIN,
                struct.pack('<IIII', size, 0x200, 0x400, offset))[1] != "\0\0":
            raise Exception('Failed to enter Flash download mode')
        self._port.timeout = old_tmo
```

@Links2004 can you please try with this fix applied to esptool.py?

Links2004, 2015-04-29:
i allways get 
```
    raise Exception('Failed to leave Flash mode')
Exception: Failed to leave Flash mode
```
at the end but the device is running

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6) 
 
load 0x40100000, len 28896, room 16  
tail 0 
chksum 0xa7 
load 0x3ffe8000, len 1340, room 8  
tail 4 
chksum 0xeb 
load 0x3ffe8540, len 4556, room 4  
tail 8 
chksum 0xd9 
csum 0xd9 
R°D
```

Links2004, 2015-04-29:
i change the handling in ```flash_finish```
to
```py
 def flash_finish(self, reboot = False):
        pkt = struct.pack('<I', int(not reboot))
        ret = self.command(ESPROM.ESP_FLASH_END, pkt)[1]
        if ret != "\0\0":
            #raise Exception('Failed to leave Flash mode')
            print 'Failed to leave Flash mode 0000 != %s ' % (ret.encode("hex")),
```

then i get 
```
Failed to leave Flash mode 0000 != 0106 
```
and all is working fine.

igrr, 2015-04-29:
It's a pity those changes from Espressif weren't mainlined into the esptool.py.
I'm applying this fix to esptool-ck, should be finished shortly.

igrr, 2015-04-29:
Short summary of the bug in SPIEraseArea (thanks to SpLab from esp8266.ru/forum):
```SPIEraseArea``` function in the esp8266 ROM has a bug which causes extra area to be erased.
If the address range to be erased crosses the block boundary, then extra ```head_sector_count``` sectors are erased.
If the address range doesn't cross the block boundary, then extra ```total_sector_count``` sectors are erased.
Where ```head_sector_count``` is the number of sectors from the start of erased area to the block boundary, and ```total_sector_count``` is the total number of sectors to be erased.
Sector size is 4096 bytes and block is 16 sectors.

igrr, 2015-04-29:
@Links2004 pushed the fix to https://github.com/igrr/esptool-ck
You can either wait for the CI build to finish (https://ci.appveyor.com/project/igrr/esptool-ck) or build it from source yourself. Kindly check either way.

Links2004, 2015-04-29:
final we have the esptool-ck and esptool.py working again :+1:

igrr, 2015-04-30:
Created an issue for esptool.py, I hope the fix will be integrated there as well:
https://github.com/themadinventor/esptool/issues/33

igrr, 2015-04-30:
updated esptool-ck for windows is available here:
https://ci.appveyor.com/api/buildjobs/u9lkd3d0s07ugct8/artifacts/esptool-0.4.2-5-g7b53d30-win32.zip

Links2004, 2015-04-30:
Info: last git version c9b5ddeebcb0938d2484d046d308211acd83f02d need 
https://ci.appveyor.com/api/buildjobs/xqi9oexaoh7xpidk/artifacts/esptool-0.4.3-win32.zip

igrr, 2015-04-30:
Even better, use the [release from github](https://github.com/igrr/esptool-ck/releases/download/0.4.3/esptool-0.4.3-win32.zip). Appveyor is known to be slow/unresponsive at times...

Makuna, 2015-04-30:
Just confirming, this fixes my problem.

closed 2015-05-05 by igrr