esp8266/Arduino #131: "Error compiling." while trying to use ArduinoJson lib

thereforeiam, 2015-04-27:
I'm trying to use [ArduinoJson] (https://github.com/bblanchon/ArduinoJson) with ESP8266 which seems like a good match :)
However when trying to compile a simple example : 

```c
// Copyright Benoit Blanchon 2014
// MIT License
//
// Arduino JSON library
// https://github.com/bblanchon/ArduinoJson

#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);

  StaticJsonBuffer<200> jsonBuffer;

  char json[] =
      "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

  const char* sensor = root["sensor"];
  long time = root["time"];
  double latitude = root["data"][0];
  double longitude = root["data"][1];

  Serial.println(sensor);
  Serial.println(time);
  Serial.println(latitude, 6);
  Serial.println(longitude, 6);
}

void loop() {
  // not used in this example
}
```

I get the following error message : 
```
~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -L~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/sdk//lib -L~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/sdk//ld -Teagle.app.v6.ld -o /var/folders/lb/xjmz955x16jgq0sd2f88wkzc0000gn/T/build2078761220582383494.tmp/JsonParserExample.cpp.elf -Wl,--start-group /var/folders/lb/xjmz955x16jgq0sd2f88wkzc0000gn/T/build2078761220582383494.tmp/JsonParserExample.cpp.o /var/folders/lb/xjmz955x16jgq0sd2f88wkzc0000gn/T/build2078761220582383494.tmp/ArduinoJson/ArduinoJson.cpp.o /var/folders/lb/xjmz955x16jgq0sd2f88wkzc0000gn/T/build2078761220582383494.tmp/core.a -lc -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -Wl,--end-group -L/var/folders/lb/xjmz955x16jgq0sd2f88wkzc0000gn/T/build2078761220582383494.tmp 
~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /var/folders/lb/xjmz955x16jgq0sd2f88wkzc0000gn/T/build2078761220582383494.tmp/JsonParserExample.cpp.elf section `.text' will not fit in region `iram1_0_seg'
~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o):(.literal+0x1c): undefined reference to `_sbrk_r'
~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o): In function `malloc_extend_top':
mallocr.c:(.text+0x5e): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x127): undefined reference to `_sbrk_r'
~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o): In function `_malloc_trim_r':
mallocr.c:(.text+0x366): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x38f): undefined reference to `_sbrk_r'
~/Downloads/Arduino-2.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o):mallocr.c:(.text+0x3a0): more undefined references to `_sbrk_r' follow
collect2: error: ld returned 1 exit status
Error compiling.
```
I'm using 1.6.1 release.

Links2004, 2015-04-27:
you need upgrade to the latest git version.
see same bug here:
https://github.com/esp8266/Arduino/issues/71#issuecomment-93452378

thereforeiam, 2015-04-28:
It did the trick but for the record, I also had to update `hardware/tools/esp8266/sdk` some header files changed...
Do you guys plan on doing a new release ?

igrr, 2015-04-29:
Currently the plan is to do the release when the issue with Windows
toolchain is resolved. Until the issue is resolved I can't package the core
separately from the IDE.

linagee, 2015-04-29:
I think the Espressif closed source libs already have some json decoding. You'd be wasting the space if you decide to upload your own Arduino library for the same purpose. (But... easier... I guess.)

closed 2015-05-19 by igrr

burner-, 2015-05-24:
I have now tried 4 hours to get this work. If I replace \Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697 contect with git version it get even more broke. If I download git version of ide and git version of esp library I most probably need compile tools. So how I update this without breaaking this more? Can you publish board manager url for nightly git version then it is possible to fix this kind critical issues just with cleaning ide folder and installing nightly build?

burner-, 2015-05-24:
igrr, do you have info or diff or something of commit where that is fixed so I try just pach that file without copying all files.