esp8266/Arduino #132: Question: Is there a way to get the Wifi signal strength ?

thereforeiam, 2015-04-27:
In order to be able to diagnose connection problem I was wondering if it was possible to get the signal strength via the WiFi API.
I realise this is not available as part of the standard arduino lib, but [GSM](http://www.arduino.cc/en/Reference/GSMScannerGetSignalStrength) has something equivalent

linagee, 2015-04-29:
Yes you can. You can do almost anything with this chip if you dive into their closed source libs. :-) (Its doing almost everything entirely in software.) I don't think you'd even need to go that far though. There appears to be some way that passive mode and the wifi scan are able to grab the RSSI. (You could probably even just use passive mode and only look at AP packets.)

duncan-a, 2015-04-29:
Does anyone know the method for this?

I'm intending to have a 'connected' LED and it would be very useful to have
an indication of signal strength to accompany that...

marvinroger, 2015-04-30:
As far as I know, there is currently no way in the Espressif SDK to retrieve the RSSI to the current connected AP. But as @linagee pointed out, it is possible looking at the sources to get the RSSI after a wifi scan, here is a working example:

```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>

const char* SSID = "YOUR_SSID";

// Return RSSI or 0 if target SSID not found
int32_t getRSSI(const char* target_ssid) {
  byte available_networks = WiFi.scanNetworks();

  for (int network = 0; network < available_networks; network++) {
    if (strcmp(WiFi.SSID(network), target_ssid) == 0) {
      return WiFi.RSSI(network);
    }
  }
  return 0;
}

void setup() {
  Serial.begin(115200);
}

void loop() {
  unsigned long before = millis();
  int32_t rssi = getRSSI(SSID);
  unsigned long after = millis();
  Serial.print("Signal strength: ");
  Serial.print(rssi);
  Serial.println("dB");

  Serial.print("Took ");
  Serial.print(after - before);
  Serial.println("ms");
  delay(1000);
}
```

Outputting
```
Signal strength: -60dB
Took 2133ms
Signal strength: -57dB
Took 2133ms
Signal strength: -57dB
Took 2128ms
Signal strength: -58dB
Took 2132ms
Signal strength: -62dB
Took 2132ms
Signal strength: -57dB
Took 2132ms

```

This is far from clean and ideal, as:

* It is not very bulletproof as it filters the networks by SSID, impossible to do now by mac address
* It is blocking for around 2 sec, which is a huge caveat for a single threaded system

But hey, it works!

duncan-a, 2015-05-01:
Marvin:
Thanks for that.
A two second blocking certainly less than ideal but a starting point is
always good to have at my level of comprehension.

Links2004, 2015-05-25:
thx to latests SDK ist possible now.
see pull #325

duncan-a, 2015-05-25:
Can somebody explain to those of us who are less 'techie' how to update the
ESP modules to the latest SDK so that we can use the new functionality???

Simple 'step by step' approach appreciated!!!

igrr, 2015-05-25:
@duncan-a 
When the update is released, you will just need to open the Board Manager and click "Install" button for the ESP8266 core.

duncan-a, 2015-05-25:
OK thanks - even I should be able to manage that!!!

Any idea when the update is likely to be released?

Tugao, 2015-08-11:
Heyyyy
I am new with esp8266
I would like to get the name of  ssid , the Signal strength and the mac address 
is it possible ?
I can´t found how to get a mac address ( ap )

thanks

Links2004, 2015-08-11:
you can use ```WiFi.getNetworkInfo()``` to get all at once or MAC only ```WiFi.BSSID()```

https://github.com/esp8266/Arduino/blob/d108a6ec30193a8e44eb9c72efc7a55853b54f09/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L291
https://github.com/esp8266/Arduino/blob/d108a6ec30193a8e44eb9c72efc7a55853b54f09/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L257
https://github.com/esp8266/Arduino/blob/d108a6ec30193a8e44eb9c72efc7a55853b54f09/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L264

Tugao, 2015-08-11:
ooo my thanks
but, howe can I use this labarys in my project 
I have others labrary liki this im my project, but
they dont work well

Links2004, 2015-08-11:
its part of the ESP8266 project, you already have it.
```cpp
#include <EPS8266WiFi.h>
```
and then use the functions

Tugao, 2015-08-11:
Yes, needed to update my old library
so Now i go this information
one more time
thanks for open my eyes

I thought this command just uas possible whit at command

Links2004, 2015-08-11:
at commands have noting to do with this project :)

goldbauch, 2015-10-03:
Maybe I get it wrong, but I'd have to pass about 8 arguments ot WiFi.getNetworkInfo(), so I'd have to scan the Network previously anyway?

Links2004, 2015-10-03:
you can take a look here:
https://github.com/esp8266/Arduino/blob/d108a6ec30193a8e44eb9c72efc7a55853b54f09/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L74

the arguments  are pointer where the information will be stored.
the first is the index.

basic:
run ```WiFi.scanNetworks``` and then use ```WiFi.getNetworkInfo``` for all the entrys

closed 2015-11-06 by Links2004

hb020, 2016-01-30:
For those looking for RSSI on the ESP8266's active WifI connection, This is one of the first topics that come up. But the explanations are about RSSI after a (rather time consuming) scan, after the remark "As far as I know, there is currently no way in the Espressif SDK to retrieve the RSSI to the current connected AP". Well, that is no longer true. One can get the RSSI of the active WiFi connection: WiFi.RSSI(). That's all that is needed.

duncan-a, 2016-01-30:
Are you implying that this is available with a new version of the Expressif
SDK?

If so, for those of us who are not super techie, how does one update the
SDK?

marvinroger, 2016-01-30:
The SDK is embedded, you don't have to download it yourself...

Just download the latest 2.1.0 (not sure it was implemented before), and when connected in STA do `WiFi.RSSI()` and you're good to go.

hb020, 2016-01-31:
It is also in 2.0 (stable/master)

gert3d, 2016-02-04:
@ marvinroger commented on 1 May 2015: I used the code for a ESP-07(without any  antenna) , (onboard antenna only), (external antenna only) and (both antenna's) but I do not see much difference in signal strength, all range between -76dBm and -87dBm, 2129ms.
However in a functional scetch recording NTP time I DO see a difference: no response without antenna's and a reliable response with whatever antenna configuration. So I wonder what is measures really.

Messaoudi0, 2016-04-05:
Hello everybody
Already thank you for all that information, I would like to know if we can get the signal to noise ratio and if possible, how?

Thank you

bitnapper, 2016-08-11:
Like @Messaoudi0 I am also interested in getting the signal to noise ratio. I need it to get the geolocation with the [mozilla location service api](https://location.services.mozilla.com/api).

tablatronix, 2016-12-23:
Anyone even find if noise or SNR are available ?

gicho, 2016-12-23:
This depends on if running in AP or station mode, but WiFi.RSSI() might be
what you look for.
Check here:
https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp

zetjede7, 2017-06-13:
if (strcmp(WiFi.SSID(network), target_ssid) == 0)  <-- cannot convert 'String' to 'const char*' for argument '1' to 'int strcmp(const char*, const char*)'

Can I ask something? I've just try the code, but there is a problem with the datatype.
Thanks. :)

marvinroger, 2017-06-13:
Actually, you don't need that anymore. Just call `WiFi.RSSI()`

byingtan, 2017-08-24:
I now going to carry out a project with esp8266 module 1 and arduino Uno to measure rssi received by the smartphone. I am confuse about the wiring between esp 8266 and arduino. Wish someone can give me some guidance.

devyte, 2017-08-24:
This is not the place to ask such questions. This is an issue tracker,
meant for tracking issues in the core libs. Please research, there are many
diagrams online.
If you still have questions, please ask at stackoverflow or esp8266.com.