esp8266/Arduino #133: Implement interface to: system_rtc_mem_write

linagee, 2015-04-27:
I propose making a "ESP.rtcMemWrite" that would pass things to "system_rtc_mem_write". Unless your program can start at any point and continue there (like, wake, get temp, send to internet, deepsleep), you need to write data to the RTC before deep sleep or else you have no way of knowing what you were doing when the processor resets (wakes) after deep sleep.

Write data into RTC, see if the data exists when you boot back up (will also need: ESP.rtcMemRead/system_rtc_mem_read), then continue from there.

The first parameter, des_addr, needs to be between 64 and 128. (Block index, blocks are 4 bytes long, only bytes 256 to 512 are usable by the user.) I propose encapsulating all of this into something easy to use. (Maybe just pass in a byte/number from 0 to 256 and translate that into a "RTC storage slot" or something.)

linagee, 2015-04-29:
I had a few long thoughts about this. Maybe you could just use EEPROM to store your state instead.... Just about the only difference would be - one state lasts past a power outage, the other would not.

Imroy, 2015-05-18:
Has anyone done anything on this yet?

linagee, 2015-05-18:
Imroy: Is there a specific application where using the EEPROM would not work? (I guess using RTC would use fewer EEPROM cycles, but that's about the most convincing argument I can think of to spend time on that.)

Imroy, 2015-05-20:
I figure if it's available it should be supported. Let programmers decide which one they want to use.

hairyhenderson, 2015-06-25:
:+1: - I'd like to use the RTC memory for short-term storage of data while the system is in deep sleep. In my application I'm using EEPROM for storing other stuff long-term and I'd rather not use it.

ardyesp, 2015-07-16:
Another upvote for RTC RAM read write feature.

MacroYau, 2016-04-01:
I have attempted to add RTC user memory access in PR #1836. Please feel free to comment on the API implementation.

closed 2016-06-23 by igrr