esp8266/Arduino #136: Prevent user from "bricking" their ESP8266 w/deepSleep

linagee, 2015-04-27:
When I was playing around with deepSleep (and before I found GPIO16 and RST needed to be hooked together), I was ESP.deepSleeping just a few clock ticks after booting.

*Maybe* there should be something that kind of prevents the user from shooting themselves in the foot like this. My module was essentially bricked. (Tooked a long time to get out of this mode, strangely, an older esp8266/Arduino release was able to successfully flash/"unbrick".) You really need to be at least a second or two "past the boot" before going to sleep. Maybe record the boot time (if we aren't already?) and when deepSleep is called, see if we are far enough past that to deepSleep. If not, wait a second first or something.)

codewise-nicolas, 2015-05-05:
Removing  power, enabling the programming pin (GPIO2 I believe) and powering up the ESP should make it boot into the bootloader and let you re-program it.
There should not be a state where you can completely "brick" the ESP with deepsleep as the bootloader can be made to kick in before your program runs.

linagee, 2015-05-06:
For some reason, it reflashed fine with the older release, but not the newer one. I think there might be  a difference of CTS/RTS or something?

codewise-nicolas, 2015-05-06:
The ESP8266 Arduino does not make use of the CTS/RTS lines. Just TX/RX, it expects you to toggle GPIO2 to enable the bootloader (Some people have come up with circuits to make use of the CTS line to do this, but in general they arent used/required)

duncan-a, 2015-05-06:
codewise-nicolas
Are you sure that "*ESP8266 Arduino does not make use of the CTS/RTS lines*"?
I know of at least two people who are doing it with no 'external circuitry'
except pullup/down resistors...

igrr, 2015-05-06:
The current version from git toggles DTR and RTS to manipulate GPIO0 and RST(or CHPD).

codewise-nicolas, 2015-05-06:
Ah good to know, i was not aware those changes had been done. Ill give them a try later.

So going back to the original problem, if you were relying on the DTR/RTS functionality to re-program your chip then it is possible that you were in a situation that would not get back into the bootloader.

duncan-a, 2015-05-07:
Which version of the EPS? Is GPIO0 being pulled high via a resistor? Is
Reset being pulled high via resistor? Etc...?

Software is capable of a lot, but not if the hardware isn't set up
correctly.

Just a thought...

tytower, 2015-05-23:
Using this code and connecting DTR to RST I get it going to sleep but it gets in a reset loop and does not start properly again . I must physically remove DTR from RST.
@igrr I still can't get the I2C towork at all ?

*
#include <SPI.h> // Required to use BMP180
#include <DHT.h>
#include <stdio.h>
#include <SFE_BMP180.h>
#include <Wire.h>
#include <ESP8266WiFi.h>

#define DHT_PIN 2    //GPIO5   DHT22 for my sensorRHT03-others in library
#define TIMEOUT 3000 //3 second timout
#define SPARKFUN_UPDATE_TIME 20000         //Update SparkFun data server every 60000 ms (1 minute).
#define debug
#define ALTITUDE 750.0 // Altitude of My HQ in  meters
unsigned long timer1 = 0;
unsigned long timer2 = 0;
char server[] = "data.sparkfun.com";    // name address for data.sparkFun (using DNS)
// Set the static IP address to use if the DHCP fails to assign

const String publicKey = "RMzp9ANyqzfjw4drzNnZ";
const String privateKey = "lzrnDbjBprioBgD6Mlvd";
const byte NUM_FIELDS = 4;
const String fieldNames[NUM_FIELDS] = {"humidity" , "pressure2" , "temp" , "temp2"};
String fieldData[NUM_FIELDS];

//Character array to hold results
char results[4];
//const int requestInterval = 20000;
long lastAttemptTime = 0; 
float humidity=0;
float temp=0;
long pressure2 = 0;
float temp2;
const int dataSDA = 4; //BMP180 SDA 
const int clockSCL = 5;//BMP180 SCL 

double T,P,p0,a;
//float temp2 = 0.0;

long lastReadingTime = 0;

SFE_BMP180 pressure;

DHT dht(DHT_PIN,DHT22,15);
WiFiClient client;
// Time to sleep (in seconds):
const int sleepTimeS = 30;

//////////////////////////////////////////////////////////////////////////////////

void setup(){ 

const char* ssid = "ME-HOME";
const char* password = "";
// Igrr-Wire.begin(0, 2); on ESP-01, else they default to pins 4(SDA) and 5(SCL)..

Wire.begin(dataSDA, clockSCL );//see above definiions
//I2c.setSpeed(1);
Wire.setClock(100000);
// Create an instance of the server
// specify the port to listen on as an argument
WiFiServer server(80);

 #ifdef debug
 Serial.begin(115200);
  delay(10);

  // Connect to WiFi network
[... 206 lines cut ...]
    timer1 = millis(); //Update timer1 with the current time in miliseconds since startup.
    delay(500);
    postData(); //Send data to sparkfun data server.
    #ifdef debug
    Serial.println("Posting by timer !");
    
    Serial.println(fieldData[0] );
    Serial.println(fieldData[1] );
    Serial.println(fieldData[2] );
    Serial.println(fieldData[3] );
    delay(500);

    
  // deepSleep time is defined in microseconds. Multiply
  // seconds by 1e6 
  
  Serial.println(" Going to Sleep !" );
  #endif
  
  ESP.deepSleep(sleepTimeS * 1000000);
  
  }
 }

*

closed 2015-11-06 by Links2004

lumbric, 2017-02-18:
In case you bricked your ESP8266 / D1 by putting it to sleep immediately in main.py [follow these instructions](https://forum.wemos.cc/topic/798/i-thought-i-had-bricked-my-wemos-d1-mini):

- connect GPIO 0 (D3 on the D1 mini WEMOS) to GND
- connect GPIO 15 (D8 on the D1 mini WEMOS) to GND
- connect GPIO 2 (D4 on the D1 mini WEMOS) to 3.3V
- Connect with USB and run in a terminal  `esptool.py -p /dev/<YOUR TTY> -b 9600 erase_flash`
- your main.py should be deleted now, you can start like with a fresh ESP8266 (upload firmware etc.)