esp8266/Arduino #137: WiFi.softAPConfig compile error

tontito, 2015-04-27:
IPAddress Ip = (192, 168, 3, 1);
IPAddress NMask = (255, 255, 255, 0);
WiFi.softAPConfig(Ip, Ip, NMask);
WiFi.softAP("ESP8266_AP", "12345678", 6);

compile error

C:\Users\VOLKI\Desktop\Arduino\Build\graphicstest.cpp.o: In function `GUIHazirla()':
graphicstest.cpp:(.text+0x37c): undefined reference to `ESP8266WiFiClass::softAPConfig(IPAddress, IPAddress, IPAddress)'
C:\Users\VOLKI\Desktop\Arduino\Build\graphicstest.cpp.o: In function `EepromToString(int, int)':
graphicstest.cpp:(.text+0x48a): undefined reference to `ESP8266WiFiClass::softAPConfig(IPAddress, IPAddress, IPAddress)'
collect2.exe: error: ld returned 1 exit status
Derleme sırasında hata oluştu.

igrr, 2015-04-27:
I'm sorry, but I'm not able to reproduce it. Here's my sketch:
```
#include "ESP8266WiFi.h"
void setup() {
  IPAddress Ip = (192, 168, 3, 1);
  IPAddress NMask = (255, 255, 255, 0);
  WiFi.softAPConfig(Ip, Ip, NMask);
  WiFi.softAP("ESP8266_AP", "12345678", 6);
}

void loop() {
}
```

Here's the compiler output:
```
Using library ESP8266WiFi in folder: /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi 

/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/SoftAPConfigTest.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/SoftAPConfigTest.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/ESP8266WiFi/ESP8266WiFi.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/ESP8266WiFi/WiFiClient.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiServer.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/ESP8266WiFi/WiFiServer.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUIN [... cut ...] tAPConfigTest.cpp.elf -bo /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/SoftAPConfigTest.cpp_00000.bin -bs .text -bs .data -bs .rodata -bc -ec -eo /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/SoftAPConfigTest.cpp.elf -es .irom0.text /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build2978641421408012178.tmp/SoftAPConfigTest.cpp_40000.bin -ec 

Sketch uses 226,156 bytes (43%) of program storage space. Maximum is 524,288 bytes.
```

igrr, 2015-05-04:
@tontito can you please check again if this issue may be closed?

tontito, 2015-05-06:
check and happy. no problem. :)

closed 2015-05-06 by tontito