esp8266/Arduino #139: Unable to read correct register values of I2C slave device

hallard, 2015-04-28:
Hi,
I'm the author of the [TH02](https://github.com/hallard/TH02) library sensor for arduino and I tried to push compatibility to my lib to be able to use it with ESP8266. Unfortunatly I didn't succeeded to do this. Case [#65](https://github.com/esp8266/Arduino/issues/65) on Wire.endTransmission if one of the starting problem (that also prevent any arduino I2C scanner to work out of the box). So I tried the scanner by changing the return value of .endTransmission to 0 if slave acked (function i2c_master_write_to in i2c.cpp) but with no luck. I never been able to see my device. Going deeper into my lib I removed all check from wire return values just to see if I can have consistent registers values in return of reading `Wire.requestFrom( 0x40, 1)`, but again with no luck.

I know some of you got SSD1306 oled working, but these device works only in write mode (no data is pulled from OLED controller) so this is a bit different than trying to read device information.. 

So my main concern, if someone succeeded to read a value from I2C device, would it be possible to have a sample code ?

BTW the i2c [scanner](http://www.esp8266.com/viewtopic.php?f=19&t=1049#p6198) of NodeMCU is working fine so my wiring is correct and my board is working.

GerryKeely, 2015-04-28:
Hi Hallard
I got the DS1307 program found here http://tronixstuff.com/2014/12/01/tutorial-using-ds1307-and-ds3231-real-time-clock-modules-with-arduino/ to work. The only addition I made was to include 

 Wire.pins(12,13); // 12 =sda  13 =sck
  Wire.setClock(100000);

under setup()

Gerry

hallard, 2015-04-28:
@GerryKeely 
Thanks, I've got one DS1307 module, I just tried. Wired to GPIO0/GPIO2 (with GPIO12/13 it was even not detected) . First checked I2C device with nodemcu firmware and i2c-scan.lua test, this told me device seen at address 0x68 => Ok seen correct
[image]

Then put the Arduino sketch you mentioned above changed according my wiring in setup to
```
Wire.pins(0,2); // 0 =sda 2 =sck
Wire.setClock(100000);
```
Then compiled with Arduino IDE and launched the sketch
```
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
0:00:00 1/1/0 Day of week: Sunday
```
So it's not working on my side. I need to clone back the repo just in case.

GerryKeely, 2015-04-28:
Hi
I presume you have set up the time using the function
setDS3231time(); as described in the article ?

Gerry

hallard, 2015-04-28:
@GerryKeely 
To be honest I just copied/pasted the code from tronix stuff , adding in setup the two lines.
```
Wire.pins(0,2); // 0 =sda 2 =sck
Wire.setClock(100000);
```

knowing time will not be correct but I thought that the second counter should be incremented every second, but this assumption was false (may be it need a 1st write to start chip clock).

So I just added the line you proposed, know what ? works fine ;-)

I now need to check working with I2C lib interaction between DS1307 and my th02 lib.

Thank you very much for your help

Charles

igrr, 2015-04-28:
New i2c driver and the fix for the return codes is comming up, should be fixed soon.

hallard, 2015-04-28:
igrr,

Thanks for the head up. In the meantime I analyzed why my TH02 is working with Arduino and why not with ESP8266. So I graphed with logic analyzer
This screen is taken from com with Arduino, reading 2 bytes starting register 1, so bus is
TH02 address + set register address 1, then repeated start + TH02 address and reading 2 bytes
[image]
This one is taken from ESP8266 reading 4 bytes starting register 0, so bus is
TH02 address + set register address 0, then repeated start + TH02 address and reading 4 bytes
[image]
And as you can see, it seems there is twice the repeated start, that seems to make the TH02 to issue a NACK.

The code is as follow
```
void setup() 
{
  Serial.begin(115200);
  Serial.println("Setup done");
  Wire.begin();
  // SDA GIPO2 / SCL GPIO0
  Wire.pins(2, 0); 
  Wire.setClock(50000); 
}
void loop() 
{
  Wire.beginTransmission(0x40);   
  Wire.write(0);
  Wire.endTransmission(false);   
  Wire.requestFrom(0x40, 4);
  Serial.print("requestFrom()=");
  Serial.print(Wire.available());
  Serial.print(" TH02 Status=0x");
  printhex(Wire.read());
  Serial.print(" datah=0x");
  printhex(Wire.read());
  Serial.print(" datal=0x");
  printhex(Wire.read());
  Serial.print(" config=0x");
  printhex(Wire.read());
  Serial.println();
}
```
And yes if you noticed, I setup the frequency to 50KHz but Analyzer show me frequency of 100KHz so I checked the compiler command line just in case, it's really 80MHz by the `-DF_CPU=80000000L`

igrr, 2015-04-29:
The point where you have drawn an arrow doesn't look like a repeated start because SCK is low. The logic analyzer correctly marked the next falling transition as a repeated start — SCK is high there.
The frequencies are off, yes. There's an issue open regarding frequency calibration.

igrr, 2015-04-29:
Also there is a NAK during address setup (first write of 0x80). Looks like the device hasn't replied on address 0x40... Is 0x40 it's 7-bit address?

hallard, 2015-04-29:
Ivan,
Yeah I saw that after posting. I think 0x40 is the good address, well it's the one I use on Arduino when it's working. I'm not an I2C protocol specialist, by in some case I need to dig into when problems occurs. Strangely this is not the first time I have problem with this device [see here](http://forum.arduino.cc/index.php?topic=290483.0)
What I do not understand looking at the both analyzer graphs is why on the first device answer with ACK and on the other with NACK, clock pulses and data are exactly the same ;-(
may be on the 2nd it's not the 1st com and the two pulses at the end (where should I have repeated start) put the device into a "bad mode" I need to investigate on this point.

igrr, 2015-05-02:
@hallard if you are comfortable building the IDE from source, or can run the Windows version which is built by CI server, you may want to try the new Wire library.

hallard, 2015-05-02:
@igrr 
Yeah, thank for the head up, I saw yesterday (well in fact very late in the night for me) that i2C has changed, I pulled the repo and compiled with no pb (I saw you changed build.xml to point on new sourceforge link for the launch4j, I wanted to open a ticket on this but now fixed before)
Once IDE compiled fine, I was unable to compile my sketch with error at link stage. I think the repo modification was not ended, I will pull it again and test ASAP and post results here
thanks

```
D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -LD:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk//lib -LD:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk//ld -Teagle.app.v6.ld -o C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp/test-i2c.cpp.elf -Wl,--start-group C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\test-i2c.cpp.o C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\ESP8266WiFi\ESP8266WiFi.cpp.o C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\ESP8266WiFi\WiFiClient.cpp.o C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\ESP8266WiFi\WiFiServer.cpp.o C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\ESP8266WiFi\WiFiUdp.cpp.o C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp/core.a -lm -lc -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -Wl,--end-group -LC:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp 
C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o: In function `TwoWire::write(unsigned char)':
Wire.cpp:(.text+0x130): undefined reference to `twi_init(unsigned char, unsigned char)'
Wire.cpp:(.text+0x143): undefined reference to `twi_init(unsigned char, unsigned char)'
C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o: In function `TwoWire::begin()':
Wire.cpp:(.text+0x190): undefined reference to `twi_setClock(unsigned int)'
C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o: In function `TwoWire::begin(unsigned char)':
Wire.cpp:(.text+0x19b): undefined reference to `twi_setClock(unsigned int)'
C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o: In function `TwoWire::begin(int)':
Wire.cpp:(.text+0x1a8): undefined reference to `twi_readFrom(unsigned char, unsigned char*, unsigned int, unsigned char)'
C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o: In function `TwoWire::setClock(unsigned int)':
Wire.cpp:(.text+0x1ca): undefined reference to `twi_readFrom(unsigned char, unsigned char*, unsigned int, unsigned char)'
C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o: In function `TwoWire::requestFrom(int, int, int)':
Wire.cpp:(.text+0x29c): undefined reference to `twi_writeTo(unsigned char, unsigned char*, unsigned int, unsigned char)'
C:\Users\CHARLE~1\AppData\Local\Temp\build3518595953392153004.tmp\Wire\Wire.cpp.o: In function `TwoWire::beginTransmission(unsigned char)':
Wire.cpp:(.text+0x2ba): undefined reference to `twi_writeTo(unsigned char, unsigned char*, unsigned int, unsigned char)'
collect2.exe: error: ld returned 1 exit status
Error compiling.
```

hallard, 2015-05-02:
Well, I took some time and pulled repo then compiled from scratch IDE (Win 8.1 X64). Here the results I have :
* the 1st thing that is new is that IDE was unable to launch esptool.exe to upload sketch due to missing execute permissions on file esptool.exe, just manually added them and it was fixed (strange, it's the 1st time it happens after build).
* 2nd the classical libiconv2.dll and libiconv-2.dll problem missing (fixed copying them to root ide folder)
* the last (the one I compiled for) is the I2C. As soon as I add include "Wire.h" (or <Wire.h>) it does not compile (well in fact it compile but does not link).

here is the test sketch
```
#include <Wire.h>

// manage GPIO16 Led and multiplexed Push button on node MCU boards
uint8_t key_led( uint8_t level)
{
  uint8_t temp;
  digitalWrite(16, 1);
  pinMode(16, INPUT);
  temp = digitalRead(16);
  pinMode(16, OUTPUT);
  digitalWrite(16, level);
  return temp;
}

void setup() 
{
  Serial.begin(115200);
  Serial.println("Setup done");
}

void loop() 
{
  // manage Blinking LED
  key_led((millis() % 800) < 600);
}
```
here is the the full build results, if I remove the include "Wire.h", building and uploading are working fine.
```

Using library Wire in folder: D:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\libraries\Wire 

D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -ID:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -ID:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266 -ID:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\variants\generic -ID:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\libraries\Wire C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\test.cpp -o C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\test.cpp.o 
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\Wire\Wire.cpp.o
D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -ID:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk//include -c -g -x assembler-with-cpp -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -ID:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266 -ID:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\variants\generic D:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266\cont.S -o C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\cont.S.o 
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\cont_util.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_noniso.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_timer.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_wiring.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_wiring_analog.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_wiring_digital.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_wiring_pulse.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_wiring_pwm.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_wiring_shift.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\libc_replacements.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\si2c.c.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\abi.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\core_esp8266_main.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\Esp.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\HardwareSerial.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\IPAddress.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\Print.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\Stream.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\Tone.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\WMath.cpp.o
Using previously compiled file: C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\WString.cpp.o
[... cut ...]
C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\Wire\Wire.cpp.o: In function `TwoWire::requestFrom(int, int, int)':
Wire.cpp:(.text+0x29c): undefined reference to `twi_writeTo(unsigned char, unsigned char*, unsigned int, unsigned char)'
C:\Users\CHARLE~1\AppData\Local\Temp\build453996467878338922.tmp\Wire\Wire.cpp.o: In function `TwoWire::beginTransmission(unsigned char)':
Wire.cpp:(.text+0x2ba): undefined reference to `twi_writeTo(unsigned char, unsigned char*, unsigned int, unsigned char)'
collect2.exe: error: ld returned 1 exit status
Error compiling.
```

igrr, 2015-05-02:
>    include "Wire.h" (or ) it does not compile (well in fact it compile but

Okay, thanks, will take a look.

igrr, 2015-05-04:
@hallard can you please try again? This should be fixed now.

hallard, 2015-05-05:
@igrr,

Thanks for the head up, it does now compile. 
In the meantime I succeeded to get it working with old I2C implementation (in core with i2c.h and i2c.c not the new ones si2c.c or twi.c). So my investigation are : 
* old I2C => Ok working 
* latest I2C => does not work, but I don't think it's related to you I2C code, this TH02 device is so tricky that I'm pretty sure it's not 100% compliant to I2C protocol (datasheet says you should disconnect VCC if you have other devices on same I2C bus, whoo, suprising !!!!) Also I'm not sure that data on SDA line are not sampled on falling edge instead of rising edge of CLK. That's why it does not work and it worked on old implementation just because of "luck". Anyway I spent too much time on this "random work" device, I need to pass to other sensor.

So I tried the new implementation with other I2C devices, One DS1307 RTC and one temp/hum sensor HTU21D, both works fine, so from my point of view all is working fine now (except this tricky TH02).

Thank you for your help, I post picture of old I2C implementation working with TH02 and one with the new not working is case someone is interested.

Old i2c
[image]

New I2C (see the latest SDA high is too short (0.8us) for the TH02
[image]

jorgegarciadev, 2016-01-24:
@hallard Hey hallard, I'm having some difficulties making a DS1307 work with i2c. Could you share the module or circuit you're using, please?

devyte, 2017-10-20:
I have to wonder about clock stretching.
Closing due to age and #3655 .

closed 2017-10-20 by devyte