esp8266/Arduino #141: Cannot Upload (flash_download_begin failed)

circuitFlower, 2015-04-28:
Sorry if this is a repost, I've been following the issues for a while, and just can't seem to get this up and running. I've got this Arduino-esp IDE installed on Mac Yosemite. I'm using a serial to usb adapter from Adafruit https://www.adafruit.com/product/954... I also have an FTDI adapter. the serial adapter puts out 5v, so I'm using a voltage divider in between to knock the voltage down to 3.3v. 

When I run the Arduino-esp IDE, I'm getting an error :

Sketch uses 208,276 bytes (39%) of program storage space. Maximum is 524,288 bytes.
/Users/aervin/Documents/Arduino/hardware/esp8266com/esp8266/tools/macosx/esptool -vv -cd none -cb 115200 -cp /dev/tty.usbserial -ca 0x00000 -cf /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build6296413658005645577.tmp/mDNS_Web_Server_home.cpp_00000.bin -ca 0x40000 -cf /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build6296413658005645577.tmp/mDNS_Web_Server_home.cpp_40000.bin 
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to none
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/tty.usbserial
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build6296413658005645577.tmp/mDNS_Web_Server_home.cpp_00000.bin success
opening port /dev/tty.usbserial at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_open
Uploading 37088 bytes from /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build6296413658005645577.tmp/mDNS_Web_Server_home.cpp_00000.bin to flash at 0x00000000
	erasing flash
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	setting timeout 5000
	setting character timeout 50
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm cmd: receiving 2 bytes of data
warning: espcomm cmd: no final C0
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader

I read that there was an update to the platform.txt file, so I took a look. There's a closed issue, suggesting that this line be changed from 

tools.pyesptool.program.pattern=python "{path}/{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin"

to 

tools.pyesptool.program.pattern="{path}/{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin"

my platform.txt file does not have this line at all, nor does it reference pyesptool. do I have the wrong esptool installed? my platform.txt calls this 

tools.esptool.upload.pattern="{path}/{cmd}" {upload.verbose} -cd {upload.resetmethod} -cb {upload.speed} -cp {serial.port} -ca 0x00000 -cf "{build.path}/{build.project_name}_00000.bin" -ca 0x40000 -cf "{build.path}/{build.project_name}_40000.bin"

which is a reference to the upload function, perhaps it's looking for the wrong esptool?

What else might be throwing the 

warning: espcomm cmd: no final C0
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader

If I run esptool.py from command line, I can do a few things, like read_mac, though some of the other commands return errors, probably because I don't know how to use them.

circuitFlower, 2015-04-28:
UPDATE: also changed title to reflect new issue.

I removed my the original Arduino software and the Arduino-esp software, and built and installed this: https://github.com/unreality/Arduino/tree/e59d27654c0ae523666623d4cb5bc2a07802b1b4 by @unreality. It's supposed to include the pyesptool, rather than the esptool. Now I'm getting a null pointer exception (same problem I ran into when trying to run this on Linux). 

Sketch uses 210,612 bytes (40%) of program storage space. Maximum is 524,288 bytes.
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:246)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:75)
	at processing.app.debug.Compiler.upload(Compiler.java:157)
	at processing.app.Sketch.upload(Sketch.java:1259)
	at processing.app.Sketch.exportApplet(Sketch.java:1201)
	at processing.app.Sketch.exportApplet(Sketch.java:1173)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2432)
	at java.lang.Thread.run(Thread.java:745)

Is this because there are basic system tools missing, now that I uninstalled the original Arduino?

circuitFlower, 2015-04-28:
UPDATE: I switched the programmer option to pyesptool option, and then received this error: 

Sketch uses 210,612 bytes (40%) of program storage space. Maximum is 524,288 bytes.
python /Applications/Arduino-esp/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool.py --port /dev/tty.usbserial write_flash 0x00000 /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build2390947998494800588.tmp/mDNS_Web_Server.cpp_00000.bin 0x40000 /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build2390947998494800588.tmp/mDNS_Web_Server.cpp_40000.bin 
Connecting...
Traceback (most recent call last):
  File "/Applications/Arduino-esp/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool.py", line 473, in <module>
    esp.connect()
  File "/Applications/Arduino-esp/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool.py", line 151, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect

I also tried connecting to it through my terminal window, using the esptool, and now that won't work. It's not supposed to use the esptool here, correct? it's supposed to use pyesptool?

I also tried switching to my Linux Virtual Machine, and received this error: 
[image]

igrr, 2015-04-28:
The last screenshot looks like a successful upload to me.

igrr, 2015-04-28:
Also if esptool.py doesn't work for you, but the upload is successful from a Linux VM, i sugges you check the drivers on OS X. For example the FTDI driver bundled with OS X is known to have some issues. Installing the FTDI-supplied driver from their website resolved upload issues for me.
Same goes for the Prolific-based usb-serial converters...

duncan-a, 2015-04-28:
Hey, that's what we've been seeing with the Windows version - see here
<https://github.com/esp8266/Arduino/issues/95#issuecomment-96399149>

circuitFlower, 2015-04-28:
@igrr hmm, well, I didn't think that was correct, because I was uploading the mDNS_Web_Server example, and when I navigated to http://esp8266.local, I didn't get anything. 

I did get it to erase the flash at one point, then it just hung, and now I get the below error on both the Linux VM, and Yosemite. I can no longer connect to the board using esptool.py in my Unix terminal either. same error. @duncan-a is this what you were referring to? I tried to erase flash using the terminal, and got the same failure to connect error. I'm looking into it now. 

Sketch uses 210,612 bytes (40%) of program storage space. Maximum is 524,288 bytes.
python /Applications/Arduino-esp/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool.py --port /dev/tty.usbserial write_flash 0x00000 /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build6955321294746670421.tmp/mDNS_Web_Server.cpp_00000.bin 0x40000 /var/folders/yy/dqkcs2dj4lq8s4vk0h10kwmnvspy9l/T/build6955321294746670421.tmp/mDNS_Web_Server.cpp_40000.bin 
Connecting...
Traceback (most recent call last):
  File "/Applications/Arduino-esp/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool.py", line 473, in <module>
    esp.connect()
  File "/Applications/Arduino-esp/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool.py", line 151, in connect
    raise Exception('Failed to connect')
Exception: Failed to connect

igrr, 2015-04-28:
Re mDNS not working for you: this was fixed in a commit yesterday, so perhaps you can pull the updated code and check again.

duncan-a, 2015-04-28:
Amanda

Yes, just what we've been getting - appears to upload correctly but refuses
to run...

circuitFlower, 2015-04-28:
@igrr I can't upload anything now, so I can't test a new sketch. for the record though, I dumped the whole thing and reinstalled @unreality 's example with the pyesptool update. so if the new mDNS example was included in that, then yes, I got it. 

@duncan-a is there someplace I can follow fixes for this?

I noticed the Linux version is up and running again, cool. I was just gonna say this is super frustrating, because my friend and I are working on this, and he has this up and running on his Linux setup, so I'm dragging my feet here.

duncan-a, 2015-04-28:
*Amanda*

See here: https://github.com/esp8266/Arduino/issues/95#issuecomment-96399149

I haven't got the 'fix' to work, but things did spring back to life briefly
today. I'm not particularly techie, as will become clear when you read the
thread...

Buena suerte - good luck

circuitFlower, 2015-04-28:
oh, thanks @duncan-a. I tried that already, but it did not bring things back to life. thanks though. I'm still stuck on the can't connect issue, can't even connect with the esptool

duncan-a, 2015-04-29:
*Amanda*
Is the ESP in the correct mode? Earlier in that topic there's a thing about
the header message that the ESP puts out as it boots - that may give you
some clues.

It's all a bit beyond me...

onkelfunny, 2015-04-29:
I have the same issue with my mac and ck-esptool:

espcomm cmd: receiving 2 bytes of data
warning: espcomm cmd: no final C0
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader

I have tried many 5 esp-modules. I can only flash this with the esptool.py

linagee, 2015-04-29:
Windows - works very nicely (CH340, no extra drivers needed), except for the 350MB download and 1GB extracted directory size.
Linux - works very nicey (CH340, no extra drivers needed), it was only 78MB or so download
Mac - I wouldn't wish this on my worst enemies. Struggled for hours with CH340 drivers, found I had to go to a .cn site, download them, do some terminal thing for yosemite (Mac people hate having to go to terminal, it would be hard for me to ask a friend to do it), found my USB TTL didn't want to work with the USB extension cable I had, found that Arduino/esp8266 didn't want to flash for all of the reasons listed above, esptool problems. Still got it to work using lots of workarounds from other bug posts. But by that time, I was 5+ hours into the problem. Not a good overall experience. ;-) I'd just tell people to fire up a Linux VM, skip the Mac uglyness.

circuitFlower, 2015-04-30:
Hey, so I got it working. I'm still working through the specific code, but I got the mDNS example to work. I switched to a brand new esp8266 board. I'm thinking when I ran the pyesptool version, it erased my flash to 100%, then hung, and maybe got stuck somewhere? I'm not sure if my instincts are accurate, but maybe it didn't actually erase the whole thing, and when the IDE was looking for certain header info, it either couldn't find it, or couldn't write to it because something else was there, maybe? Anyway, I've got it up and running on my Ubuntu VM, just working out the bugs in my code now.

igrr, 2015-05-06:
The latest version of esptool (0.4.3) fixes a bug which could cause the ```espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed``` error. Can anybody verify that?

sticilface, 2015-05-06:
I was trying that. But then get the build failed with the i2c

igrr, 2015-05-06:
The i2c.cpp file shouldn't be there — please check your git status

gerardwr, 2015-05-06:
I had many many months of problem-free uploading. I started getting many upload problems some weeks ago after installing new updates, like:
```
espcomm cmd: receiving 2 bytes of data
warning: espcomm cmd: no final C0
```
On May 2nd I updated again, and the upload problems disappeared, without changes to my hardware.

Wouldn't be surprised that the fix of esptool solved the problems.

BTW : I'm on Mac 10.10.2.

onkelfunny, 2015-05-06:
if i use this connections: https://github.com/esp8266/Arduino/blob/esp8266/docs/ESP01_connect.jpg with a FTDI-TTL-Board, i can flash with the current version.

with my nodemcu board (the yellow) i can't flash

I have OSX 10.0.3

Links2004, 2015-05-06:
nodemcu is not an ESP01 it has more pins.
you need pull down GPIO15 also (no ESP01 this is done on the board)

igrr, 2015-05-06:
I think the issue with nodemcu board is due to the different behavior of
ch340 usb-ttl driver in terms of serial timeouts. If I had a nodemcu board
I would fix that, but without it it's kinda nontrivial.

onkelfunny, 2015-05-06:
do you have any idea why i can flash with esptool.py? it is slow but it works with the ch340

igrr, 2015-05-06:
I suspect pyserial (used be esptool.py) doesn't have the bugs in serial
timeouts handling which are present in my code.

sticilface, 2015-05-06:
Using the latest version.  

I got what looked like the beginnings of a successful upload.. but then it failed. and failed every time after.  
here is one of the outputs... This was one the nodemcu board, but it does it for an FDTI adapter too.

``````

Sketch uses 237,464 bytes (45%) of program storage space. Maximum is 524,288 bytes.
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool -vv -cd ck -cb 115200 -cp /dev/tty.wchusbserialfa130 -ca 0x00000 -cf /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build760706363100022971.tmp/WiFiWebServer.cpp_00000.bin -ca 0x10000 -cf /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build760706363100022971.tmp/WiFiWebServer.cpp_10000.bin 
esptool v0.4.3 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/tty.wchusbserialfa130
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build760706363100022971.tmp/WiFiWebServer.cpp_00000.bin success
opening port /dev/tty.wchusbserialfa130 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 34752 bytes from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gn/T/build760706363100022971.tmp/WiFiWebServer.cpp_00000.bin to flash at 0x00000000
	erasing flash
	size: 0087c0 address: 000000
	first_sector_index: 0
	total_sector_count: 9
	head_sector_count: 9
	adjusted_sector_count: 5
	adjusted_size: 005000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 10000
	setting character timeout 100
	done
	setting timeout 1
	setting character timeout 1
	done
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing boot loader
``````

dogrocker, 2015-05-09:
@sticilface try this #208

InnovativeIdeas, 2015-09-17:
NOTE: for OS X users and the Prolific chip, the current version of esptool.py WILL NOT reliably  auto baud or SYNC to the ESP or transfer data reliably to it .  My test setup is 10.9.5 and ESP-12.  the esptool.py code simply cant work as there are issues that dont have a workaround 

Ive patched and hacked it so that now I can connect every time all the way to 115200, 

In my experience, it slack of understanding of serial comms thats behind a lot of these issues, it took me 6 hours with a serial decoding logic analyser to see what was happening as many of the comms issues are timing related 

your mileage may vary of course :)

tablatronix, 2015-12-20:
@InnovativeIdeas care to elaborate somewhere on timing issues?

icossinib, 2016-01-03:
Hi all, 

I have the 
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed

while trying to flash the esp8266 with Arduino on OSX, however, the pyesptool work perfectly with the same .bin file that arduino creates. Unfortunately, now I depend on Arduino to compile my code.

I am using OS X El Capitán with a generic usb-to serial that I got with my raspberry kit. The only requirement is to install the prolific driver. No needs on tweaking or any other special actions.

igrr, 2016-01-03:
There is a known issue with esptool-ck and PL2303 on OS X. I tried
troubleshooting it for a while, but wasn't able to figure it out.

icossinib, 2016-01-03:
Send me a link to the issue, and I will try to investigate and maybe help you.

However, please notice that esptool.py works fine, so it looks to me like an issue with serial communication parameters.

onkelfunny, 2016-01-03:
@icossinib how long are the wires between the esp and the serial flasher?

I have the same problem with os x if the wires are longer then 5 cm ... with windows i have no problems and can use longer wires

icossinib, 2016-01-03:
RX/TX wires? definitely longer than 5 cm
they are soldered to my usb-to-serial. I’ll try it and let you know

icossinib, 2016-01-03:
anyway, if I use the esptool.py, with the same wire configuration, the flashing works perfectly.

igrr, 2016-01-03:
https://github.com/igrr/esptool-ck/issues/9

See my notes there — basically I found that sometimes the characters
returned by the driver do not match what is sent down the wire.

tablatronix, 2016-01-03:
Not sure if its related but I noticed (on mac) I could not upload faster than `115200` without getting odd errors, and wrong direction errors. After switching to pyesptool I can flash at 921600` just fine.

icossinib, 2016-01-03:
I am using 9600 (because it was the default value, I am about to change it).

nshafiqah7, 2016-07-11:
@igrr hi. i have problem uploading the sketch to esp826. i used the ESP12 DEV KIT,
this is the message error i got.
[image]

nshafiqah7, 2016-07-11:
how can  solved it

krzychb, 2016-07-12:
@nshafiqah7,

Basing on the first error message that says "error: Failed to open COM1" you are likely to have one of two issues:

1. No ESP module is connected to COM1
2. COM1 is already opened by some other application, so IDE can not access it.

Verify to which COM port your ESP is connected to, and if this port is not already used by some other application like e.g. external serial monitor.

You can also check [this troubleshooting  guide](https://github.com/esp8266/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.md).

devyte, 2018-02-27:
This is documented, is an old issue, and there have been several changes to the esptool. Closing.

closed 2018-02-27 by devyte