esp8266/Arduino #143: SoftwareSerial

RussMathis, 2015-04-29:
How would I go about connecting a second serial device to digital pins? Trying to connect a RDM6300 RFID read but have had no success on the UART (rx/tx) lines in. Maybe some SoftwareSerial exists for the ESP?

igrr, 2015-04-30:
UART1 is TX only, it may be used to print debug messages.

You can either use GPIO1/3 pins with UART0 (which are normal pins you use to upload the sketch), or call ```Serial.swap();``` in your setup() function, which changes the UART pins to use GPIO13 and GPIO15 as RX and TX.

By using the swap function you can have your usb-serial connected to GPIO1/3 for sketch upload and your RFID reader connected to GPIO13/15 at the same time.

RussMathis, 2015-04-30:
Can you call Serial.swap() repeatedly in other places of code besides setup(). For example, when data arrives at GPIO13/15, (RFID scanned) then output back to the GPIO1/3 pins by calling Serial.swap() again?

By the way, great job on this project! The ESP is a fantastic new arena for development. For those like myself with low level skills suited to the Arduino your work has opened a larger arena to explore. THANK YOU VERY MUCH.

RussMathis, 2015-04-30:
If seems that GPIO15 must be grounded to run/start the chip? How would you go about wiring pin GPIO15 for use with UART1? Common closed on a relay that would be changed to open from setup()?
Any input would help. Thanks!

duncan-a, 2015-04-30:
The GPIOs should be pulled up or down via a resistor (around 10K ohm is
normally adequate) so, for GPIO15 to conform with the bootup requirement,
connect it to ground via a 10K - the Serial signal will overcome that
easily in use.

igrr, 2015-05-05:
> Can you call Serial.swap() repeatedly in other places of code besides setup()

Yes you can — it will swap between GPIO3/1 and GPIO13/15.

RussMathis, 2015-05-05:
Having issue with swap: currently have 10K from GPIO15 to Gnd. 
The following code works and sends "test" to the monitor window once per second;
void setup(){
    Serial.begin(9600);
    while(true){
        Serial.println("test");
        delay(1000);
    }
}
void loop(){}

However, this code does not output print "test" to the monitor window, in face nothing
is ever seen in the monitor window;
void setup(){
    Serial.begin(9600);
    while(true){
        Serial.println("test);
        Serial.flush();
        Serial.swap();
        delay(1000);
        Serial.swap();
    }
}

What is wrong?

igrr, 2015-05-06:
Do you have the usb-serial RX line connected to the GPIO15 in this case?

RussMathis, 2015-05-06:
I finally got the example I posted to work by updating the hardware folder. 

... I'm thinking it might be my version of the ESP IDE. I've been trying to get the latest version since my version does not have Serial2 defined in the HardwareSerial.h file.
Finally, after getting the latest source Download.zip from: https://github.com/rogerclarkmelbourne/arduino-esp8266/, 
 and putting it in my sketch folder's hardware directory I can compile and upload successfully from Arduino software 1.6.3.

However, Visual Micro does not seem to work since it can't find the esptool to upload. Working on that now...

Any input?

Thanks for the help iggr

igrr, 2015-05-12:
Sorry I don't have any experience with Visual Micro...

closed 2015-05-22 by RussMathis

flyandi, 2015-08-27:
This is older but I have a question regarding the swap, too. Is it ok, to constantly between the two pin assignments, e.g. 

Read from TX/RX 
Swap
Read
Swap
Read .. Write
Swap
Read
...
and so on?

Links2004, 2015-08-27:
@flyandi  you can swap as often you like.
but keep in mind that the ```write``` and ```read``` are asynchronous.
also you may loos bytes of the RX from the non active part. 

if you use ```flush``` after ```write``` and 
you need add some ```delay```  before ```read``` then it shut work.

flyandi, 2015-08-27:
Cool. Thanks Links2004.