esp8266/Arduino #144: "Setup" doesn't seem to be called....

WoodsterDK, 2015-04-30:
Hi there,

I have a project, where the ESP8266 just hangs after boot.
The bootloader puts out:
" ets Jan 8 2013,rst cause:1, boot mode:(3,1)

ets_main.c"

I can mask out some code - it is not even being called - and then it starts up correctly - so it seems to be related to the size of the project.

Any ideas or thoughts would be appreciated.... :)

igrr, 2015-04-30:
See #130 — this is a flash tools issue.
Should fixed in the last couple of commits today.

WoodsterDK, 2015-04-30:
Thanks iggr,

you were absolutely right - ; )

Keep up the good work..

closed 2015-04-30 by WoodsterDK

Rosmianto, 2017-10-11:
In my case, this issue happened when I try to upload a sketch with wrong flash size (4MB flash to 2MB ESP8266 module).