esp8266/Arduino #157: Reboots when combining Ticker and MQTT client PubSubClient

Toshik, 2015-05-01:
When use Ticker and periodical MQTT publish routine inside ticker, ESP resets, seems by WDT.
Separately Ticker and PubSubClient works well, but together - leads to reboot.

Here is meta-sketch:

```
#include <PubSubClient.h>
#include <ESP8266WiFi.h>
#include <Ticker.h>

WiFiClient wifiClient;
PubSubClient client(server, 1883, handleMQTTMessage, wifiClient);
Ticker t;

void setup() 
{
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }

    t.attach_ms(3000, sendData);
}

void sendData()
{
    if (client.publish("test", "test payload"))
        Serial.println("Publish ok");
    else
        Serial.println("Publish failed");
}

```

Related topic: http://www.esp8266.com/viewtopic.php?f=32&t=2432

igrr, 2015-05-01:
Long story short: don't do lengthy stuff (print, network io) from interrupts.

Long story:
At this point I decided not to support continuations in the context of Ticker calls, because this causes some issues. Consider the following case:
```
Ticker t1;
Ticker t2;
void sleeper1() { delay(3000); Serial.println("1"); }
void sleeper2() { delay(2000); Serial.println("2"); }
void setup() { 
  Serial.begin(9600); 
  t1.attach_ms(1000,sleeper); 
  t2.attach_ms(1000,sleeper); 
}
```
What would be the desirable behaviour in this case? Should different tickers be executed in series? Should they be executed in series with the loop() function as well? (meaning that if you hit a delay(2000) in ticker you loop() function will not run for 2 seconds as well). Serial execution is easier to implement, but it is not very intuitive in this context.
Parallel execution of tickers, on the other hand, would actually require support of full-fledged multitasking. We do support continuations, but it is an extremely limited form of multitasking, and only two contexts of execution are supported — system context and sketch context. Introducing additional tasks (and allocating them dynamically) is not worth it IMO.
So this is why Ticker behaves like an interrupt.

Perhaps there are other options to solve this issue (aside from serializing everything and doing multithreading) — I'm open to suggestions here.

Toshik, 2015-05-03:
Well, thanks for explanation.
For simple use now works "flags" approach:

```
volatile bool doSomething = false;

void tickerCallback()
{
    doSomething = true
}

void loop()
{
    if (doSomething)
    {
        realDoSomething();
        doSomething = false;
    }
}
```
At least for MQTT it works perfect.

igrr, 2015-05-03:
Yeah, setting the flag is a good approach. Perhaps I need to add a variant of Ticker which would be called from loop() function (like ```Ticker::update();```)

Toshik, 2015-05-03:
Well, how it done in NodeMCU firmware?
There are also 7 timers, which works like Ticker, as I understand.
There is no WDT issue.

igrr, 2015-05-03:
We have a context switching implementation here to support blocking network IO, unlike NodeMCU.
So if you only do async stuff from the Ticker callback, you're good. If you use blocking calls, there will be a problem (see explanation above).
Suggestions how to deal with this are welcome though.

Toshik, 2015-07-06:
Here I have built very simple Scheduler based on Ticker, it very simplifies Ticker usage:
https://github.com/Toshik/TickerScheduler

closed 2015-11-06 by Links2004

a4aLien, 2018-03-16:
I was facing the same. When calling `client.publish` in an external function, triggered by ticker, my ESP would crash. 

Using Toshik's "flags" approach worked wonderfully well.