esp8266/Arduino #158: Linux 32bit build

dannyvai, 2015-05-01:
Hi,
Why there is no 32bit arduino ide build?
Some people use old computers to program micro-controllers like the esp8266 and arduino.. 
They use 32bit computers...

Btw after building the arduino in the programmer menu-list there shouldn't be the esptool?

sticilface, 2015-05-01:
1) Because igrr doesn't have access to a 32 bit machine to build it. 
2) it was only asked for last week

There is already a thread about this. 
+ the option for esptool was removed when it was put as the primary method for uploading. So esptool is default. I have to change some things to put it back / use esptool.py

igrr, 2015-05-05:
closing as duplicate of #124 — lets move the discussion there

closed 2015-05-05 by igrr

igrr, 2015-05-19:
Added 32-bit Linux tools to the board manager package.