esp8266/Arduino #161: ili9341 hspi not working 02.05.2015 windows build

tontito, 2015-05-02:
hi again.

02.05.2015 windows build
(https://ci.appveyor.com/api/buildjobs/dbj2egv85gam6s8t/artifacts/build/windows/arduino-1.6.1-windows.zip)
download and copile ili9341 example no error compiling but not working screen always white screen.

igrr, 2015-05-02:
Is this different from #120?

UNUMobile, 2015-05-03:
hi
me too. I checkout to "Commits on Apr 30" is working. but 01.05.2015 always white screen.

this commit changed pins_arduino.h , but still white screen.
https://github.com/esp8266/Arduino/commit/83ed3d5d8ff7377a4862e69780326bfe0549316d

tontito, 2015-05-03:
different from #120. Always color problem but 02.05.2015 build not working.

tontito, 2015-05-03:
hi igrr
it's different issue

checkout to "Commits on Apr 30" is working. but 01.05.2015 always white screen.

83ed3d5 commit changed pins_arduino.h and spi library.

problem found.

visit site:
https://ci.appveyor.com/project/igrr/Arduino/history

download:
https://ci.appveyor.com/project/igrr/arduino/build/0.0.78

work ili9341 tft

download:
https://ci.appveyor.com/project/igrr/arduino/build/0.0.84
all file change. 0.0.78
except file:
pins_arduino.h
Arduino.h
core_esp8266_wiring_analog.c
core_esp8266_wiring_digital.c
core_esp8266_wiring_pwm.c
esp8266_peri.h
core_esp8266_timer.c
si2c.c
si2c.h

work ili9341 tft

UNUMobile, 2015-05-07:
checkout to each version, I got the problem.
https://github.com/esp8266/Arduino/commit/7ece3e436dbf3c42099661b3c614760f17a6dc3b in this commit init 0-5 pin default to INPUT, but ili9341 used GPIO2.

so, add pinMode(2, OUTPUT); in setup() function
it's work!

tontito, 2015-05-11:
My screen damaged not tested yet.

tontito, 2015-07-12:
add pinMode(2, OUTPUT); in setup() . Good working lcd

closed 2015-07-12 by tontito