esp8266/Arduino #162: Problem cloning library

tytower, 2015-05-03:
This may also be related to the 32 bit / 64 bit error , but I get this message when trying to clone the source -Is the reason known to anyone?

# git clone https://github.com/esp8266/Arduino.git
Cloning into 'Arduino'...
error: Problem with the SSL CA cert (path? access rights?) while accessing https://github.com/esp8266/Arduino.git

Links2004, 2015-05-03:
ssl looks good here.
https://www.ssllabs.com/ssltest/analyze.html?d=https%3A%2F%2Fgithub.com%2Fesp8266%2FArduino.git
you shut check you local installation and update you ca-certificates.
```apt-get update && apt-get upgrade```

tytower, 2015-05-03:
Funny I can clone other libraries here on git hub but not this one . I use Puppy Linux and run as root . I dont have apt-get at all and use a Puppy Package Manager , not that I think any of that should affect this cloning operation

tytower, 2015-05-03:
Please look a little closer at this . I updated what I could and ended up with 100Meg of unnecessary crud on my system and it made not the slightest bit of difference . I think the problem is there with you

webguy16, 2015-05-04:
Not on linux... but it's currently working for me.
[image]

igrr, 2015-05-04:
A bit of googling shows a lot of answers how to solve this issue, like this
one: http://stackoverflow.com/a/4454754

This is obviously a client side problem, your trusted certificates are not
up to date or not accessible to git for some reason.

tytower, 2015-05-04:
OK Thanks - Google is excellent if you know what you are looking for . I have no idea on this certificate business.
Its clearly security you choose to set up on your repository and thats your affair but it would help if there was a warning that it is secured (maybe there is ,I didn't see one)

My way round this was to clone via 
 "env GIT_SSL_NO_VERIFY=true git clone https://github.com/esp8266/Arduino" 
No security but it saves wasting a day trying to find out whats wrong

igrr, 2015-05-05:
> you choose to set up on your repository

I don't think Github offers any particular "choice" in terms of HTTPS certificates.

You can although use git protocol instead of https when cloning (as in ``` git clone git@github.com:esp8266/Arduino.git```) — this might resolve the certificate issue as well.

closed 2015-05-05 by igrr