esp8266/Arduino #164: UART pins as GPIO

Toshik, 2015-05-03:
I know, that we can use TX/RX pins as GPIO 1/3

Now I am try to use DS18x20 on RX (GPIO3) pin (executing example from OneWire library) , ESP but it does not detects sensor.
When I use GPIO2 - it works fine.

Do I miss something?

igrr, 2015-05-03:
Perhaps you have an external pull-up resistor on GPIO2 but not GPIO3?

Toshik, 2015-05-03:
Yes, I use pullup resistor 4k7 with DS18B20 and I just switches GPIO2 - GPIO3 with that resistor

May connected USB-UART module be cause?

igrr, 2015-05-03:
Why exactly do you need to have a USB-UART on the DS18B20 pin?

Toshik, 2015-05-03:
Generally I want DS18B20 run at RX pin, so now it interconnected with USB-UART.
I have made kind of dev board for ESP-01, but forgot to make TX/RX on/off jumpers :)

igrr, 2015-05-03:
Try disconnecting the USB-UART and see if it works then.

Toshik, 2015-05-04:
I have made jumpers on my dev board to disconnect rx/tx pins from UART-USB adapter. 
But, it did not solve issue - DS18B20 still not recognized on GPIO3
Maybe OneWire library somehow conflicts with Serial one?

Toshik, 2015-05-05:
I have found out that OneWire sure conflicts with Serial.

It works perfect if I do:

```
Serial.end();
readDSData(); // Read DS18B20 data
Serial.begin(...)
```

Such approach works regardless on connected UART adapter.

Toshik, 2015-05-05:
Seems, best solution would be ability to disable RX upon Serial initialization

Toshik, 2015-05-12:
Ok, as I see \hardware\esp8266com\esp8266\cores\esp8266\HardwareSerial.cpp : 309
```
pinMode(3, SPECIAL);
```

makes Pin 3 unusable at least for OneWire.

themakerman, 2015-10-21:
@Toshik Hey,Can you tell me how can i reflash my ESP once i have made its UART pins work like GPIO.I am stuck.I want the UART pins to work like UARTS as they where first.I have ESP8266-01

Toshik, 2015-10-22:
@themakerman Normally, you flash firmware by resetting ESP and booting it into "flash" mode, so there is no matter, what mode you apply to any pin inside your program.

closed 2015-11-27 by Links2004