esp8266/Arduino #166: Can't receive udp packets

anesthetized, 2015-05-03:
I'm working to implement a simple mDNS client (ask who's foobar.local and get response) and i'm stuck as it correctly sends packets, my pc with avahi-daemon responds accordingly (i'm checking with wireshark), but esp doesn't seem to receive any packet!

Wireshark screenshot proof:
[image]

First line is esp's mDNS request, my pc answers on both unicast (second line) and multicast (third and fourth). Unicast response has 5353 src port and a random one as dst port. Multicast ports are both 5353.
It is not a unicast-or-multicast problem, since it seems to receive generic packets when i comment out the lookup() function.

Code (with multicast):
```Cpp
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <Ticker.h>

Ticker nameLookup;

const char* ssid     = "ssid";
const char* password = "passwd";
const char* hostname = "hostname";

const unsigned int loc_port = 5353;
WiFiUDP udp;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println(WiFi.localIP());

  udp.begin(5353);
  nameLookup.attach(5, lookup);
}

void loop() {

  int packet = udp.parsePacket();
  if(packet!=0){
    Serial.println("gotcha!");
    delay(100);
  }

}

void lookup(){
    Serial.println("Hostname Lookup sent");
 
    char header[13] = { 0x00, 0x00, // ID
                        0x00, 0x00, // FLAGS
                        0x00, 0x01, // QDCOUNT
                        0x00, 0x00, // ANCOUNT
                        0x00, 0x00, // NSCOUNT
                        0x00, 0x00, // ARCOUNT
                        0x00};      // hostname length

    char trailer[11] = { 0x05, 0x6c, 0x6f, 0x63, 0x61, 0x6c, 0x00, //local + null terminator
                         0x00, 0x01,   // QDTYPE
                         0x00, 0x01};  // QCLASS

    unsigned short int hostname_len = strlen(hostname);
    unsigned short int packet_size = 24 + hostname_len;
    char* packet = new char[packet_size];
    
  // Forge the request packet
    header[12] = hostname_len;         // Set hostname length byte
    memcpy(packet, header, 13);
    memcpy(packet+13, hostname, hostname_len);
    memcpy(packet+13+hostname_len, trailer, 11);
    
  // Send mDNS request
    udp.beginPacket(IPAddress(224, 0, 0, 251), 5353);
    udp.write(packet, packet_size);
    udp.endPacket();

    delete[] packet;
}

```
(My understanding is that such library is wanted by many, and I want to contribute to the project! :D)

igrr, 2015-05-04:
Thanks for the report, i will check on this.

In the meantime, please read [this explanation](https://github.com/esp8266/Arduino/issues/157#issuecomment-98132083) why network (or any other blocking IO) should not be done from Ticker callbacks right now. 
The code you posted will likely work because UDP is essentially non-blocking, but if you would do some TCP (or any other blocking stuff) from your callback, that would likely cause the stack to lock up, leading to a WDT reset.

anesthetized, 2015-05-04:
Thanks Igor, i will read this and adjust my code accordingly. In my experience with ticker or without (lookup() call in loop()).

Makuna, 2015-05-04:
could this be related to my issue where only ~8 packets are received before it stops completely?

https://github.com/esp8266/Arduino/issues/151

closed 2015-05-19 by igrr