esp8266/Arduino #167: Wire broken

av1024, 2015-05-03:
1. Wire.pins(sda, scl) removed but SDA/SCL defined as `const uint8_t` in pins_arduino.h so can not be redefined (or I don't know how)
2. Wire.cpp does not compile in e1a6e3d755 (may be si2c.h not included).

---
    Wire/Wire.cpp.o: In function `TwoWire::write(unsigned char)':
    Wire.cpp:(.text+0x130): undefined reference to `twi_init(unsigned char, unsigned char)'
    Wire.cpp:(.text+0x143): undefined reference to `twi_init(unsigned char, unsigned char)'
    Wire/Wire.cpp.o: In function `TwoWire::begin()':
    Wire.cpp:(.text+0x190): undefined reference to `twi_setClock(unsigned int)'
    Wire/Wire.cpp.o: In function `TwoWire::begin(unsigned char)':
    Wire.cpp:(.text+0x19b): undefined reference to `twi_setClock(unsigned int)'
    Wire/Wire.cpp.o: In function `TwoWire::begin(int)':
    Wire.cpp:(.text+0x1a8): undefined reference to `twi_readFrom(unsigned char, unsigned char*, unsigned int, unsigned char)'
    Wire/Wire.cpp.o: In function `TwoWire::setClock(unsigned int)':
    Wire.cpp:(.text+0x1ca): undefined reference to `twi_readFrom(unsigned char, unsigned char*, unsigned int, unsigned char)'
    Wire/Wire.cpp.o: In function `TwoWire::requestFrom(int, int, int)':
    Wire.cpp:(.text+0x29c): undefined reference to `twi_writeTo(unsigned char, unsigned char*, unsigned int, unsigned char)'
    Wire/Wire.cpp.o: In function `TwoWire::beginTransmission(unsigned char)':
    Wire.cpp:(.text+0x2ba): undefined reference to `twi_writeTo(unsigned char, unsigned char*, unsigned int, unsigned char)'
    collect2: error: ld returned 1 exit status
    Error compiling.

igrr, 2015-05-04:
Should be fixed in 2e3e98b

av1024, 2015-05-04:
Compilation success but TWI not worked. The test code in BMP085.begin() fails (read8 returns 0xFF instead of 0x55)

1. deprecated `pins()` realization is not valid because does not store passed values and next `begin()` uses pre-defined SDA/SCL constants, not passed to `pins()`
2. Looks like something else broken for me: at least reading from Serial (I use it for CLI-configuration) causes wdt reset :(

The read8() code from adafruit_BMP085:

```C++
    uint8_t Adafruit_BMP085::read8(uint8_t a) {
      uint8_t ret;

      Wire.beginTransmission(BMP085_I2CADDR); // start transmission to device 
    #if (ARDUINO >= 100)
      Wire.write(a); // sends register address to read from
    #else
      Wire.send(a); // sends register address to read from
    #endif
      Wire.endTransmission(); // end transmission
  
      Wire.beginTransmission(BMP085_I2CADDR); // start transmission to device 
      Wire.requestFrom(BMP085_I2CADDR, 1);// send data n-bytes read
    #if (ARDUINO >= 100)
      ret = Wire.read(); // receive DATA
    #else
      ret = Wire.receive(); // receive DATA
    #endif
      Wire.endTransmission(); // end transmission
    
      return ret;
    }
```

The Serial handler implementation (called from loop()):

```C++
    void onSerial() {
      while(Serial.available()) {
        char c = (char) Serial.read();
        Serial.write(c);
        cli.putch(c);
        if (cli.isDone()) {
          Serial.println();
          break;
        }
      } 
      if (cli.isDone()) {
       if (!cli.handle(Serial)) {
         Serial.println("\nERR");
       }
     }
    }
```

igrr, 2015-05-05:
Wire.pins should be fixed now (but it's better to use ```Wire.begin(sda, scl)```)

av1024, 2015-05-06:
pins() fix confirmed.
But TWI still not work for me on 4e387e4ffe
Unfortunately I burned my sensor during reflashing it back so I can't test anymore  at least two weeks ((

closed 2015-05-19 by igrr