esp8266/Arduino #170: Math functions are not included and fails compilation under Windows

robotiko, 2015-05-04:
Hi, 
I'm trying to use stepper control lib AccelStepper.h lib (http://www.airspayce.com/mikem/arduino/AccelStepper/index.html) but it fails at compilation due to missing sqrt function. 
>AccelStepper.cpp:(.text+0x679): undefined reference to `sqrt'
The code is the example code that is quite simple and I tried adding explicitly the math.h  but had same results. 

I just downloaded (yesterday) the latest release for windows from github project page (arduino-1.6.1-p1-windows.zip) and using it in a  windows 7 32bits version.

How can I include the math lib in the build?  Do I have to update the arduino version?

Thank you for such great project and help.

Links2004, 2015-05-04:
you need a newer version then the libm is included.
you can try the version from the auto build
https://ci.appveyor.com/project/igrr/Arduino/build/artifacts/
or build it your self :)

robotiko, 2015-05-04:
Hi Links2004 ,
thank you for your suggestion. 
I downloaded the built from the Appveyor, but seems to be a Win64x version. Could find a 32bits version so  I went ahead.. and built it ;)  . 
Compiling the test app with the new build has a newer error to what looks like a hardcoded route from development "d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\errno/../../../../../newlib/libc/errno/errno.c:13"

Compile log : 

C:\Users\mot\AppData\Local\Temp\build729029768781264497.tmp/core.a(libc_replacements.c.o): In function `__errno':
libc_replacements.c:(.irom0.text+0x378): multiple definition of `__errno'
c:/temparduino/arduino-esp8266/build/windows/work/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libc.a(lib_a-errno.o):d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\errno/../../../../../newlib/libc/errno/errno.c:13: first defined here
collect2.exe: error: ld returned 1 exit status
Error compiling.

Stuck again

antelder, 2015-05-05:
This is http://www.esp8266.com/viewtopic.php?f=28&t=2898 isn't it?
I tried the lastest build from https://ci.appveyor.com/project/igrr/Arduino/build/artifacts/ but that fails to start with:
Exception in thread "main" java.lang.UnsatisfiedLinkError: jnidispatch (/com/sun
/jna/win32-amd64/jnidispatch.dll) not found in resource path
        at com.sun.jna.Native.loadNativeLibrary(Native.java:471)
        at com.sun.jna.Native.<clinit>(Native.java:91)
        at processing.app.windows.Platform.<clinit>(Platform.java:287)
        at java.lang.Class.forName0(Native Method)
        at java.lang.Class.forName(Class.java:190)
        at processing.app.BaseNoGui.initPlatform(BaseNoGui.java:601)
        at processing.app.Base.initPlatform(Base.java:177)
        at processing.app.Base.main(Base.java:91)

Is there anything i can do to get past that?

robotiko, 2015-05-06:
relate to #156

robotiko, 2015-05-06:
HI  antelder,
I had the same issue and just solved by building manually the project (see my reply to Links2004).
That will fix the arduino app issues but also introduces some build issues in compilation for some libs (like the accelstepper) but may fix your problem because the math.h seem to work.

antelder, 2015-05-06:
Thanks for replying robotiko. Was trying to avoid building the source as thought it would probably mean hours of messing about with gcc etc...but gave it a go now and turns its easy and just works, yay...however trying that new build gives me exactly the same UnsatisfiedLinkError: jnidispatch exception

robotiko, 2015-05-06:
Hi antelder, 
I had the same worry and the same experience, build is quite straightforward. 
However, my arduino IDE didn't work properly just making the build, I had to run it with "ant run".
I don't have any issue with ide after that. 
Another quick and VERY dirty solution that seems to work really good is to overwrite the non IDE folders in the Arduino installer that is in the latest releases (https://github.com/igrr/Arduino/releases/download/1.6.1-esp8266-1/arduino-1.6.1-p1-windows.zip)  with the ones with the ESP fixes from the alpha ( https://ci.appveyor.com/project/igrr/Arduino/build/artifacts/). 

Lets see of that helps.

antelder, 2015-05-06:
I wasn't sure which are the "non IDE folders" but firkling around in the directories i see there are a couple of folders with esp8266 in their name so i've replaced those with copies from the latest build and it does seem to have fixed the problem compiling pow.  Thanks!

igrr, 2015-05-06:
__errno is really declared in two places — libc_replacements.c and libc, so it's not a paths related issue. However this error doesn't seem to happen for me on OS X, perhaps windows version of the toolchain has been built with different newlib flags.

igrr, 2015-05-06:
@antelder are you running into this ```Exception in thread "main" java.lang.UnsatisfiedLinkError``` error on 32-bit or 64-bit Windows?

Links2004, 2015-05-06:
the problem may came from this commit 
https://github.com/esp8266/Arduino/commit/83ed3d5d8ff7377a4862e69780326bfe0549316d
there we add the libc back.
Is libc really working now or it is only working be course the compiler choose our replacement functions in the most cases.
If the libc is working we can remove the most replacement functions excepted the AVR special stuff.
otherwise we shut remove the libc again.

robotiko, 2015-05-06:
HI igrr, when I mean  that it can be a path issue , it is due to the error it logs : 
 "....d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\errno/../../../../../newlib/libc/errno/errno.c:13"path 

I have no d:\ivan\projects route in the machine .

About the startup error of the ide, that  @antelder mentions, I'm running in Win7 32-bit and it doesn't start either. 
It gives a really weird error when clicking on the arduino.exe from https://ci.appveyor.com/project/igrr/Arduino/build/artifacts/..  "%1 Is Not a Valid Win32 Application".. however building it directly from repo seems to work.

igrr, 2015-05-06:
As I said, it's not a path issue — the linker spits out the file path that
was used when building libc — and it is not used anywhere in the build
process. The error comes from the fact that we have two definitions of
errno: in our core and in newlib.

The appveyor builds are done on a 64-bit host, so they won't work on 32-bit
windows machines.

antelder, 2015-05-06:
64-bit Windows 7

robotiko, 2015-05-06:
igrr, thank you for having a look at this. 

Since maths.h seems to be included and now the problem is reduced to multiple definition of `__errno'. 
Do we close this issue and keep #186 open? or leave this one that got more traction?

About the 32bit incompatibility of the appveyor builds, it would be a good idea to add a note somewhere (maybe in the readme) to avoid additional issues and confusions.

Links2004, 2015-05-06:
added but not merge at the moment see #190

igrr, 2015-05-12:
fixed in the latest trunk

igrr, 2015-05-19:
Regarding ```UnsatisfiedLinkError: jnidispatch exception``` — you can now grab official Arduino 1.6.4 from the arduino.cc website, and install esp8266 support using board manager package.

closed 2015-05-19 by igrr

robotiko, 2015-05-19:
Thanks igrr.
Tested and working on W7 32b