esp8266/Arduino #174: One esp-01 as STA can't connect to another esp-01 as STA_AP

WoodsterDK, 2015-05-05:
Hi there,

I'm trying to make a esp-01 udp to serial gateway with one esp-01 with static ip 10.0.0.1, that should host the WLAN for another gateway as STA with static IP address 10.0.0.2.

I can get the WLAN running, and have my mobile or laptop connect to it - and ping the 10.0.0.1 device, but the STA device connects the WLAN but can not be found by a WLAN analyzer or pinged.

The status is connected - but no ping is possible.
I even tried to enable the DHCP for the STA_AP, and both my mobile, laptop and the STA device gets an IP address - but I still can't ping the STA device....
The strange thing is, that if i let the STA device connect to my normal WLAN then I have no problem with pinging it.

Any ideas, or example sketches that shows it's working?

Martin

igrr, 2015-05-05:
That won't work because routing is not enabled in the network stack from Espressif.

WoodsterDK, 2015-05-05:
Sorry - what won't work... the STA_AP or the STA part?

igrr, 2015-05-05:
> that should host the WLAN
If i understood your question correctly, the WLAN is hosted by an esp8266.
Routing between clients connected to such a WLAN will not work — because it is not enabled in the network stack. This explains why two devices connected to esp's softAP can not ping each other.

WoodsterDK, 2015-05-05:
Ahh.. okay - I think I have had both my laptop and mobile connected at once - with the pinging working - but I'm not sure about that. Will have to check.
If the routing is not enabled what is the AP then usefull for other than the initial setup of the STA?

igrr, 2015-05-05:
I don't think the AP is currently useful for anything other than initial setup.
There is an open-source version of the library, https://github.com/kadamski/esp-lwip, we can switch to it and enable routing. But with the current version of the SDK esp-lwip seems to be broken, so have to wait until it gets fixed.

WoodsterDK, 2015-05-06:
OK - well I will just add a real hotspot, and use both in STA mode - just thought I could create a closed WIFI environment for my system.
Thanks for you reply.

closed 2015-05-06 by WoodsterDK