esp8266/Arduino #178: Need a more complete MQTT library

CptanPanic, 2015-05-05:
MQTT is one of the most used internet of things protocols, so as the best IoT platform the ESP8266 needs to support this.  The linked to library PubSubClient is ok, but it doesn't support the complete MQTT protocol such as ensuring delivery of message (QOS > 0).  The other ESP firmwares such as nodemcu, etc. have these features, but it would be great if we could get this working in the arduino framework.

duncan-a, 2015-05-05:
I'd second that - but I suspect that it isn't a simple process to include
it.

Toshik, 2015-05-05:
Try this one: https://github.com/hemantsangwan/pubsubclient

CptanPanic, 2015-05-05:
That one from @hemantsangwan doesn't have publish qos option.  

> Try this one: https://github.com/hemantsangwan/pubsubclient

duncan-a, 2015-05-05:
Thanks guys...

marvinroger, 2015-05-05:
Might also be interesting: https://github.com/mikaelpatel/Cosa/tree/367c8fe48388b5c26ba22844527f47996289e3bc/libraries/MQTT

Imroy, 2015-05-06:
That repository from hemantsangwan recently merged from my tree and he hasn't done anything with it since. And I don't know if he's working on it. So you might as well use mine directly.

I'm currently working on making it more complete, but it will take at least a few days. I hope I'm not going too overboard... :P

igrr, 2015-05-06:
@Imroy great work on that library — it would actually be great to bundle your version with the Arduino IDE for the ESP. Let me know when it gets more or less stable so that I can add a copy to our distro.

Toshik, 2015-05-06:
@Imroy Sorry, I just didn't see yours fork.
In "Fork map" https://github.com/knolleary/pubsubclient/network it looks like hemantsangwan's fork ahead

Toshik, 2015-05-06:
@Imroy Do you support github issues in your fork?
I have ran into ESP reboot if WiFi disconnects then reconnects and trying to send something to MQTT

igrr, 2015-05-06:
This reboot is not related to the code inside mqtt library. More like
something in the Wifi library...

Toshik, 2015-05-06:
@igrr I thought so, but tested code like this:

```
void loop()
{
    if (WiFi.status() == WL_CONNECTED)
    {
        Serial.print("test wf: ");
        Serial.println(counter++);
        wifiClient.write("0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789", 130);
        Serial.println("test wf: done");
    }
}
```

And it did not caused any reboots on WiFi disconnenctions...

Imroy, 2015-05-07:
Here's my work on making PubSubClient more flexible. I ended up making a class for every MQTT message type and moving/reimplementing much of Nick's original work into those classes. The PubSubClient class is still there, but uses the classes to create/get messages. This means the "old" interface is still in place and works the same. See the auth and qos examples to see how to create a Connect or Publish object and set some options before actually using it.

https://github.com/Imroy/pubsubclient/tree/OOP-monster

CptanPanic, 2015-05-07:
Looks nice, I will check it out soon.

dogrocker, 2015-05-08:
Hello, @Imroy 

I have use https://github.com/Imroy/pubsubclient
and try your `mqtt_basic` example and I can't publish to my MQTT broker
it can't connect to the broker client.connect("arduinoClient"), 
I don't know what the problem.

I change code to

```
String server = "gis.it.kmitl.ac.th";
PubSubClient client(server, 1883);
```

Imroy, 2015-05-08:
@dogrocker mqtt_basic doesn't print anything when the messages come in. Are you adding printing in the callback, or are you using the mqtt_subscriber example in my OOP-monster branch?

dogrocker, 2015-05-08:
@Imroy I add this to debug the messages that come in.

```
void callback(char* topic, byte* payload, unsigned int length) {
  // handle message arrived
  String payloads = String((char*)payload);
  Serial.println(payloads);
}
```

full code

```
/*
 Basic MQTT example 
 
  - connects to an MQTT server
  - publishes "hello world" to the topic "outTopic"
  - subscribes to the topic "inTopic"
*/

#include <ESP8266WiFi.h>
#include <PubSubClient.h>

const char *ssid =	"";		// cannot be longer than 32 characters!
const char *pass =	"";		//

// Update these with values suitable for your network.
//IPAddress server(172, 16, 0, 2);
String server = "gis.it.kmitl.ac.th";
void callback(char* topic, byte* payload, unsigned int length) {
  // handle message arrived
  String paysloads = String((char*)payload);
  Serial.println(paysloads);
}

PubSubClient client(server, 1883);

void setup()
{
  // Setup console
  Serial.begin(115200);
  delay(10);
  Serial.println();
  Serial.println();

  client.set_callback(callback);

  WiFi.begin(ssid, pass);

  int retries = 0;
  while ((WiFi.status() != WL_CONNECTED) && (retries < 10)) {
    retries++;
    delay(500);
    Serial.print(".");
  }
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("");
    Serial.println("WiFi connected");
  }

  if (client.connect("arduinoClientasdsdsasdd")) {
    client.publish("outTopic","hello world");
    client.subscribe("test/get");
  } else {
    Serial.println("MQTT connect failed");
    Serial.println("Will reset and try again...");
    abort();
  }
}

void loop()
{
  client.loop();
}
```

Toshik, 2015-05-08:
@Imroy Somewhere in receive message routine there are memory leaks. Even without callback each mqtt message to ESP consumes  >70 bytes of memory, depending on message/topic length

dogrocker, 2015-05-08:
@Toshik Thanks, do you know how to fix this?

Imroy, 2015-05-08:
@dogrocker That looks like you're using the master branch, which is just a somewhat cleaned up version of Nick's original code. You're getting that string because 'payload' is just a pointer into PubSubClient's single, recycled packet buffer. It's not a null-terminated string. So it's printing out your received message and whatever else happens to be in the buffer, until it happens to hit a null byte.

If you switch over to my OOP-monster branch, the callback function receives an MQTT::Publish object, on which you can call methods to get nice strings.

Imroy, 2015-05-08:
@Toshik Is this on my new OOP-monster branch? I am doing dynamic memory allocation at a few places, but I tried to keep it to a minimum. How do you find the amount of memory free/used?

dogrocker, 2015-05-08:
@Imroy Thanks for answer and make the new MQTT for esp8266,

```

dogrocker, 2015-05-08:
@Imroy Thanks for answer and make the new MQTT for esp8266,
but I have try you `OOP-monster` branch but I can't compile it

I don't know what the error is.
```
/Applications/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: mqtt_subscriber.cpp.elf section `.text' will not fit in region `iram1_0_seg'
/Applications/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o):(.literal+0x1c): undefined reference to `_sbrk_r'
/Applications/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-mallocr.o): In function `malloc_extend_top':
mallocr.c:(.text+0x5e): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x127): undefined reference to `_sbrk_r'
/Applications/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o): In function `_malloc_trim_r':
mallocr.c:(.text+0x366): undefined reference to `_sbrk_r'
mallocr.c:(.text+0x38f): undefined reference to `_sbrk_r'
/Applications/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib/libc.a(lib_a-freer.o):mallocr.c:(.text+0x3a0): more undefined references to `_sbrk_r' follow
collect2: error: ld returned 1 exit status
Error compiling.
```

dogrocker, 2015-05-08:
Now I can compile `OOP-monster` by recompile the new Arduino_ide with ant
but new problem now is about Arduino_ide can't upload the sketch #208  :(

dogrocker, 2015-05-09:
@Imroy Thanks, I'm try your `OOP-monster` subscribe example it work very well !

I'll try with my code soon.

Toshik, 2015-05-09:
@Imroy, yes, it is OOP branch. I use builtin function to get free heap: getFreeHeap()

Imroy, 2015-05-09:
@Toshik Thanks, I see the "leak" too. It might just be that the heap is becoming fragmented from the out-of-order mallocs and frees. I'll look into making the allocation more static.

Imroy, 2015-05-11:
@Toshik Memory leak fixed! I had destructors on some classes for freeing up dynamically-allocated memory but they weren't being called because (1) I was using free() on an object pointer instead of delete, and (2) the base class didn't have a virtual destructor. Three changed lines is all it needed!

marvinroger, 2015-05-11:
Would you mind opening an issue tracker on your repo @Imroy ? If your fork becomes a "standard" for MQTT, this is a must !

Imroy, 2015-05-12:
@marvinroger Okay, it now has an issue tracker. I've also merged my "OOP monster" branch into master since everything seems to be working now.

CptanPanic, 2015-05-12:
Great

Diaoul, 2015-05-26:
I'm new on the topic but what about https://github.com/256dpi/arduino-mqtt ?

closed 2015-11-06 by Links2004

capedra, 2019-02-27:
@Diaoul Great library!! It's working great with TLS and QOS 2. Thanks a lot!!