esp8266/Arduino #179: Hardware Reset

rokison, 2015-05-05:
Is there a way to trigger a hardware reset?
It would be great after the AP stores an SSID and password into the EEPROM it rebooted automatically to try and connect to it.

Zeron, 2015-05-05:
From the readme file:

ESP.reset() resets the CPU.

rokison, 2015-05-05:
Couldn't find that in the readme I have.
It also doesn't compile, saying "ESP was not declared in this scope".
Sure I'm making a rookie error, but any further guidance much appreciated!

rokison, 2015-05-05:
Am I just using an old build?!

Zeron, 2015-05-05:
ESP.reset() was added with this commit: https://github.com/esp8266/Arduino/commit/7b70acf3377d08e52d4ed3da364eb36da037c7d6  on April 11

rokison, 2015-05-06:
I am feeling like a complete noob but I can't get a working version with the ESP feature.
Is there any way the "download" for windows linked to from the home page of this repository could be updated to contain this functionality. I'm afraid I'm not quite capable of building from source!

duncan-a, 2015-05-06:
The last Windows version I managed to download and install (
https://ci.appveyor.com/api/buildjobs/8s54dteakqvt3e3q/artifacts/build/windows/arduino-1.6.1-windows.zip)
appears to be Win64 only - trying to run Arduino.exe briefly flashes up an
error message that it's not a valid Win32 file...

I am a complete noob, so there may be something I'[m not understanding...

igrr, 2015-05-06:
@duncan-a  The appveyor builds are 64-bit, so you can't run them on 32-bit Windows.

duncan-a, 2015-05-07:
OK - where can I find a recent 32 bit version?

Many of the problems I had with the .zip file on 'Releases' (downloaded
about a month ago) are, reputedly, fixed now...

As an aside, I can't get the 1.6.1 IDE to recognise the RF24 library (for
the nRF24L01)

closed 2015-05-19 by igrr