esp8266/Arduino #182: Esp8266 Unique Instructions for BitBang

Makuna, 2015-05-06:
I have found several very useful asm instructions for bitbang that should probably be placed in a common and accessible location.

Let me know of a great spot to place these and I will create a pull for them.  Also provide feedback on formatting to better fit in with this code base.

one is CCOUNT, this is similar to the ARM CYCCNT, it is the count of cycles.
```C++
#define RSR_CCOUNT(r)     __asm__ __volatile__("rsr %0,ccount":"=a" (r))
static inline uint32_t get_ccount(void)
{
    uint32_t ccount;
    RSR_CCOUNT(ccount);
    return ccount;
}
```
the other is a true enable and disable interrupts.  The current version do not block the level 14 interrupts so this can cause problems.  These should be used sparingly but are required for libraries like NeoPixels.

```
// Read Set Interrupt Level
#define RSIL(r)  __asm__ __volatile__("rsil %0,15 ; esync":"=a" (r))
// Write Register Processor State
#define WSR_PS(w)  __asm__ __volatile__("wsr %0,ps ; esync"::"a" (w): "memory")

static inline uint32_t esp8266_enter_critical()
{
    uint32_t state;
    RSIL(state);
    return state;
}

static inline void esp8266_leave_critical(uint32_t state)
{
    WSR_PS(state);
}
```

igrr, 2015-05-06:
You can declare those functions in Arduino.h or Esp.h.
I would suggest naming them with ```xt_``` prefix rather than ```esp8266_``` because these are xtensa-specific, not chip-specific.

Makuna, 2015-05-06:
this pull https://github.com/esp8266/Arduino/pull/191 is tracking this issue.

Makuna, 2015-05-10:
pull was merged

closed 2015-05-10 by Makuna