esp8266/Arduino #184: Can't compile anymore...

sticilface, 2015-05-06:
Using latest commits  1.6.1-esp8266-1-211-g8dc2570

this is the wifi scan example.  I was getting errors on a previous build as well, with SDA, SDK errors. 

`````````

Build options changed, rebuilding all
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp: In function 'void i2c_init(int, int)':
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:78:13: error: 'ESP_PINS_OFFSET' was not declared in this scope
     pinMode(ESP_PINS_OFFSET + sda_pin, OUTPUT_OPEN_DRAIN);
             ^
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp: In function 'void i2c_release()':
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:85:13: error: 'ESP_PINS_OFFSET' was not declared in this scope
     pinMode(ESP_PINS_OFFSET + s_sda_pin, INPUT);
             ^
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp: In function 'size_t i2c_master_read_from(int, uint8_t*, size_t, bool)':
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:161:17: error: 'ESP_PINS_OFFSET' was not declared in this scope
         pinMode(ESP_PINS_OFFSET + s_sda_pin, INPUT);
                 ^
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp: In function 'void twi_init()':
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:190:19: error: declaration of C function 'void twi_init()' conflicts with
 void twi_init(void) {
                   ^
In file included from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:40:0,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:22:
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/twi.h:29:6: error: previous declaration 'void twi_init(unsigned char, unsigned char)' here
 void twi_init(unsigned char sda, unsigned char scl);
      ^
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp: In function 'uint8_t twi_readFrom(uint8_t, uint8_t*, uint8_t, uint8_t)':
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:196:99: error: declaration of C function 'uint8_t twi_readFrom(uint8_t, uint8_t*, uint8_t, uint8_t)' conflicts with
 uint8_t ICACHE_FLASH_ATTR twi_readFrom(uint8_t addr, uint8_t* data, uint8_t size, uint8_t sendStop) {
                                                                                                   ^
In file included from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:40:0,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:22:
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/twi.h:33:9: error: previous declaration 'uint8_t twi_readFrom(unsigned char, unsigned char*, unsigned int, unsigned char)' here
 uint8_t twi_readFrom(unsigned char address, unsigned char * buf, unsigned int len, unsigned char sendStop);
         ^
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp: In function 'uint8_t twi_writeTo(uint8_t, uint8_t*, uint8_t, uint8_t, uint8_t)':
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:200:112: error: declaration of C function 'uint8_t twi_writeTo(uint8_t, uint8_t*, uint8_t, uint8_t, uint8_t)' conflicts with
 uint8_t ICACHE_FLASH_ATTR twi_writeTo(uint8_t addr, uint8_t* data, uint8_t size, uint8_t wait, uint8_t sendStop) {
                                                                                                                ^
In file included from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:40:0,
                 from /Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/i2c.cpp:22:
/Users/amelvin/git/Arduino/build/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/twi.h:32:9: error: previous declaration 'uint8_t twi_writeTo(unsigned char, unsigned char*, unsigned int, unsigned char)' here
 uint8_t twi_writeTo(unsigned char address, unsigned char * buf, unsigned int len, unsigned char sendStop);
         ^
Error compiling.
``````

igrr, 2015-05-06:
i2c.cpp [is not in the trunk anymore](https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/cores/esp8266) — could you please check your working copy again?

sticilface, 2015-05-06:
Should it be there? Ie. Should I copy it over?

igrr, 2015-05-06:
You should delete it if you still have it.
But if you did ```git clone``` this file shouldn't be there.

sticilface, 2015-05-06:
Ah, ok.  I did git pull.... 

if i clone it works fine.  For the git gurus what is the best way to update a local 'git clone' without redownloading everything.  I though git pull, was the way forward, but that is not quite giving me an identical copy.

igrr, 2015-05-07:
```git pull origin esp8266``` works for me

closed 2015-05-07 by igrr

sticilface, 2015-05-07:
Thank you.  I’m very new to git!
> On 7 May 2015, at 11:16, Ivan Grokhotkov <notifications@github.com> wrote:
> 
> git pull origin esp8266