esp8266/Arduino #186: Error compiling with math dependency libc_replacements.c: multiple definition of `__errno'

robotiko, 2015-05-06:
I get this build failure  libc_replacements.c:(.irom0.text+0x378): multiple definition of `__errno'
when compiling a math dependency  (AccelStepper examples) 

Looks like i t has a hardcoded route to :"d:\ivan\projects\arduinoesp\toolchain\"
Compiles under arduino boards without issues. 

Using latest alpha build under windows (edaa9142031cba35b77de7f410145df03de43d05).

Trace Log: 

C:\Users\mot\AppData\Local\Temp\build2898262615022444565.tmp/core.a(libc_replacements.c.o): In function `__errno':
libc_replacements.c:(.irom0.text+0x378): multiple definition of `__errno'
c:/users/mot/desktop/arduino-1.6.1/hardware/tools/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libc.a(lib_a-errno.o):d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libc\errno/../../../../../newlib/libc/errno/errno.c:13: first defined here
collect2.exe: error: ld returned 1 exit status

related to #66 and #170

igrr, 2015-05-06:
Looks like a duplicate of #170.

igrr, 2015-05-12:
Fixed in the latest trunk. Closing this one and keeping #170 open until release.

closed 2015-05-12 by igrr