esp8266/Arduino #188: New Built Version Please

rokison, 2015-05-06:
The current built version for windows 32bit as linked from the main page is quite old, and lacking several key features such as the ESP library.
Please could a more recent built version be uploaded for those of us without the technical competence to build ourselves!

RussMathis, 2015-05-06:
Second this request, the last build was in March and there are many changes. Especially Serial2 support.
Thanks!

igrr, 2015-05-06:
Current trunk has several regressions, my plan was to address them and have a more stable release by the end of next week.

RussMathis, 2015-05-06:
Also, please an article or doc on how to update our libraries/hardware folders so we can simply update those instead of requiring a complete build? I've read some articles describing how to update the hardware and library folders but these seem out of date?

igrr, 2015-05-06:
@RussMathis once we implement #13 this will be taken care of by the Arduino board manager.

marvinroger, 2015-05-06:
Buy you can download the builds from [AppVeyor](https://ci.appveyor.com/project/igrr/Arduino/build/artifacts).

RussMathis, 2015-05-06:
OK, I'll give the AppVeyor build a try. Thanks.

RussMathis, 2015-05-06:
I tried the AppVeyor build. But when trying to start the arduino.exe file it comes up with a dialog titled "Launch4j" saying "An error occurred while starting the applicationr" and will not launch.

igrr, 2015-05-19:
Board manager package released.

closed 2015-05-19 by igrr