esp8266/Arduino #189: analogWrite causes serial output

yasharrashedi, 2015-05-06:
I'm using ESP8266-07 and IDE ARDUINO 1.6.1 - 2015.03.10 (built from source) 
digitalWrite works great!
tested analogWrite with pin 4 and 13, analogWrite causes following serial output (every second):

``` bash
wdt reset
load 0x40100000, len 30336, room 16 
tail 0
chksum 0x84
load 0x3ffe8000, len 2404, room 8 
tail 12
chksum 0xfa
ho 0 tail 12 room 4
load 0x3ffe8970, len 1444, room 12 
tail 8
chksum 0x1f
csum 0x1f
rl
```

my sketch:
``` cpp
int led = 4;
void setup() {
  pinMode(led, OUTPUT);
}

void loop() {
  analogWrite(led, 10);
  delay(1000); 
  analogWrite(led, 100);
  delay(1000);
}
```

reaper7, 2015-05-07:
For me analogWrite with any values <> 0 giving wdt reset

JChatNoir, 2015-05-07:
Tested with esp8266 with usb power and 3.3v regulator and with 3.3 battery pack.
Same output.

i think that the watchdog sees that it not have the time for doing the rest of the calculations...
So it call a reset

see here:

http://www.esp8266.com/viewtopic.php?f=14&t=537

btw: Sorry about my english, im from Colombia, and TKS A LOT to the developers of this Arduino IDE for ESP8266... realy i love your work

auspex-nl, 2015-05-08:
Just tried using analogWrite on my ESP-03, getting that same watchdog reset, I used an empty loop:

```arduino
byte r = 128;
byte g = 1;
byte b = 1;

void setup() {
  analogWrite(12, r);
  analogWrite(13, g);
  analogWrite(14, b);
}  

void loop() {
}
```

cherowley, 2015-05-08:
ESP-12 and I'm getting the wdt reboots too:

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 30608, room 16 
tail 0
chksum 0x7f
load 0x3ffe8000, len 2620, room 8 
tail 4
chksum 0x06
load 0x3ffe8a40, len 3032, room 4 
tail 4
chksum 0x4b
csum 0x4b
rl

Just need pwm to do my led dimming projects with this fantastic bit of software :)

igrr, 2015-05-12:
Fixed in the latest trunk, new release coming this week.

yasharrashedi, 2015-05-13:
pulled latest version and built, now pwm works fine. thank all developers for the great support.

auspex-nl, 2015-05-13:
PWM does indeed work now (no resets) but it does glitch every once in a while. I'm not sure if I should open another issue for this or that it's related to this one. 

I have a RGB ledstrip connected to pins 12,13,14 trough some logic-gate mosfets, now if I use analogWrite to output a PWM signal on those pins it works, but once in a while output of all 3 pins is pulled high (no PWM) for a (very) short duration. It seems to happen very irregularly, sometimes minutes between flashes, sometimes as little as ~5 seconds. Now I think they might be caused by the CPU doing other stuff (wifi?)
The same flashes do happen shortly after applying the power (~4 times about a half a second or so after applying power). I suspect the wifi is interfering with the PWM.

I just tested it with the esp on 160MHz and even though I see a couple of flashes (1-3) on start, it doesn't seem to flash anymore after that.

igrr, 2015-05-13:
There will obviously be irregularities in software PWM output because some of the Wifi interrupts have higher priority than timer1. If you need really stable PWM output I suggest you use an external device and talk to it over I2C (an AVR micro will do).

closed 2015-05-19 by igrr