esp8266/Arduino #194: Latest definition of true/false in c_types.h broke compilation

hallard, 2015-05-07:
Just pulled the repo and built from scratch. It seems true and false definition have changed from file c_types.h, and now I'm unable to compile anymore. for information here below the error messages.

```
In file included from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/c_types.h:9:0,
                 from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/ets_sys.h:11,
                 from D:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_noniso.c:27:
D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/c_types.h:79:25: error: two or more data types in declaration specifiers
 typedef unsigned char   bool;
                         ^
In file included from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/ets_sys.h:11:0,
                 from D:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_noniso.c:27:
D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/c_types.h:79:1: warning: useless type name in empty declaration [enabled by default]
 typedef unsigned char   bool;
 ^
D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/c_types.h:81:0: warning: "true" redefined [enabled by default]
 #define true            (1)
 ^
In file included from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/c_types.h:9:0,
                 from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/ets_sys.h:11,
                 from D:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_noniso.c:27:
d:\arduino-esp8266\build\windows\work\hardware\tools\esp8266\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\4.8.2\include\stdbool.h:34:0: note: this is the location of the previous definition
 #define true 1
 ^
In file included from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/ets_sys.h:11:0,
                 from D:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_noniso.c:27:
D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/c_types.h:82:0: warning: "false" redefined [enabled by default]
 #define false           (0)
 ^
In file included from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/c_types.h:9:0,
                 from D:\arduino-esp8266\build\windows\work/hardware/tools/esp8266/sdk/include/ets_sys.h:11,
                 from D:\arduino-esp8266\build\windows\work\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_noniso.c:27:
d:\arduino-esp8266\build\windows\work\hardware\tools\esp8266\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\4.8.2\include\stdbool.h:35:0: note: this is the location of the previous definition
 #define false 0
 ^
Error compiling.

```

igrr, 2015-05-07:
just fixed in 0018e30

closed 2015-05-07 by igrr

hallard, 2015-05-07:
Correct, this does work again, thanks