esp8266/Arduino #195: Compile failed on Mac (latest commit 102c8242b3)

NSchnitzler, 2015-05-07:
I get the following compiler errors with the actual version:
```
 In file included from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/c_types.h:9:0,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/ets_sys.h:11,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_noniso.c:27:
/Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/c_types.h:79:25: error: two or more data types in declaration specifiers
 typedef unsigned char   bool;
                         ^
In file included from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/ets_sys.h:11:0,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_noniso.c:27:
/Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/c_types.h:79:1: warning: useless type name in empty declaration [enabled by default]
 typedef unsigned char   bool;
 ^
/Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/c_types.h:81:0: warning: "true" redefined [enabled by default]
 #define true            (1)
 ^
In file included from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/c_types.h:9:0,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/ets_sys.h:11,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_noniso.c:27:
/Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/include/stdbool.h:34:0: note: this is the location of the previous definition
 #define true 1
 ^
In file included from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/ets_sys.h:11:0,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_noniso.c:27:
/Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/c_types.h:82:0: warning: "false" redefined [enabled by default]
 #define false           (0)
 ^
In file included from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/c_types.h:9:0,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/sdk//include/ets_sys.h:11,
                 from /Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_noniso.c:27:
/Users/norbertschnitzler/Downloads/Arduino-esp8266/build/macosx/work/Arduino.app/Contents/Java/hardware/tools/esp8266/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.2/include/stdbool.h:35:0: note: this is the location of the previous definition
 #define false 0
 ^
Fehler beim Kompilieren.
```

igrr, 2015-05-07:
same as #194

NSchnitzler, 2015-05-07:
You are right with that. The last commit works! But i get this error now:
```
Uploading 33152 bytes from /var/folders/lh/rxf23rrs43n6tpy40fhx34sh0000gq/T/build3718364451220291383.tmp/ESP_SPI_MIN.cpp_00000.bin to flash at 0x00000000
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
```

I added this to programmers.txt
```pyesptool.name=pyesptool
pyesptool.communication=serial
pyesptool.protocol=esp
pyesptool.program.protocol=esp
pyesptool.program.tool=pyesptool
pyesptool.program.extra_params
```

platform.txt:
```
tools.pyesptool.cmd=esptool.py
tools.pyesptool.cmd.windows=esptool.exe
tools.pyesptool.path={runtime.ide.path}/hardware/tools/esp8266/pyesptool
tools.pyesptool.program.params.verbose=""
tools.pyesptool.program.params.quiet="" 
tools.pyesptool.program.pattern=python "{path}/{cmd}" {program.verbose} --port {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin
```
and of course the esptool.py is in the /tools/pyesptool folder.
Any idea to fix this?
Will there be a out-of-the-box working package for mac?

igrr, 2015-05-07:
Sorry I don't use esptool.py myself so can't comment on that.
Perhaps something's wrong with the hardware setup?

igrr, 2015-05-07:
Edit: sorry, i see now that the error output shows you are in fact using esptool, not esptool.py

NSchnitzler, 2015-05-07:
ok, changing board.txt to 
```modwifi.upload.tool= pyesptool
```
brought success. thank you

closed 2015-05-07 by igrr

NSchnitzler, 2015-05-08:
Hi Ivan,

i think the hardware setup is fine, because the Upload works from  0x00000000 to 0x00008000.
The Message
IOError: [Errno 2] No such file or directory: ‘/var/folders/lh/rxf23rrs43n6tpy40fhx34sh0000gq/T/build7388804888345301808.tmp/ESP_SPI_MIN.cpp_40000.bin'
tells me, he can’t find the named file and it’s right. 
The only bin-files in the folder were:
ESP_SPI_MIN.cpp_00000.bin      
ESP_SPI_MIN.cpp_10000.bin 
is this an error at the platform.txt, because there is no file _4000.bin created?

regards
Norbert

igrr, 2015-05-08:
Yes please do check the latest hardware.txt: the section address has been changed:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/platform.txt#L94

NSchnitzler, 2015-05-08:
After modifying the pyesptool param it’s working now!

isaavedram, 2015-11-15:
" After modifying the pyesptool param it’s working now!  "  what did you modify?

NSchnitzler, 2015-11-15:
The modifications were meant in the broken link above my last post, but i can't remember them.