esp8266/Arduino #201: NullPointerException on upload scketch

leandrogs, 2015-05-07:
I'm trying to upload my code into my ESP using today's IDE build and it gives me this error:

```
Sketch uses 187,424 bytes (35%) of program storage space. Maximum is 524,288 bytes.
java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:246)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:75)
	at processing.app.debug.Compiler.upload(Compiler.java:157)
	at processing.app.Sketch.upload(Sketch.java:1259)
	at processing.app.Sketch.exportApplet(Sketch.java:1201)
	at processing.app.Sketch.exportApplet(Sketch.java:1173)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2432)
	at java.lang.Thread.run(Thread.java:745)
```

I'm using Java7 and I have no error compiling the code, only uploading...

tdicola, 2015-05-08:
I was running into this on a Linux build (with the esp Arduino 1.6.1 release) but it turned out I didn't have the esptool option selected under the programmer menu.  Try checking you have esptool selected and see if that fixes it for you too.

leandrogs, 2015-05-08:
I just download it again a few hours ago and worked just fine...

igrr, 2015-05-12:
@leandrogs can we close this issue?

leandrogs, 2015-05-12:
Yes, we can @igrr . Thanks for your time :)

closed 2015-05-12 by igrr

RossRogers, 2018-09-07:
Breadcrumbs for others...  I got this in arduino IDE when doing Sketch->Upload Using Programmer when my board needed to use Sketch->Upload  ( not "Upload Using Programmer")