esp8266/Arduino #202: Please insert SpifFs

tontito, 2015-05-07:
Please add to library SpifFs

https://github.com/Sermus/ESP8266_ArduinoIDE/tree/master/Espressif/ESP8266/cores/esp8266/Sming/Sming/Services/SpifFS 

 :)
Please @Sermus

Sermus, 2015-05-08:
Tontito, unfortunately i have very limited time to support this for family reasons. Once i have a minute i'll look at this.

sticilface, 2015-05-09:
I asked peter anderson if he'd be willing to help.  I'm way out of my league here.  Can get his details from https://github.com/pellepl/spiffs.  He said he would, but more as an advisor for someone having problems trying to get it to work!

igrr, 2015-05-13:
Okay, spiffs support is almost here, but I can't figure out why the fs images created on the ESP and on the host (OS X) are not compatible.
Same code, same config. Here's the output of SPIFFS_vis running on the host side:
```
# for the fs image created on host side
$ ./make_spiffs -s 0x10000 -l spiffs_dump_64k_my2.bin
10752	edit.htm
   0 //ddddd/dd/dd/dd/dd/dd/dd/dd/dd	era_cnt: N/A
   1 /dd/dd/dd/dd/dd/dd/dd/dd/dd/dd/	era_cnt: N/A
   2 ddi____________________________	era_cnt: N/A
   3 _______________________________	era_cnt: N/A
   4 _______________________________	era_cnt: N/A
   5 _______________________________	era_cnt: N/A
   6 _______________________________	era_cnt: N/A
   7 _______________________________	era_cnt: N/A
era_cnt_max: 1
last_errno:  -10072
blocks:      8
free_blocks: 5
page_alloc:  44
page_delet:  21
# for fs image created on ESP
$ ./make_spiffs -s 0x10000 -l spiffs_dump_64k_2.bin
   0 _______________________________	era_cnt: N/A
   1 i/dd/ddddd////////////dddddddd_	era_cnt: N/A
   2 ddddddddddddddddddddddddddddddd	era_cnt: 11878
   3 i/dd/ddddd////////////ddddddddd	era_cnt: N/A
   4 _______________________________	era_cnt: N/A
   5 _______________________________	era_cnt: N/A
   6 _______________________________	era_cnt: N/A
   7 _______________________________	era_cnt: N/A
era_cnt_max: 11879
last_errno:  -10072
blocks:      8
free_blocks: 5
page_alloc:  64
page_delet:  28
```
Running SPIFFS_check on host emits lots of errors for the the fs image created by ESP.
@pellepl looking at the above output of SPIFFS_vis, do you have any clue to what might be working differently on the PC side vs ESP side? Thanks in advance.

tontito, 2015-05-14:
https://github.com/anakod/Sming
include sample 

@sermus ESP8266 for Arduino ide include Sming core

igrr, 2015-05-14:
Yes, that's what we used.
Never mind, just found the issue, now it works! Will be merged soon!

pellepl, 2015-05-14:
Hmmmm, could it be endianess problems? Ah just got pinged you got it
working, great. What was the problem?

tontito, 2015-05-14:
i`m very very happy :))))

I expected just this property OTA Air update.

good luck.

Thank you :)

igrr, 2015-05-14:
@pellepl on the ESP side the FS was actually offset by 0x1000 bytes. We thought it started at 0x6B000 but it actually started at 0x6C000 due to a bug. Therefore when we dumped the fs image, we had it prefixed with another 0x1000 of empty/garbage bytes.

pellepl, 2015-05-14:
Classic :)

ficeto, 2015-05-14:
```

void sdToSpi(String sdPath, String spiPath){
  File sdFile = SD.open(sdPath.c_str());
  int fileSize = sdFile.size();
  FSFile spiFile = FS.open(spiPath.c_str(), FSFILE_WRITE);
  if(spiFile.write(sdFile) != fileSize){
    Serial.println("Wrote less SPIFFS data than expected!");
  } else {
    Serial.println("SPIFFS file saved!");
  }
  spiFile.close();
  sdFile.close();
}
```

closed 2015-05-19 by igrr