esp8266/Arduino #207: No heap available, failed to malloc 1616

robertgregor, 2015-05-08:
Hello, I have a sketch with the ESP8266WebServer. I have in the sketch a lot of Strings, which are going out i.e. XML documents. Seems to me, that when there is a lot of these strings in the sketch I am getting "No heap available". If I comment out some of the Strings, then everything is fine. Seems like all the Strings are put to the heap??? How to say in the sketch, that the string should be in program space and only be printed when they are needed?
I am using usually this:

 if (server.method() == HTTP_GET) {
  if (server.arg(F("xsd")) == "1") {
    server.send(200, "text/xml", "");
    WiFiClient client = server.client();
    client.print(XSD_SIMPLE_SW_1);
    client.flush();
    client.print(XSD_SIMPLE_SW_2);
    client.flush();
    client.print(XSD_SIMPLE_SW_3);
    client.flush();
    client.print(XSD_SIMPLE_SW_4);
    client.flush();
    client.print(XSD_SIMPLE_SW_5);
    client.flush();
    client.print(XSD_SIMPLE_SW_6);
    client.flush();

And i.e.:
const String XSD_SIMPLE_SW_1 = "<?xml version=\"1.0\" encoding=\"UTF-8\"?><xs:schema xmlns:tns=\"http://remoteswitch.remotehome.com/\" xmlns:xs=\"http://www.w3.org/2001/XMLSchema\" version=\"1.0\" targetNamespace=\"http://remoteswitch.remotehome.com/\"><xs:element name=\"configure\" type=\"tns:configure\"></xs:element><xs:element name=\"configureResponse\" type=\"tns:configureResponse\"></xs:element><xs:element name=\"getStatus\" type=\"tns:getStatus\"></xs:element><xs:element name=\"getStatusResponse\" type=\"tns:getStatusResponse\"></xs:element><xs:element name=\"setCurrentBorderValue\" type=\"tns:setCurrentBorderValue\"></xs:element><xs:element name=\"setCurrentBorderValueResponse\" type=\"tns:setCurrentBorderValueResponse\"></xs:element><xs:element name=\"switchOff\" type=\"tns:switchOff\"></xs:element><xs:element name=\"switchOffResponse\" type=\"tns:switchOffResponse\"></xs:element><xs:element name=\"switchOn\" type=\"tns:switchOn\"></xs:element>";

Links2004, 2015-05-08:
```String``` is a class and is put on the heap
use normal variables to store const data like:
```cpp
static const char XSD_SIMPLE_SW_1[] = "TEXT";
client.write(&XSD_SIMPLE_SW_1[0], sizeof(XSD_SIMPLE_SW_1));
```

Makuna, 2015-05-08:
@Links2004 static string include a null terminator.  So your sample will include a '\0' in the output.  You should be able to do just the following with your definition...
```
client.write(XSD_SIMPLE_SW_1);
```

Links2004, 2015-05-08:
```client.write(XSD_SIMPLE_SW_1);``` will not work its then only send one char.
```client.write(&XSD_SIMPLE_SW_1[0], sizeof(XSD_SIMPLE_SW_1)-1);``` or ```client.print(XSD_SIMPLE_SW_1);``` will do the jop but ```print``` is slower.

robertgregor, 2015-05-08:
I see, OK, thanks. But I belive, if I am using String in the method for example, than at the end of the method, all the objects are removed from the heap. 
What happen, if I an using i.e. client.print("abcdef"); Is abcdef also on the heap all the time?

closed 2015-05-08 by robertgregor

Links2004, 2015-05-09:
```"dfsadf"``` is for the compiler not a String. it is an ```const char *```