esp8266/Arduino #208: Building from source But no esptool Programmer

dogrocker, 2015-05-08:
I had build Arduino from source

```
$ git clone https://github.com/esp8266/Arduino.git
$ cd Arduino/build
$ ant dist
```
OSX

but when I try to upload sketch it no have `esptool` in `Tools > Programmer`
I choose `Tools > Board to generic esp8266 module`

So I try to add `programmers.txt` to
```
esptool.name=esptool
esptool.communication=serial
esptool.protocol=esp
esptool.program.protocol=esp
esptool.program.tool=esptool
esptool.program.extra_params=
```

I also trying to change `platform.txt`
```
tools.esptool.cmd=esptool
tools.esptool.cmd.windows=esptool.exe
tools.esptool.path={runtime.ide.path}/hardware/tools/esp8266

tools.esptool.upload.protocol=esp
tools.esptool.upload.params.verbose=-vv
tools.esptool.upload.params.quiet=
tools.esptool.upload.pattern="{path}/{cmd}" {upload.verbose} -cd {upload.resetmethod} -cb {upload.speed} -cp "{serial.port}" -ca 0x00000 -cf "{build.path}/{build.project_name}_00000.bin" -ca 0x10000 -cf "{build.path}/{build.project_name}_10000.bin"
```

to

```
tools.esptool.cmd=esptool.py
tools.esptool.cmd.windows=esptool.exe
tools.esptool.path=/usr/local/bin/

tools.esptool.program.params.verbose=-vv
tools.esptool.program.params.quiet=
tools.esptool.program.pattern="{path}{cmd}" {program.verbose} -b {upload.speed} -p {serial.port} write_flash 0x00000 "{build.path}/{build.project_name}_00000.bin" 0x40000 "{build.path}/{build.project_name}_40000.bin"
```
but it not usefull I also can't upload the sketch

marvinroger, 2015-05-08:
Choosing esptool in Programmer is not need anymore, see #1d6a74.

dogrocker, 2015-05-09:
@marvinroger Thanks, But what should I select in `Tools > Programmer` ?
I try to upload by select `NodeMCU (ESP8266 ESP-12 Module)` 
but got an error like this

```
Sketch uses 242,476 bytes (5%) of program storage space. Maximum is 4,194,304 bytes.
Uploading 35056 bytes from /var/folders/6m/krhdlzr93md8fhw1gwr6xmhw0000gn/T/build3726772748833940538.tmp/sketch_may09a.cpp_00000.bin to flash at 0x00000000
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
```

dogrocker, 2015-05-09:
Full error

```
Sketch uses 242,476 bytes (5%) of program storage space. Maximum is 4,194,304 bytes.
/Users/kanin/Desktop/Arduino/build/macosx/Arduino.app/Contents/Java/hardware/tools/esp8266/esptool -vv -cd ck -cb 115200 -cp /dev/tty.wchusbserial1420 -ca 0x00000 -cf /var/folders/6m/krhdlzr93md8fhw1gwr6xmhw0000gn/T/build7947418630947899196.tmp/sketch_may09a.cpp_00000.bin -ca 0x10000 -cf /var/folders/6m/krhdlzr93md8fhw1gwr6xmhw0000gn/T/build7947418630947899196.tmp/sketch_may09a.cpp_10000.bin 
esptool v0.4.3 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/tty.wchusbserial1420
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/6m/krhdlzr93md8fhw1gwr6xmhw0000gn/T/build7947418630947899196.tmp/sketch_may09a.cpp_00000.bin success
opening port /dev/tty.wchusbserial1420 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 35056 bytes from /var/folders/6m/krhdlzr93md8fhw1gwr6xmhw0000gn/T/build7947418630947899196.tmp/sketch_may09a.cpp_00000.bin to flash at 0x00000000
	erasing flash
	size: 0088f0 address: 000000
	first_sector_index: 0
	total_sector_count: 9
	head_sector_count: 9
	adjusted_sector_count: 5
	adjusted_size: 005000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 10000
	setting character timeout 100
	done
	setting timeout 1
	setting character timeout 1
	done
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
```

dogrocker, 2015-05-09:
OK I think now I solve this error by using esptool.py

I had change platform.txt from

```
tools.esptool.cmd=esptool
tools.esptool.cmd.windows=esptool.exe
tools.esptool.path={runtime.ide.path}/hardware/tools/esp8266

tools.esptool.upload.protocol=esp
tools.esptool.upload.params.verbose=-vv
tools.esptool.upload.params.quiet=
tools.esptool.upload.pattern="{path}/{cmd}" {upload.verbose} -cd {upload.resetmethod} -cb {upload.speed} -cp "{serial.port}" -ca 0x00000 -cf "{build.path}/{build.project_name}_00000.bin" -ca 0x10000 -cf "{build.path}/{build.project_name}_10000.bin"
```

to

```
tools.esptool.cmd=esptool.py
tools.esptool.cmd.windows=esptool.exe
tools.esptool.path=/usr/local/bin/

tools.esptool.upload.protocol=esp
tools.esptool.upload.params.verbose=-vv
tools.esptool.upload.params.quiet=
tools.esptool.upload.pattern="{path}/{cmd}" -b {upload.speed} -p "{serial.port}" write_flash 0x00000  "{build.path}/{build.project_name}_00000.bin" 0x10000 "{build.path}/{build.project_name}_10000.bin"
```

but in the lasted Arduino_IDE is use 0x40000 but now it use 0x10000 I don't know why I can't use 0x40000

because 01000h User application, slot 1
41000h User application, slot 2 right ?
ref: https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map

igrr, 2015-05-19:
Esptool can now work with CH340 usb-ttl chips found in NodeMCU board. You can get it with the latest board manager package.

dogrocker, 2015-05-19:
Thanks, @igrr

closed 2015-05-23 by igrr

acestu, 2015-10-06:
Hi, now that there is no esptool , which programmer do you select ?

Thanks
Acestu

col000r, 2015-11-03:
Same problem. What do I select as a programmer now? Arduino as ISP?

marvinroger, 2015-11-03:
It does not matter, it won't be the selected one that will be used anyway.

acestu, 2015-11-05:
Great thanks, sorted now...